kitesurfer1404/WS2812FX #5: problem with the file "WS2812FX.h

papybricol, 2016-12-19:
Hello,
I try to use your Library,  but with Arduino 1.6.13,
 and with Arduino : 1.6.5 (Windows 10), Carte : "Arduino Uno".
IDE arduino replies in compilation:
"In file included from test-effets-neopixels-fx.ino:3:0:
C:\Users\jean-louis\Documents\1-ROBOTIQUE...techno\1-ARDUINO\2-DOSSIER des sketchs\libraries\WS2812FX/WS2812FX.h:123:37: error: 'neoPixelType' has not been declared
     WS2812FX(uint16_t n, uint8_t p, neoPixelType t) : Adafruit_NeoPixel(n, p, t) {
                                     ^
Erreur lors de la compilation."

Perhaps, it is in this line of       file  " .........h"  ??
class WS2812FX : public Adafruit_NeoPixel {  typedef void (WS2812FX::*mode_ptr)(void);
  public:
    WS2812FX(uint16_t n, uint8_t p, **neoPixelType** t) : Adafruit_NeoPixel(n, p, t) {

thanks for your help
(Excuse my english language, it's very bad!)
papybricol

kitesurfer1404, 2016-12-19:
Do you have the latest NeoPixel library installed? Can you post your code?

papybricol, 2016-12-19:
The files of the NeoPixel library was dated    of 03/07/2014.
To morrow,  i will download the latest version.

I tried with the exemple "serial control"   and  this other program, the messages are the same for the 2.

`#include <Adafruit_NeoPixel.h>              //error if absent
#include <WS2812FX.h>

#define LED_COUNT 16
#define LED_PIN 8

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

unsigned long previousMillis = 0;
const long interval = 10000; 

int effet = -1;

void setup() {
   ws2812fx.init();
   ws2812fx.setBrightness(25);
   ws2812fx.setSpeed(200);
   ws2812fx.setColor(255, 255, 0);
   ws2812fx.setMode(FX_MODE_CHASE_RAINBOW);
   ws2812fx.start();
   Serial.begin(115200);
}

void loop() {
  ws2812fx.service();

  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval) {
    ws2812fx.stop();
    effet++;
    if(effet >= MODE_COUNT) effet = 0;
    Serial.println(ws2812fx.getModeName(effet));
    ws2812fx.setMode(effet);
    ws2812fx.start();
    previousMillis = currentMillis;
  }
}`

kitesurfer1404, 2016-12-19:
I found some hints 03/07/2014 might be to old while googleing the error message.

Not related to the issue: You don't have to call ws2812fx.stop() before you change the effect. You might want to try:
`if (currentMillis - previousMillis >= interval) {`
`  ws2812fx.setMode((ws2812fx.getMode() + 1) % ws2812fx.setModeCount());`
`  Serial.println(ws2812fx.getModeName(ws2812fx.getMode()));`
`  previousMillis = currentMillis;`
`}`

;-)

papybricol, 2016-12-19:
YES!!!
It is OK (compilation) with the latest version  1.0.6 .of the neoPixel Library (  with Arduino IDE 1.6.5)
But it is necessary to add the command     `#include <Adafruit_NeoPixel.h>`
in the 2 programs.

GREAT THANKS !

Tomorrow i will test with IDE  1.6.13 and connect the leds

papybricol, 2016-12-20:
Hello,
With IDE 1.6.13,  everything  is Ok (for compilation)
No need to add   ` #include <Adafruit_NeoPixel.h>`

Have a good day