kitesurfer1404/WS2812FX #6: WS2812FX error on compiling for ESP8266 and Arduino

questuk, 2016-12-20:
Hi,

I have been trying to get your  WS2812FX working with my ESP8266, but i get errors on compile ?

I am using:
name=Adafruit NeoPixel version=1.0.6
ESP8266 using Arduino 1.6.5  (Reluctant to update this as Adafruit_NeoPixel works with all my other projects)

Regards

Gaz

> In file included from esp8266_webinterface.ino:43:0:
> G:\DATA\Software - Data\Arduino_Sketchbooks\Sketchbook + PubSubClient V2.3\libraries\WS2812FX-master/WS2812FX.h:55:31: fatal error: Adafruit_NeoPixel.h: No such file or directory
>  #include <Adafruit_NeoPixel.h>
>                                ^
> compilation terminated.
> Error compiling.

kitesurfer1404, 2016-12-22:
If you don't want to update, try adding
`#include <Adafruit_NeoPixel.h>`
to the beginning of your code file. See: https://github.com/kitesurfer1404/WS2812FX/issues/5
Let me know, if it helps.

questuk, 2016-12-22:
Hi,

Thanks for replying, here are the results.
Thanks Gaz

If i try to compile your example serial_control. I get the following:

> Arduino: 1.6.5 (Windows 7), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 4M (3M SPIFFS)"
> 
> In file included from serial_control.ino:1:0:
> G:\DATA\Software - Data\Arduino_Sketchbooks\Sketchbook + PubSubClient V2.3\libraries\WS2812FX-master/WS2812FX.h:55:31: fatal error: Adafruit_NeoPixel.h: No such file or directory
>  #include <Adafruit_NeoPixel.h>
>                                ^
> compilation terminated.
> Error compiling.

  
If I then add as you suggest #include <Adafruit_NeoPixel.h> at the start of the ino sketch it DOES compile !

I then tried your other example esp8266_webinterface and again as you suggest #include <Adafruit_NeoPixel.h> at the start of the ino sketch it does NOT compile ?
  
```
#include <Adafruit_NeoPixel.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WS2812FX.h>

#include "index.html.h"
#include "main.js.h"`
```

> Arduino: 1.6.5 (Windows 7), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 4M (3M SPIFFS)" 
> 
> esp8266_webinterface.ino:80:0: warning: "DEFAULT_COLOR" redefined [enabled by default]
> In file included from esp8266_webinterface.ino:44:0:
> G:\DATA\Software - Data\Arduino_Sketchbooks\Sketchbook + PubSubClient V2.3\libraries\WS2812FX-master/WS2812FX.h:60:0: note: this is the location of the previous definition
>  #define DEFAULT_COLOR 0xFF0000
>  ^
> esp8266_webinterface.ino:81:0: warning: "DEFAULT_BRIGHTNESS" redefined [enabled by default]
> In file included from esp8266_webinterface.ino:44:0:
> G:\DATA\Software - Data\Arduino_Sketchbooks\Sketchbook + PubSubClient V2.3\libraries\WS2812FX-master/WS2812FX.h:57:0: note: this is the location of the previous definition
>  #define DEFAULT_BRIGHTNESS 50
>  ^
> esp8266_webinterface.ino:82:0: warning: "DEFAULT_SPEED" redefined [enabled by default]
> In file included from esp8266_webinterface.ino:44:0:
> G:\DATA\Software - Data\Arduino_Sketchbooks\Sketchbook + PubSubClient V2.3\libraries\WS2812FX-master/WS2812FX.h:59:0: note: this is the location of the previous definition
>  #define DEFAULT_SPEED 150
>  ^
> esp8266_webinterface.ino:83:0: warning: "DEFAULT_MODE" redefined [enabled by default]
> In file included from esp8266_webinterface.ino:44:0:
> G:\DATA\Software - Data\Arduino_Sketchbooks\Sketchbook + PubSubClient V2.3\libraries\WS2812FX-master/WS2812FX.h:58:0: note: this is the location of the previous definition
>  #define DEFAULT_MODE 0
>  ^
> esp8266_webinterface.ino: In function 'void setup()':
> esp8266_webinterface:102: error: 'modes_setup' was not declared in this scope
> esp8266_webinterface:113: error: 'wifi_setup' was not declared in this scope
> esp8266_webinterface:116: error: 'srv_handle_index_html' was not declared in this scope
> esp8266_webinterface:117: error: 'srv_handle_main_js' was not declared in this scope
> esp8266_webinterface:118: error: 'srv_handle_modes' was not declared in this scope
> esp8266_webinterface:119: error: 'srv_handle_set' was not declared in this scope
> esp8266_webinterface:120: error: 'srv_handle_not_found' was not declared in this scope
> esp8266_webinterface.ino: In function 'void loop()':
> esp8266_webinterface:138: error: 'wifi_setup' was not declared in this scope
> esp8266_webinterface.ino: In function 'void srv_handle_index_html()':
> esp8266_webinterface:207: error: 'class ESP8266WebServer' has no member named 'send_P'
> esp8266_webinterface.ino: In function 'void srv_handle_main_js()':
> esp8266_webinterface:211: error: 'class ESP8266WebServer' has no member named 'send_P'
> 'modes_setup' was not declared in this scope
> 
>   This report would have more information with
>   "Show verbose output during compilation"
>   enabled in File > Preferences.
>

kitesurfer1404, 2016-12-22:
Do you have the latest ESP8266WebServer lib installed?
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer

questuk, 2016-12-22:
Hi,

Mine is dated 23/7/15

I did try and replace that with the one you pointed at but still got errors 

Regards

Gaz

kitesurfer1404, 2016-12-22:
I'll try to find out the details after 01/01/2017 or maybe I can find someone at #33c3 to help out with it. Stay tuned.

kitesurfer1404, 2016-12-23:
I had some time and wanted to investigate this issue further. I tested the examples with Arduino IDE 1.8.0 and version 2.3.0 of the Arduino Board Code from https://github.com/esp8266/Arduino

` 'modes_setup' was not declared in this scope`
that's strange, because it is defined in the example itself.

Please try to move the setup() and loop() to the bottom of the example. Maybe your compiler needs the functions defined before they are called the first time in the code. Can't remember, if this was an issue with older Arduino IDEs.

questuk, 2016-12-23:
Hi,

I tried the above and got 

```
Build options changed, rebuilding all
esp8266_webinterface.ino:80:0: warning: "DEFAULT_COLOR" redefined [enabled by default]
In file included from esp8266_webinterface.ino:44:0:
G:\DATA\Software - Data\Arduino_Sketchbooks\Sketchbook + PubSubClient V2.3\libraries\WS2812FX-master/WS2812FX.h:60:0: note: this is the location of the previous definition
 #define DEFAULT_COLOR 0xFF0000
 ^
esp8266_webinterface.ino:81:0: warning: "DEFAULT_BRIGHTNESS" redefined [enabled by default]
In file included from esp8266_webinterface.ino:44:0:
G:\DATA\Software - Data\Arduino_Sketchbooks\Sketchbook + PubSubClient V2.3\libraries\WS2812FX-master/WS2812FX.h:57:0: note: this is the location of the previous definition
 #define DEFAULT_BRIGHTNESS 50
 ^
esp8266_webinterface.ino:82:0: warning: "DEFAULT_SPEED" redefined [enabled by default]
In file included from esp8266_webinterface.ino:44:0:
G:\DATA\Software - Data\Arduino_Sketchbooks\Sketchbook + PubSubClient V2.3\libraries\WS2812FX-master/WS2812FX.h:59:0: note: this is the location of the previous definition
 #define DEFAULT_SPEED 150
 ^
esp8266_webinterface.ino:83:0: warning: "DEFAULT_MODE" redefined [enabled by default]
In file included from esp8266_webinterface.ino:44:0:
G:\DATA\Software - Data\Arduino_Sketchbooks\Sketchbook + PubSubClient V2.3\libraries\WS2812FX-master/WS2812FX.h:58:0: note: this is the location of the previous definition
 #define DEFAULT_MODE 0
 ^
esp8266_webinterface.ino: In function 'void srv_handle_index_html()':
esp8266_webinterface:156: error: 'class ESP8266WebServer' has no member named 'send_P'
esp8266_webinterface.ino: In function 'void srv_handle_main_js()':
esp8266_webinterface:160: error: 'class ESP8266WebServer' has no member named 'send_P'
'class ESP8266WebServer' has no member named 'send_P'
```

kitesurfer1404, 2017-01-01:
I think I can not help any further. I installed a fresh Arduino IDE on a machine that has never seen Arduino before with all needed libs in their current versions and did not get any errors. I'm going to label this as "help wanted". Maybe someone else can help you. I'm sorry I cannot resolve this issue now.

questuk, 2017-01-02:
OK  thanks for your help, I wll upgrade to the latest Arduino IDE at some stage, then I am sure it will work !

Just one last question which version of Arduino IDE did you install on the PC that had never seen Arduino?

Thanks

On 01/01/2017 15:40, kitesurfer1404 wrote:

I think I can not help any further. I installed a fresh Arduino IDE on a machine that has never seen Arduino before with all needed libs in their current versions and did not get any errors. I'm going to label this as "help wanted". Maybe someone else can help you. I'm sorry I cannot resolve this issue now.

kitesurfer1404, 2017-01-02:
I used 1.8.0 (Linux 64 Bit) of Arduino, I followed https://github.com/esp8266/Arduino/ and added
`http://arduino.esp8266.com/stable/package_esp8266com_index.json` 
to the IDE, updated the packages and installed the ESP8266 Board stuff.

questuk, 2017-01-09:
Hi Kitesurfer,

Just to update on this i installed Arduino  1.8.0  and everything went fine, now all working, you created a great project !

The one thing I wanted and didn't realize it didn't have in it's effects, was a 'candle flicker effect' .
As you know your code much better than I can ever do, would it be possible for you to add it to your code ?

I don't mind if its just some code I have to add to the WS2812FX.h  just for me, or if you feel it would be useful for others include it in your project.

I can test it for you , no problem.

Thanks 

Regards 

Gaz

kitesurfer1404, 2017-01-09:
Hi,

thanks for the feedback. I'm going to close this issue.

There is a candle flicker effect in a pull request. You might want to try that out and give some feedback. I had not time to test it, yet. I'm going to integrate that - hopefully soon. Did not expect that much feedback, so there was no development branch at first. I created that and I'm also working on some sound-to-light possibilities. I hope to get all to the master at the weekend. Please stay tuned, maybe with some patience.

Kind regards,
Harm

closed 2017-01-09 by kitesurfer1404