kitesurfer1404/WS2812FX #8: Aircoookie branch does not compile ?

questuk, 2017-01-10:
Hi,

I tried the branch version by Aircoookie but got the following on compiling.

I am using portable Arduino 1.8.0

I renamed the folder 'WS2812FX-master-fire' to keep it separate from your original. 

Apologies if its something I am doing wrong.

Thanks

Gaz

> Arduino: 1.8.0 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"
> 
> In file included from G:\DATA\Temporary\arduino-1.8.0 -1\portable\sketchbook\libraries\WS2812FX-master-fire\examples\esp8266_webinterface\esp8266_webinterface.ino:43:0:
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h:315:29: error: ISO C++ forbids declaration of 'mode_fire_flicker' with no type [-fpermissive]
> 
>        mode_fire_flicker(void),
> 
>                              ^
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h:316:34: error: ISO C++ forbids declaration of 'mode_fire_flicker_soft' with no type [-fpermissive]
> 
>        mode_fire_flicker_soft(void),
> 
>                                   ^
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h:317:32: error: ISO C++ forbids declaration of 'mode_fire_flicker_int' with no type [-fpermissive]
> 
>        mode_fire_flicker_int(int),
> 
>                                 ^
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h:319:5: error: ISO C++ forbids declaration of 'boolean' with no type [-fpermissive]
> 
>      boolean _running;
> 
>      ^
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h:319:5: error: expected ';' at end of member declaration
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h:319:5: error: declaration of 'int WS2812FX::boolean' [-fpermissive]
> 
> In file included from sketch\esp8266_webinterface.ino.cpp:1:0:
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:186:17: error: changes meaning of 'boolean' from 'typedef uint8_t boolean' [-fpermissive]
> 
>  typedef uint8_t boolean;
> 
>                  ^
> 
> In file included from G:\DATA\Temporary\arduino-1.8.0 -1\portable\sketchbook\libraries\WS2812FX-master-fire\examples\esp8266_webinterface\esp8266_webinterface.ino:43:0:
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h:319:13: error: '_running' does not name a type
> 
>      boolean _running;
> 
>              ^
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h: In constructor 'WS2812FX::WS2812FX(uint16_t, uint8_t, neoPixelType)':
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h:171:44: error: cannot convert 'int (WS2812FX::*)()' to 'WS2812FX::mode_ptr {aka void (WS2812FX::*)()}' in assignment
> 
>        _mode[FX_MODE_FIRE_FLICKER]          = &WS2812FX::mode_fire_flicker;
> 
>                                             ^
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h:172:44: error: cannot convert 'int (WS2812FX::*)()' to 'WS2812FX::mode_ptr {aka void (WS2812FX::*)()}' in assignment
> 
>        _mode[FX_MODE_FIRE_FLICKER_SOFT]     = &WS2812FX::mode_fire_flicker_soft;
> 
>                                             ^
> 
> G:\DATA\Temporary\arduino-1.8.0 -1\portable\Sketchbook\libraries\WS2812FX-master-fire/WS2812FX.h:226:7: error: '_running' was not declared in this scope
> 
>        _running = false;
> 
>        ^
> 
> exit status 1
> Error compiling for board NodeMCU 1.0 (ESP-12E Module).
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
>

kitesurfer1404, 2017-01-10:
> I renamed the folder 'WS2812FX-master-fire' to keep it separate from your original.
That might be the problem. Rename the original "WS2812FX_backup" and the Aircookie branch "WS2812FX". That might work.

questuk, 2017-01-10:
Hi,

I have done that now and still the same errors.

Regards

Gaz

kitesurfer1404, 2017-01-10:
Same here. There is an error in Aircookies code. You can fix it manually: WS2812FX.h line 314 replace ; with , and line 317 replace , with ;

questuk, 2017-01-10:
Hi,

I have corrected as you suggest and it does now compile, but I am getting the following on the com port serial monitor:

> Starting...
> 
> Exception (3):
> epc1=0x40100576 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40007571 depc=0x00000000
> 
> ctx: cont 
> sp: 3fff09f0 end: 3fff0c30 offset: 01a0
> 
> >>>stack>>>
> 3fff0b90:  00000000 0000000a 3ffe87ce 3ffefc08  
> 3fff0ba0:  3fffdad0 0000000b 3ffefb4c 40207c3d  
> 3fff0bb0:  3ffe8d58 00000000 3ffefb4c 3ffefc08  
> 3fff0bc0:  00001390 3ffefa2c 3ffefa2c 40208014  
> 3fff0bd0:  3fffdad0 00000000 3ffefa2c 40208063  
> 3fff0be0:  3fffdad0 00000000 3ffefb4c 402023c9  
> 3fff0bf0:  feefeffe feefeffe feefeffe feefeffe  
> 3fff0c00:  feefeffe feefeffe feefeffe 3ffefc08  
> 3fff0c10:  3fffdad0 00000000 3ffefc02 402087c0  
> 3fff0c20:  feefeffe feefeffe 3ffefc10 40100718  
> <<<stack<<<

kitesurfer1404, 2017-01-16:
What kind of powersupply do you use? Are the LEDs powered by the same powersupply? Most issues with the ESP I had were power related. Does the code run without LEDs attached? Does it run with just one or two LEDs activated in the code `#define LED_COUNT 2`

questuk, 2017-01-17:
Hi,

I have had time to try it again.

I have tested it with no LEDS attached and still it crashes as above.
It seems as though the added code is causing it to crash ?
Have you got the above code to work using Leds ? 

Regards

Gaz

kitesurfer1404, 2017-01-18:
Not tried on hardware yet, just compiling. Strange behaviour. I had something similar yesterday with another code. I used

`ESP.eraseConfig();`
`ESP.reset();`

in setup() once and removed it afterwards. That erased my broken config and all settings. Afterwards the ESP was running fine.
Maybe you can find some information for _Exception (3)_ online. I'll try to test the code next week.

Aircoookie, 2017-01-29:
I'm sorry for the semicolon error in my branch, it is corrected now. Those exceptions only occur if something in the code is illegal. It happened to me once because of a method calling itself recursivly.
Overall the ESP seems to behave unpredictably in some situations, for example I used some NTP code that works perfectly, the ESP CPU continues running but after 1-48 hours it doesn't respond to web requests any more. Strange stuff. questuk, does the exception only occur with my branch, or also with kitesurfers?

questuk, 2017-01-30:
HI ,

Kitesurfer, your code works 100% all of the time. Altough I have only run it for about 2 hours, I will leave it on for a longer test.

Aircookie fails 100% of the time.

I am using Arduino 1.8 and NodeMCU hardware, which has ESP12E inside, I think.

Regards

Gaz

Aircoookie, 2017-01-30:
It is strange that my code doesn't work for you given that it differs so slightly from kitesurfers. I tested my code with ESP-12E and it ran for 100+ hours without problems. The only thing I can imagine the issue being is that I use outdated Arduino 1.6 and ESP library from August 2016. Maybe it works in that version but no longer in newer releases. I'll try updating my dependencies and report back soon.

questuk, 2017-02-01:
Hi,

Just to let you know, I ran Kitesurfers original for 36 hours non stop and it worked perfectly.

So the problem is somewhere in the Aircookie version I think ?

Regards

Gaz

kitesurfer1404, 2017-02-02:
I guess, I found something. Line 68 in WS2812FX.h
https://github.com/Aircoookie/WS2812FX/blob/36ba73c88741496b066347640a9ae36c815c33f4/WS2812FX.h#L68

Try to replace 45 with 47. Otherwise the array-size of

`    const char*`
`      _name[MODE_COUNT];`

and

`    mode_ptr`
`_mode[MODE_COUNT];`

will not be correct which might end in strange behavior and/or not compiling code. Have no hardware right now to check, but I guess that might be it.

questuk, 2017-02-02:
Hi,

Well done it now compiles and runs correctly in my browser and can select 'Fire Flicker'.

Works with your corrections:
> There is an error in Aircookies code. You can fix it manually: WS2812FX.h line 314 replace ; with , and line 317 replace , with ;
**and**
> I guess, I found something. Line 68 in WS2812FX.h
> https://github.com/Aircoookie/WS2812FX/blob/36ba73c88741496b066347640a9ae36c815c33f4/WS2812FX.h#L68
> 
> Try to replace 45 with 47

Aircoookie, 2017-02-02:
I guess I'll be faster. I'm sorry, I corrected this mistake in my project but not in the repo.
Glad it is working now!

kitesurfer1404, 2017-02-02:
Had to repurpose a set of LEDs to try your code right now. Nice work! I like the effects!
I also added external trigger functionality for sound-to-light. beta testers needed ;-) https://github.com/kitesurfer1404/WS2812FX/issues/13

closed 2017-02-02 by kitesurfer1404