kitesurfer1404/WS2812FX #9: Color scale on android chrome

k1gavrilov, 2017-01-16:
Thank you for this wonderful library and examples. I use an example from the web server on esp. it's great.
but in the chrome browser on smartphones color scale it is not working properly.
shade similar, but very few colors.
on the PC everything is fine.
maybe you can help.

kitesurfer1404, 2017-01-18:
I know this issue. Same here. Firefox on Android seems to work a little bit better. But the color scale is not the way I want it in the end.
Feel free to come up with a better solution. ;-) I could imagine two or three horizontal bars at the top. One for color, one for saturation. Maybe one for brightness.
I have this on my list but not so much time to fix it in the next weeks. Bear with me but stay tuned. Maybe someone improves the example and does a pull-request to the development branch. *hint*

k1gavrilov, 2017-01-19:
Thank you for your help. Try another browser. I'm still a noob at this. but the backlight is already working with his wife and baby happy. I will try.

aj-gh, 2017-03-12:
[FlexiColorPicker](https://github.com/DavidDurman/FlexiColorPicker) seems to work pretty well for that purpose; maybe that would be an alternative?

closed 2018-05-24 by moose4lord