kitesurfer1404/WS2812FX #12: i want to edit or add some effect

LedArts, 2017-02-02:
hello sir, i want to add more effects. i am using arduino but when i open auto cycle mode it doesn't show .h files how can i edit this program????

kitesurfer1404, 2017-02-02:
The modes are stored in WS2812FX.h and WS2812FX.cpp in the main-directory. You can change them there and add more effects to the library. auto_mode_cycle.ino is only a example for using the library, not the effects itself. 
You can also add your own effects in your own code and call them from your loop-method as you would do with the Adafruit NeoPixel library without WS2812FX.

Hope this helps.

LedArts, 2017-02-02:
yes sir it helps a lot. thank you sir

closed 2017-02-02 by kitesurfer1404

LedArts, 2017-02-02:
i have one more question sir. where n how can i open .cpp or .h files in arduino?????

kitesurfer1404, 2017-02-02:
I'm not sure if that is possible. You can edit them using a normal text-editor (e.g. notepad or wordpad in windows). You might want to take a look at notepad++ which is a great text editor that also highlights code etc.

LedArts, 2017-02-02:
ok sir thank you