kitesurfer1404/WS2812FX #13: Call for beta-testers

kitesurfer1404, 2017-02-02:
Please check out the development branch and try it. Your feedback is welcome. 
Please add general feedback here and feel free to open new issues on errors. 

Thanks for your support!

DeuxVis, 2017-02-03:
Anything we need to pay special attention to, or just general usage testing ?

kitesurfer1404, 2017-02-03:
At the moment: general usage testing. I think it's pretty stable. I have it running on several installations with different kind of controllers. Seems to work. 
Sound-to-Light needs testing. 
I was thinking about removing the name strings and put them into a second library because of code size. But I don't know if that's something beginners might find too difficult to use. Maybe some kind of compile-option? I'm not into compiler options that much yet to be able to judge it's usefullness.

kitesurfer1404, 2017-02-04:
I got feedback that the ESP-demo is not working on iPad. I don't have any i-devices. Can someone check that demo on iPad?

Aircoookie, 2017-02-04:
Thank you for pulling my request! I tried out the trigger function, it works as expected, however I don't think it it optimal for the sound-to-light function (untested, since I have no mic), because it only works for a few effects, like the random color (8). Therefore it would be more useful to just have a FX mode that sets the brightness according to the analog input IMHO.
However I still think the trigger function is useful, for example to update the color instantly rather than at the next service() refresh.

toblum, 2017-02-05:
Hi, I integrated your library in my Project McLighting (also on Github https://github.com/toblum/McLighting) today. It worked quite well and I was able to switch to your library quite fast. I hope thats ok for you. I referenced your project in the code and readme. 
I did not notice that there is a dev branch, but i will try out the new Features soon.

DeuxVis, 2017-02-06:
I just had a quick run through the basic examples - automatic cycling and serial command - this weekend.
Arduino Uno, a single 42 x WS2812 strip.
Everything worked fine.

kitesurfer1404, 2017-02-06:
Merged dev to master. Seems to be stable and I can live with the current functionality. Will add a better sound-to-light example soon.
@Aircoookie : won't add a brightness-effect, because one can easily use .setBrightness for that which updates the brightness immediately without need of service(). I'll take care in my example to show that in detail.
@toblum : Thanks for testing. Nice project with McLighting. I added a reference in the README as well. 
@DeuxVis : Thanks for testing and feedback.

I was thinking of a second lib using the FastLED library. But I did not try that one yet. Don't know how much code can be reused with that lib or if it is even possible to add a switch to the code that decides which lib to use (NeoPixel / FastLED). I'm looking into that in a not so distant future, I hope.

Aircoookie, 2017-02-06:
Yeah, using .setBrightness() directly is what I meant, it requires no additional library functions. For example I use it in my WLED project (https://github.com/Aircoookie/WLED) to have a smooth fade transition. Regarding FastLED, it worked for me without problems on an AVR Arduino Uno but never on an ESP. Since then I used the Adafruit library and it works fine on the ESP too. Maybe the reason for FastLED not working for me could just be that I connect the strip in a lazy way, I've never had any issues using no resistor/voltage step-up/capacitor for the input pins. I've started to work on some new FX, will add them to my fork in a few days!

DeuxVis, 2017-02-07:
@kitesurfer1404 If you are considering FastLed maybe give a glance at https://github.com/Makuna/NeoPixelBus too - No idea if this is better, just mentioning its existence.

toblum, 2017-02-07:
@DeuxVis I saw a good comparison of the three libraries at: https://github.com/krzychb/EspFire2012 from the maker of EspFire2012.

closed 2018-05-24 by kitesurfer1404