kitesurfer1404/WS2812FX #14: Only 19 LED's of 144 working

anthonyjclarke, 2017-02-05:
Hi, firstly I am new here!  So please be patient.

Using sample code sketch on a 144LED Strip - http://www.banggood.com/5V-WS2812B-43W-144-SMD-5050-RGB-LED-Strip-Waterproof-IP65-Individual-Addressable-p-1042712.html and only 19 of the LED's are being used.

Tried the Adafruit NeoPixel Library and made no difference?

Any suggestions?

#define LED_COUNT 144
#define LED_PIN 11
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

kitesurfer1404, 2017-02-05:
Hi! When using the NeoPixel lib didn't help, it might be a hardware issue. Take a closer look at the 19th LED and it's connections to the next LED. Maybe you can also try to insert the signal behind the 19th LED instead of the beginning of the strip. The strip has three copper contacts between each LED. Most of the time + and - on the outside and signal in the center. Try to insert power from the other side of the strip. And/or signal somewhere behind the 19th LED. 

That might give a hint on any unsoldered LED pins. Maybe pushing the 19th or 20th LED down with careful force might "fix" it temporarily. Then try to resolder the LED carefully (not too hot, not too long).

DarkBeardRUS, 2017-02-16:
Hi
What power supply do you use?
Each LED consumes 60mA. At 144 LEDs glow when you need white light maximum brightness should be at approximately 5V 9A.
Be sure to connect the power to the LED strip at both ends!

anthonyjclarke, 2017-02-23:
Appreciate both your help, tried all permutations, must be a dodgy LED Strip.  Will try buying another, unless you have any other testing suggestions?

kitesurfer1404, 2017-02-24:
Last suggestion maybe: remove (cut before and behind) LED 20 and try to run the rest of the strip. Re-solder first 19 LED-strip to 21-144 LED-strip.

closed 2017-04-25 by kitesurfer1404