kitesurfer1404/WS2812FX #15: change led count in setup() or loop()

renat2985, 2017-02-16:
How I can change the led count in setup() or loop() void?
Thanks

kitesurfer1404, 2017-02-16:
You might want to take a look at the updateLength function of the NeoPixel library.
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.h#L134
I did not test this. But it seems to be the right function to call.

renat2985, 2017-02-16:
Not Work. 
[image]

kitesurfer1404, 2017-02-16:
You have to call it on your WS2812FX-object. Like WS2812FX.updateLength(ledCount); But with the corresponding name of the object you declared in your code.

renat2985, 2017-02-16:
Sketch compile ok, but led number not updated.

kitesurfer1404, 2017-02-16:
Why do you want to update the length?

kitesurfer1404, 2017-02-16:
You might need to use the empty constructor and then call updateType and updateLength. See
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.cpp#L45
(untested).

DarkBeardRUS, 2017-03-10:
At the beginning, when initializing an LED tape, it costs a length of 0. The change occurs during the Setup, when the sketch reads the configuration file. Then the length of the LED strip is updated.
void LightEffect :: updateLength (uint32_t  _ulc) {
   _led_count = _ulc;
   Adafruit_NeoPixel :: updateLength (_led_count);
  }
But I just increase the size of the LED strip, did not try to reduce it.

renat2985, 2017-04-25:
@kitesurfer1404 Please add a simple change led count with setup or loop.

MTJoker, 2017-05-01:
I have the same issue- my number of leds, the led pin and the type is read during run-time from a config file when the led object is already instantiated. So an empty constructor needs to be called on instantiation, and the parameters adjusted later. This works perfectly with the Neopixel library itself. But it does not work with yours, as your class is derived and does not provide an empty constructor. I have done an enhancement to allow this. If you want I can post a pull-request.

kitesurfer1404, 2017-05-01:
Cool. Please do! :+1:

MTJoker, 2017-05-01:
Unfortunately can't send pull request right now (tells me authentification failed).
Will need to figure out what's the problem... :-)

idolpx, 2017-07-26:
I needed this ability too and added it here.

https://github.com/kitesurfer1404/WS2812FX/pull/32

closed 2018-05-24 by moose4lord