kitesurfer1404/WS2812FX #17: Issue 304.

popekpl, 2017-02-21:
Hi, I have a problem, because I have a strip with 492 WS2812B leds , and unfortunately I can't use whole of these leds with this library.  Everything works with 304 leds or lower. When I'm trying to declare more than 304 leds my stripe is not showing anything.

`#include <WS2812FX.h>

#define LED_COUNT 492
#define LED_PIN 7

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(100);
  ws2812fx.setSpeed(200);
  ws2812fx.setMode(FX_MODE_RAINBOW_CYCLE);
  ws2812fx.start();
}

void loop() {
 
  ws2812fx.service();
}`

How can I fix this and how can I use this with all my long stripe?
I'm looking forward to your answear.

kitesurfer1404, 2017-02-21:
What controller are you using?

popekpl, 2017-02-21:
Arduino Nano

popekpl, 2017-02-21:
But everything works with Adafruit_NeoPixel. I can use 492 leds.

kitesurfer1404, 2017-02-21:
I think that might be a RAM problem. With only 2kb of RAM you can have a maximum of ~680 pixels when your code does not need any RAM. That's not possible, so the real limit is well under that.
When using the Adafruit library you don't have any effects and so on that use RAM, too. So the limit is higher than with the WS2812FX. I don't think there is anything that can be changed without changing the concept of this library.
What does the compiler output say on code and RAM usage?
Also, see https://learn.adafruit.com/adafruit-neopixel-uberguide/arduino-library#pixels-gobble-ram

popekpl, 2017-02-21:
18500 (60% of RAM)

popekpl, 2017-02-21:
So what should I do ?

kitesurfer1404, 2017-02-21:
I guess there are these options:
1. just copy the effects you need to your own code. Not that helpfull, I guess.
2. fork the library and kick all effects you don't need. There are some heavy RAM eating ones.
3. try to remove all the string-stuff, if not needed. I guess that might be using a lot of RAM when using serial.print to put out the names. But I'm not sure on that.
4. feel free to give feedback - or better pull requests - on RAM improvements ;-)
5. upgrade hardware ;-) ESP8266 or Arduino Mega should have enough RAM to play with 500 LEDs.

Maybe someone can assist on this. I'll leave this open and tag it "help needed".

popekpl, 2017-02-21:
Ok, thank you. Can you tell me how can I delete useless effects ?

kitesurfer1404, 2017-02-21:
In WS2812FX.h:
Update the mode count: https://github.com/kitesurfer1404/WS2812FX/blob/master/WS2812FX.h#L69 
Re-number the modes: https://github.com/kitesurfer1404/WS2812FX/blob/master/WS2812FX.h#L71
Remove FXs from _mode (delete corresponding lines) https://github.com/kitesurfer1404/WS2812FX/blob/master/WS2812FX.h#L126
Remove FXs from _name https://github.com/kitesurfer1404/WS2812FX/blob/master/WS2812FX.h#L175
Remove FX functions here https://github.com/kitesurfer1404/WS2812FX/blob/master/WS2812FX.h#L272

In WS2812FX.cpp
Remove mode functions https://github.com/kitesurfer1404/WS2812FX/blob/master/WS2812FX.cpp#L183

I guess, that might be it.

popekpl, 2017-02-21:
Ok, I will try :)

popekpl, 2017-02-21:
So I remove this, but I have compilation problem: 

`Arduino:1.8.0 (Windows 10), Płytka:"Arduino Nano, ATmega328"

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX.cpp: In member function 'void WS2812FX::stop()':

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX.cpp:86:13: error: 'strip_off' was not declared in this scope

   strip_off();

             ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX.cpp: At global scope:

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX.cpp:192:26: error: no 'void WS2812FX::strip_off()' member function declared in class 'WS2812FX'

 void WS2812FX::strip_off() {

                          ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX.cpp: In member function 'void WS2812FX::mode_blink()':

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX.cpp:260:15: error: 'strip_off' was not declared in this scope

     strip_off();

               ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX.cpp: In member function 'void WS2812FX::mode_blink_rainbow()':

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX.cpp:376:15: error: 'strip_off' was not declared in this scope

     strip_off();

               ^

exit status 1
`

I attached .CPP and .H files. PLZ HELP ME .
[ISSUE .CPP .H.zip](https://github.com/kitesurfer1404/WS2812FX/files/791745/ISSUE.CPP.H.zip)

kitesurfer1404, 2017-02-22:
Do not remove strip_off from WS2812FX.cpp ;-)
Removing all mode_... is OK. But not the "utility" functions.

Also, by removing ALL mode_... functions you'll end up with something not very usefull.

popekpl, 2017-02-22:
But I didn't remove strip_off. And I want to use only 9 modes.

popekpl, 2017-02-22:
This is my .cpp file : 

`

#include "Arduino.h"
#include "WS2812FX.h"

#define CALL_MODE(n) (this->*_mode[n])();

void WS2812FX::init() {
  Adafruit_NeoPixel::begin();
  WS2812FX::setBrightness(_brightness);
  Adafruit_NeoPixel::show();
}

void WS2812FX::service() {
  if(_running || _triggered) {
    unsigned long now = millis();

    if(now - _mode_last_call_time > _mode_delay || _triggered) {
      CALL_MODE(_mode_index);
      _counter_mode_call++;
      _mode_last_call_time = now;
      _triggered = false;
    }
  }
}

void WS2812FX::start() {
  _counter_mode_call = 0;
  _counter_mode_step = 0;
  _mode_last_call_time = 0;
  _running = true;
}

void WS2812FX::stop() {
  _running = false;
  strip_off();
}

void WS2812FX::trigger() {
  _triggered = true;
}

void WS2812FX::setMode(uint8_t m) {
  _counter_mode_call = 0;
  _counter_mode_step = 0;
  _mode_last_call_time = 0;
  _mode_index = constrain(m, 0, MODE_COUNT-1);
  _mode_color = _color;
  Adafruit_NeoPixel::setBrightness(_brightness);
  //strip_off();
}

void WS2812FX::setSpeed(uint8_t s) {
  _counter_mode_call = 0;
  _counter_mode_step = 0;
  _mode_last_call_time = 0;
  _speed = constrain(s, SPEED_MIN, SPEED_MAX);
  //strip_off();
}

void WS2812FX::increaseSpeed(uint8_t s) {
  s = constrain(_speed + s, SPEED_MIN, SPEED_MAX);
  setSpeed(s);
}

void WS2812FX::decreaseSpeed(uint8_t s) {
  s = constrain(_speed - s, SPEED_MIN, SPEED_MAX);
  setSpeed(s);
[... 198 lines cut ...]
    Adafruit_NeoPixel::setPixelColor(0, color_wheel(_mode_color));
  }

  Adafruit_NeoPixel::show();

  _counter_mode_step = (_counter_mode_step + 1) % 2;

  _mode_delay = 50 + ((50 * (uint32_t)(SPEED_MAX - _speed)) / _led_count);
}

void WS2812FX::mode_merry_christmas(void) {
  for(uint16_t i=0; i < _led_count; i++) {
    if((i + _counter_mode_step) % 4 < 2) {
      Adafruit_NeoPixel::setPixelColor(i, 255, 0, 0);
    } else {
      Adafruit_NeoPixel::setPixelColor(i, 0, 255, 0);
    }
  }
  Adafruit_NeoPixel::show();

  _counter_mode_step = (_counter_mode_step + 1) % 4;
  _mode_delay = 100 + ((100 * (uint32_t)(SPEED_MAX - _speed)) / _led_count);
}

`

kitesurfer1404, 2017-02-22:
Oh, my fault. You removed it from the .h file in line 271.
https://github.com/kitesurfer1404/WS2812FX/blob/master/WS2812FX.h#L271

popekpl, 2017-02-22:
Ohh, ok yes. And now I have new compilation errors ;P . 

: 

Arduino:1.8.0 (Windows 10), Płytka:"Arduino Nano, ATmega328"

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:308:3: error: expected unqualified-id before 'if'

   if(_counter_mode_call == 0) {

   ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:312:31: error: '_mode_color' was not declared in this scope

   uint8_t breath_brightness = _mode_color; // we use _mode_color to store the brightness

                               ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:314:3: error: expected unqualified-id before 'if'

   if(_counter_mode_step < 8) {

   ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:316:5: error: expected unqualified-id before 'else'

   } else {

     ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:321:3: error: expected unqualified-id before 'if'

   if(breath_brightness == breath_brightness_steps[_counter_mode_step]) {

   ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:325:3: error: expected unqualified-id before 'for'

   for(uint16_t i=0; i < _led_count; i++) {

   ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:325:21: error: 'i' does not name a type

   for(uint16_t i=0; i < _led_count; i++) {

                     ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:325:37: error: 'i' does not name a type

   for(uint16_t i=0; i < _led_count; i++) {

                                     ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:328:45: error: '_brightness' was not declared in this scope

   int b = map(breath_brightness, 0, 255, 0, _brightness);  // keep brightness below brightness set by user

                                             ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:329:35: error: expected constructor, destructor, or type conversion before '(' token

   Adafruit_NeoPixel::setBrightness(b);                     // set new brightness to leds

                                   ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:330:28: error: expected constructor, destructor, or type conversion before ';' token

   Adafruit_NeoPixel::show();

                            ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:332:3: error: '_mode_color' does not name a type

   _mode_color = breath_brightness;                         // we use _mode_color to store the brightness

   ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:333:3: error: '_mode_delay' does not name a type

   _mode_delay = breath_delay_steps[_counter_mode_step];

   ^

C:\Program Files (x86)\Arduino\libraries\WS2812FX\WS2812FX1.cpp:334:1: error: expected declaration before '}' token

 }

 ^

exit status 1

I atached my new files.
[ISSUE .CPP .H.zip](https://github.com/kitesurfer1404/WS2812FX/files/794320/ISSUE.CPP.H.zip)

kitesurfer1404, 2017-02-22:
I can't reproduce the error. Compiles for me.

popekpl, 2017-02-22:
Ok, sorry :P .Now I'm going to check if it works.

popekpl, 2017-02-22:
Yes it works. THANK YOU 👍 <3

kitesurfer1404, 2017-02-22:
Cool. Have fun. :)

closed 2017-02-22 by kitesurfer1404