kitesurfer1404/WS2812FX #20: handling patterns in loop section

Nikoskir, 2017-03-13:
Hello my name is Niko
first of all thanks for sharing these information with all of us.
My request is about handling patterns in a way that can be called when a message is received.
below is an implementation that i want to achieve with nrf24.

receiver side:

#include "nRF24L01.h"
#include "RF24.h"
#include "SPI.h"
int ReceivedMessage[1] = {000};
RF24 radio(9,10);
const uint64_t pipe = 0xE6E6E6E6E6E6;

void setup(void){
radio.begin();
radio.openReadingPipe(1,pipe);
radio.startListening();
}

void loop(){

while (radio.available())
{
radio.read(ReceivedMessage, 1);

if (ReceivedMessage[0] == 1)
{do something}
if (ReceivedMessage[0] == 2)
{do something}
if (ReceivedMessage[0] == 3)
{do something}
if (ReceivedMessage[0] == 4)
{do something}

}
(run the pattern);
}

the receiver node will get these messages every 5 seconds.
So every pattern will be active for 5 seconds with its own parameters(speed and color)
in this way i can put 30 nodes which will perfectly synchronized.
If i put all nodes with one start message after a while they will not be together because of timing difference of each arduino.
And of course all these patterns need to be delay free as your library does!

thanks for reading this.

Best regards 

Niko

kitesurfer1404, 2017-04-25:
Hi, sorry for this delay. I had no time to code lately.
I think a look at the trigger-method might help. I implemented this to step through the effects by external trigger signals (microphone). But it can be used to implement some kind of synchronization, too, I guess.
Example use of trigger: https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/external_trigger/external_trigger.ino

I think you might want to implement some kind of step-counter on each arduino, run the effects with the trigger-method, count each step. Over radio you can send a reference step-count. Each arduino might then slow down/delay some steps of speed up/do steps without delay to catch up. Rough guess.

I'll run into this problem at the end of this year, I guess, when doing some synced ESPs for christmas lighting.

Keep me posted, if this helps. Thanks!

Nikoskir, 2017-05-05:
Hi,
thanks for the reply
i've managed to make it work with this code below
works like a charm...
the changes are the most every 10 seconds
playing around with speed it looks very nice
perfectly synchronized  except the random ones which of course cannot.
at the weekend i will finish this sketch with all the possible combinations.
i use 20 of them strips with nrf modules.
if you can advice a change inside this sketch it will be much appreciated.
i will also check the idea you posted before and see how it will work.

Thanks 
Niko

#include <WS2812FX.h>
#include "nRF24L01.h"
#include "RF24.h"
#include "SPI.h"
int ReceivedMessage[1] = {000};
RF24 radio(9, 10);
const uint64_t pipe = 0xE6E6E6E6E6E6;
WS2812FX ws2812fx = WS2812FX(30, 8, NEO_GRB + NEO_KHZ800);

void setup() {
  radio.begin();
  radio.setPALevel(RF24_PA_MIN);
  radio.openReadingPipe(1, pipe);
  radio.startListening();
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setSpeed(245);
  ws2812fx.setColor(0, 255, 0);
  ws2812fx.start();

}

void loop() {
  while (radio.available())
  {
    radio.read(ReceivedMessage, 1);

    if (ReceivedMessage[0] == 1)
    {
      ws2812fx.setSpeed(255);
      ws2812fx.setMode(FX_MODE_RUNNING_RED_BLUE);
    }
    if (ReceivedMessage[0] == 2)
    {
      ws2812fx.setSpeed(245);
      ws2812fx.setMode(FX_MODE_COLOR_WIPE_RANDOM);
    }
    if (ReceivedMessage[0] == 3)
    {
      ws2812fx.setSpeed(220);
      ws2812fx.setMode(FX_MODE_COMET);
    }
    if (ReceivedMessage[0] == 4)
    {
      ws2812fx.setMode(FX_MODE_DUAL_SCAN);
    }
    if (ReceivedMessage[0] == 5)
    {
      ws2812fx.setSpeed(255);
      ws2812fx.setMode(FX_MODE_THEATER_CHASE_RAINBOW);
    }
    if (ReceivedMessage[0] == 6)
    {ws2812fx.setSpeed(230);
      ws2812fx.setMode(FX_MODE_RUNNING_LIGHTS);
    }
    if (ReceivedMessage[0] == 7)
    { ws2812fx.setColor(255, 0, 0);
      ws2812fx.setMode(FX_MODE_RUNNING_LIGHTS);
    }
    if (ReceivedMessage[0] == 8)
    { ws2812fx.setColor(0, 0, 255);
      ws2812fx.setMode(FX_MODE_RUNNING_LIGHTS);
    }
    if (ReceivedMessage[0] == 9)
    {ws2812fx.setSpeed(245);
      ws2812fx.setMode(FX_MODE_COLOR_SWEEP_RANDOM);
    }
    if (ReceivedMessage[0] == 10)
    {
      ws2812fx.setMode(FX_MODE_RUNNING_RED_BLUE);
    }
    if (ReceivedMessage[0] == 11)
    {
      ws2812fx.setMode(FX_MODE_FIREWORKS_RANDOM);
    }
    if (ReceivedMessage[0] == 12)
    {
      ws2812fx.setMode(FX_MODE_COMET);
    }
    if (ReceivedMessage[0] == 13)
    {
      ws2812fx.setMode(FX_MODE_RAINBOW_CYCLE);
    }
    if (ReceivedMessage[0] == 14)
    {
      ws2812fx.setMode(FX_MODE_COLOR_WIPE_RANDOM);
    }
    if (ReceivedMessage[0] == 15)
    {
      ws2812fx.setMode(FX_MODE_COMET);
    }
}
  ws2812fx.service();
}