kitesurfer1404/WS2812FX #21: Max number of LEDs?

jeremykoerber, 2017-05-02:
Hi! First off, love this library! Just one problem I'm finding. I'm trying to control 288 LEDs, and this library seems to max out at 235. I have 2 x 1 meter strips with 144 LEDs each chained in my front window. Unfortunately, I'm not getting any use out of those last 53 lights. Other libraries like FastLED and Adafruit Neopixel seem to work just fine, but I really love the FX in this library. The strips are WS2812B running off an Arduino Uno R3. Should it be possible to control all 288? If so, any ideas on what I might try to get this library to work with all my lights? If not, just curious why and if there's a chance the limit could be raised. Thank you very much for the hard work on this!

Fireflaker, 2017-05-09:
Same question. The limitation is probably on the ram of the AtMega chip. Using ESP8266 might help although I am not sure. 
How many LEDs can it handle with a ESP8266? Does it run properly on ESP32?

idolpx, 2017-07-26:
It is a memory limit issue on the microcontroller you are using.
Each pixel needs about 3 bytes of memory and your program uses memory as well.

The Arduino Uno R3 uses and ATMega328P which has 2KB of memory.
Your program uses that memory too.  What every is left over is how much you have available for controlling your LED strip.  

If you optimize your code you may be able to squeeze out a few more pixels.

kitesurfer1404, 2017-09-06:
I'll close this one. But please re-check. I just merged the pull request shifting all the FX names to flash instead of keeping them in RAM. So there might be additional RAM free for your LEDs now.

closed 2017-09-06 by kitesurfer1404