kitesurfer1404/WS2812FX #27: Add more realistic fire effect

grote, 2017-06-11:
The current two fire effects seem to be only based on the current color, but don't do any color variations that are typical for real flickering fire.

It would be nice, if another fire effect would be added which looks more like [this one](http://www.walltech.cc/neopixel-fire-effect/).

idolpx, 2017-07-26:
The code actually looks almost identical to the one you referenced.
The only difference is that it does use the selected color to rather than starting with a yellow like the other.  If you set the color to 255, 215, 40 and call mode_fire_flicker_int(150) you should get a similar effect.

grote, 2017-07-26:
Thanks, so we can probably close this one.

idolpx, 2017-07-26:
I added this as "Fire Flicker (intense)" in my fork.
Still need to call setColor(255, 69,3) and setSpeed(70) before changing to this mode to get the fiery colors of the other code.

closed 2017-09-06 by kitesurfer1404