kitesurfer1404/WS2812FX #34: Does set color have to be called at a specific point?

mcdonnelldean, 2017-08-06:
Using this code,

```c
#include <WS2812FX.h>

#define LED_COUNT 4
#define LED_PIN 2

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setColor(255, 69, 3);
  ws2812fx.setBrightness(100);
  ws2812fx.setSpeed(70);
  ws2812fx.setMode(46);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}
```

The lights seem to stay default green although they do flash briefly for a second. Am I doing this wrong to change the color?

kitesurfer1404, 2017-09-06:
I guess that's some kind of issue with FX_MODE_FIRE_FLICKER_SOFT when running at low brightness. I had this kind of behavior too, when trying to change colors. Please try with higher brightness and/or different colors.

mcdonnelldean, 2017-09-07:
Actually (I thought I responded before and sorry for not) It turns out my LEDs where RGB over GRB so the colors were reversed so what I thought I was setting as red ended up being green! Which makes perfect sense. 

I will go ahead and close, for others if you have the same you need to swap `NEO_GRB` with `NEO_RGB` that will fix it (usually older models have RGB)

mcdonnelldean, 2017-09-07:
@kitesurfer1404 On the green tinge at lower brightness, ya, I think that's just the limitation of the pixel, I get green bleed at low brightness for any flicker that is red based so I'm ok with that by increasing the brightness and offsetting the green a little. So all good.

closed 2017-09-07 by mcdonnelldean

kitesurfer1404, 2017-09-07:
I know the RGB/GRB issue... I love the 12mm WS2811 Pixels, they are RGB. All newer strips are GRB. I always get that wrong. :-1