kitesurfer1404/WS2812FX #37: Adapt for WS2801

r41d, 2017-09-01:
I own some WS2801 and would like to control them using this awesome library.
What would be the preferred method?

1. Including support in WS2812FX with a bunch of #ifdef (well, this would obviously turn out ugly)
2. Including support in WS2812FX by abstracting the Adafruit WS2801/Neopixel library calls
3. Just port WS2812FX from `Adafruit_NeoPixel` to `Adafruit-WS2801-Library` in a fork (then being named WS2801FX)

Would the maintainer accept a PR that implemented 2. ?
If yes, I would work on this, if not, I would attempt 3.

r41d, 2017-09-05:
Since I was in need of this project for WS2801, I decided to port it.
Here is the result: https://github.com/r41d/WS2801FX
Remarks appreciated.

kitesurfer1404, 2017-09-06:
Cool to see this being ported. I was quite busy with my photobooth project (new code coming soon). Now the time has come to start the christmas light projects, so I'll work on this lib again.

I have no WS2801 here. But I guess I'll have to look into all different kinds of "smart" LEDs soon to get this working with most of them. Next might be UCS1904, because there are some LEDs on their way from china to my door.

Porting to FastLED was suggested before. I have no experience with that lib, only with the Adafruit one. But I might switch this lib over.

Need to review all the nice pull requests and issues, soon. So many great ideas in there.

closed 2018-05-24 by moose4lord