kitesurfer1404/WS2812FX #38: New to programing

oboeman123, 2017-09-06:
I am working on a project that uses a push button to cycle through different lighting sequences.  My question is how can I take and select only the functions I want?  I would like to then modify colors, timings, etc.  I am sorry for the lack of knowledge, but any help would be very much appreciated.

kitesurfer1404, 2017-09-06:
I did this some days ago in a hurry for another project. The idea is to store all wanted FX-numbers in an array and increment the index of this array to the current mode and then update the WS2812FX.
Switches are connected to PINs 2 to 5 and switch them to GND. All button-states are read to one byte. Therefore the bit-state is checked in the loop(). Not pretty but working.

There is also a button for speed, color (off color_wheel) and brightness-steps.

Let me know if this helps.

kitesurfer1404, 2017-09-06:
`#include <WS2812FX.h>`
` `
`#define LED_COUNT 30`
`#define LED_PIN 13`
` `
`#define DEFAULT_MODE_INDEX 13`
`#define DEFAULT_COLOR_INDEX 20  // of color_wheel`
`#define DEFAULT_SPEED 240`
`#define DEFAULT_BRIGHTNESS 50`
` `
`#define PIN_SW_BRIGHTNESS 2`
`#define PIN_SW_SPEED 3`
`#define PIN_SW_COLOR 4`
`#define PIN_SW_MODE 5`
` `
`#define SW_DEBOUNCE_TIME 75`
` `
`uint8_t modes[] = {`
`  FX_MODE_STATIC,`
`  FX_MODE_BLINK,`
`  FX_MODE_BREATH,`
`  FX_MODE_COLOR_WIPE,`
`  FX_MODE_COLOR_WIPE_RANDOM,`
`  FX_MODE_RANDOM_COLOR,`
`  FX_MODE_SINGLE_DYNAMIC,`
`  FX_MODE_MULTI_DYNAMIC,`
`  FX_MODE_RAINBOW,`
`  FX_MODE_RAINBOW_CYCLE,`
`  FX_MODE_SCAN,`
`  FX_MODE_DUAL_SCAN,`
`  FX_MODE_FADE,`
`  FX_MODE_RUNNING_LIGHTS,`
`  FX_MODE_THEATER_CHASE,`
`  FX_MODE_THEATER_CHASE_RAINBOW,`
`  FX_MODE_TWINKLE,`
`  FX_MODE_TWINKLE_RANDOM,`
`  FX_MODE_TWINKLE_FADE,`
`  FX_MODE_TWINKLE_FADE_RANDOM,`
`  FX_MODE_SPARKLE,`
`  FX_MODE_FLASH_SPARKLE,`
`  FX_MODE_HYPER_SPARKLE,`
`  FX_MODE_BLINK_RAINBOW,`
`  FX_MODE_CHASE_COLOR,`
`  FX_MODE_CHASE_RANDOM,`
`  FX_MODE_CHASE_RAINBOW,`
`  FX_MODE_CHASE_BLACKOUT,`
`  FX_MODE_CHASE_BLACKOUT_RAINBOW,`
`  FX_MODE_COLOR_SWEEP_RANDOM,`
`  FX_MODE_RUNNING_COLOR,`
`  FX_MODE_RUNNING_RED_BLUE,`
`  FX_MODE_RUNNING_RANDOM,`
`  FX_MODE_LARSON_SCANNER,`
`  FX_MODE_COMET,`
`  FX_MODE_FIREWORKS,`
`  FX_MODE_FIREWORKS_RANDOM,`
`  FX_MODE_MERRY_CHRISTMAS  `
`};`
` `
` `
` `
`WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_RGB + NEO_KHZ800);`
` `
`unsigned long now = 0;`
` `
`uint8_t mode_index = DEFAULT_MODE_INDEX;`
`uint8_t color_index = DEFAULT_COLOR_INDEX;`
` `
`uint8_t button_state = 0;`
`uint8_t last_button_state = 0;`
`unsigned long last_pushed_button_time = 0;`
` `
`void setup() {`
`  pinMode(PIN_SW_BRIGHTNESS, INPUT_PULLUP);`
`  pinMode(PIN_SW_SPEED, INPUT_PULLUP);`
`  pinMode(PIN_SW_COLOR, INPUT_PULLUP);`
`  pinMode(PIN_SW_MODE, INPUT_PULLUP);`
`  `
`  ws2812fx.init();`
`  ws2812fx.setBrightness(DEFAULT_BRIGHTNESS);`
`  ws2812fx.setSpeed(DEFAULT_SPEED);`
`  ws2812fx.setColor(color_wheel(color_index));`
`  ws2812fx.setMode(modes[mode_index]);`
`  ws2812fx.start();`
`}`
` `
`void loop() {`
`  now = millis();`
`  ws2812fx.service();`
` `
`  button_state = digitalRead(PIN_SW_BRIGHTNESS) << 3 | digitalRead(PIN_SW_SPEED) << 2 | digitalRead(PIN_SW_COLOR) << 1 | digitalRead(PIN_SW_MODE);`
`  `
`  if(button_state < 15 && last_button_state == 15 && (now - last_pushed_button_time > SW_DEBOUNCE_TIME)) {`
`    if(button_state == 14) {`
`      mode_index = (mode_index + 1) % (sizeof(modes) / sizeof(uint8_t));`
`      ws2812fx.setMode(modes[mode_index]);`
`    }`
` `
`    if(button_state == 13) {`
`      color_index = 5 + ((color_index + 10) % 255);`
`      ws2812fx.setColor(color_wheel(color_index));`
`    }`
` `
`    if(button_state == 11) {`
`      ws2812fx.setSpeed(max(5, (ws2812fx.getSpeed() + 50) % 255));`
`    }`
` `
`    if(button_state == 7) {`
`      ws2812fx.setBrightness(max(10, (ws2812fx.getBrightness() + 50) % 255));`
`    }`
`  }`
` `
`  if(button_state < 15) {`
`    last_pushed_button_time = now;`
`  }`
` `
`  last_button_state = button_state;`
`}`
` `
` `
`uint32_t color_wheel(uint8_t pos) {`
`  pos = 255 - pos;`
`  if(pos < 85) {`
`    return ((uint32_t)(255 - pos * 3) << 16) | ((uint32_t)(0) << 8) | (pos * 3);`
`  } else if(pos < 170) {`
`    pos -= 85;`
`    return ((uint32_t)(0) << 16) | ((uint32_t)(pos * 3) << 8) | (255 - pos * 3);`
`  } else {`
`    pos -= 170;`
`    return ((uint32_t)(pos * 3) << 16) | ((uint32_t)(255 - pos * 3) << 8) | (0);`
`  }`
`}`

oboeman123, 2017-09-06:
Thanks for your help.

closed 2018-05-24 by moose4lord