kitesurfer1404/WS2812FX #39: New effects

Vika1283, 2017-09-06:
You have a very good project, I'm really looking forward to new effects

jllangley, 2017-09-27:
I created this new effect I call it Christmas 2, it is based off of the circus combustus effect.  Very good work thanks for sharing your code Kitsurfer1404.

/*
 * Alternating red/green/blue pixels running.
 */
void WS2812FX::mode_merry_christmas_2(void) {
  for(uint16_t i=0; i < _led_count; i++) {
    if((i + _counter_mode_step) % 9 < 3) {
      Adafruit_NeoPixel::setPixelColor(i, 255, 0, 0);
    } else if ((i + _counter_mode_step) % 9 < 6){
      Adafruit_NeoPixel::setPixelColor(i, 0, 255, 0);
    } else {
      Adafruit_NeoPixel::setPixelColor(i, 0, 169, 255);

    }
  }

pjmsullivan, 2017-12-23:
@kitesurfer1404  - awesome project!  I just set up an LED light strand to surprise my family over Christmas.  I have some code from another project that uses the fastled library that makes really cool random ambient effects.  I'm not a coder, but take a look, and if you like these effects, maybe you could consider adding them to your library.
`        EVERY_N_MILLISECONDS(10) { //like a delay(10)?
                nblendPaletteTowardPalette(currentPalette, targetPalette, maxChanges); // FOR NOISE ANIMATION

                { gHue++; }

                if (setEffect == "Noise") {
                        setPower = "ON";
                        for(int i = 0; i < NUM_LEDS; i++) { // Just ONE loop to fill up the LED array as all of the pixels change.
                                uint8_t index = inoise8(i*scale, dist+i*scale) % 255; // Get a value from the noise function. I'm using both x and y axis.
                                leds[i] = ColorFromPalette(currentPalette, index, 255, LINEARBLEND); // With that value, look up the 8 bit colour palette value and assign it to the current LED.
                        }
                        dist += beatsin8(10,1, 4); // Moving along the distance (that random number we started out with). Vary it a bit with a sine wave.
                } // Noise

                if (setEffect == "Ripple") {
                        for (int i = 0; i < NUM_LEDS; i++) leds[i] = CHSV(bgcol++, 255, 15); // Rotate background colour.
                        switch (step) {
                        case -1:                                            // Initialize ripple variables.
                                center = random(NUM_LEDS);
                                colour = random8();
                                step = 0;
                                break;
                        case 0:
                                leds[center] = CHSV(colour, 255, 255);      // Display the first pixel of the ripple.
                                step++;
                                break;
                        case maxsteps:                                      // At the end of the ripples.
                                step = -1;
                                break;
                        default:                                               // Middle of the ripples.
                                leds[(center + step + NUM_LEDS) % NUM_LEDS] += CHSV(colour, 255, myfade / step * 2); // Simple wrap from Marc Miller
                                leds[(center - step + NUM_LEDS) % NUM_LEDS] += CHSV(colour, 255, myfade / step * 2);
                                step++;                                      // Next step.
                                break;
                        }
                }
        }`

pjmsullivan, 2018-05-24:
@moose4lord - I see the FastLED example.  I can't wait to try it out.  Thanks.