kitesurfer1404/WS2812FX #40: idea - grid effects!

rhamblen, 2017-09-06:
Hi I was looking at the 3LED strings siuch as 
[ebay 100 x 3 led string ](http://www.ebay.co.uk/itm/20-200-Pcs-3-LED-5050-SMD-RGB-Module-Waterproof-Light-Lamp-Flexible-Strip-DC-12V/372049885325?_trkparms=aid%3D555018%26algo%3DPL.SIM%26ao%3D1%26asc%3D20140221143405%26meid%3D900978b54c324439b3a46b03ed10c731%26pid%3D100010%26rk%3D5%26rkt%3D6%26sd%3D401388283845&_trksid=p2047675.c100010.m2109)

a hundred of these would make a nice array around .7m x 1m with 15 x 20 pixels.  given you can by other volumes of these you can get a smaller or larger array.
equally you can cut up and join a 5M strip and then put end connecters to make an array.

So the interesting thing is could we run great effects on an array of lights rather than a single strip of lights? or even get a way to program a sort of moving image.  (bit like the 80s with old computers and pixel graphics!)

Just a thought!
CLose this if not practical.

kitesurfer1404, 2017-09-07:
Nice idea. But these 3 LED modules are not WS2812, as far as I can tell. So you can not control every single pixel. Just R/G/B for all LEDs by doing PWM on the power.
Doing a matrix with WS2812 strips is possible and has been done before. Just search for "WS2812 matrix strip" on youtube. 
https://www.youtube.com/results?search_query=WS2812+matrix+strip

I'll close this issue. Not because it's not practical. But because video/matrix animation is not in the focus of this lib. I think it needs to be done. Would love to see that by myself. But it would clutter the WS2812FX code when doing matrix stuff.

Please feel free to add any information you can find on WS2812 strip-matrix projects and code here for reference. Maybe some programmers compile it all down to one new lib for that.

closed 2017-09-07 by kitesurfer1404

rhamblen, 2017-09-10:
The whole grid/matrix question is answered by using two things...
1. Glediator from solder labs www.solderlab.de 
there is a download for a arduino controller (some people recommend using a Teensy 3.x for speed)
2. Jinx software http://www.live-leds.de/
this is another controller software but you need to set it to use the arduino controller and it can use glediator protocol.  So you end up back at the glediator site again.

In both cases the arduino controller is connected by serial (usb) to the computer which will drive the patterns.  

You can do some pretty fantastic things and you can snake a long led string to make the matrix.