kitesurfer1404/WS2812FX #41: please also support the ESP32

micropet, 2017-09-08:
The program runs error-free on an ESP8266.
On an ESP32 different Leds flicker, or stay on.

moose4lord, 2017-10-30:
I finally got my ESP32 and I see the same flickering. It's not a problem with WS2812FX, it's a timing issue with the ESP32. See [https://forums.adafruit.com/viewtopic.php?f=47&t=117327](https://forums.adafruit.com/viewtopic.php?f=47&t=117327), which mentions a workaround if your string of LEDs is short (less then 34 LEDs). If you are trying to control a longer string of LEDs, you'll probably have to wait for the Adafruit NeoPixel library to be updated to work properly with the ESP32.

micropet, 2017-11-02:
Thank you. 
So not easy to solve.

Greetings Peter

micropet, 2018-04-19:
Yes. Thank you.

moose4lord, 2018-04-19:
@micropet what makes you say this has been fixed? I have heard rumblings on the internet that the NeoPixelBus library has a solution that kind'a works, but as far as I know the Adafruit_Neopixel library is still broken.

Dema323, 2018-06-03:
I started using some guys neopixelbus lib and works fine now

moose4lord, 2018-06-08:
OK, I dusted off my ESP32 and gave it a try. The version of NeoPixelBus released to the Arduino Library Manager (v2.2.9 which is quite old) still seems to have issues with the ESP32's interrupt processing, although I can get it to work by judiciously placing a delay(1) in my test sketch. The more up-to-date version of NeoPixelBus on GitHub works better, since it has the delay(1) trick built right into the bit bang code, as well as a call to taskENTER_CRITICAL() to disable task switching during the LED update.

WS2812FX behaves like the old version of NeoPixelBus. It updates the LEDs erratically, but I can get it to work by adding a delay(1) in my test sketch. I tried adding taskENTER_CRITICAL() to the WS2812FX bit bang code, but it did not help, so I'm not sure what's going on with the ESP32. Guess I'll be sticking with my trusty ESP8266 for a while longer. :)

moose4lord, 2018-07-19:
Archiving this conversation for now. I'm sure it will pop up again as the ESP32 software matures.

closed 2018-07-19 by moose4lord

Yurik72, 2019-02-25:
Hi All, 
Please have a look my idea
https://github.com/Yurik72/ESPHomeController/wiki/WS2812-driver-to-remove-flickering
to keep ws2812fx as a class and do not change to other library