kitesurfer1404/WS2812FX #42: Add support for animation direction, and virtual strips

fonix232, 2017-09-09:
Okay, since the title is limited, let me explain the virtual strips part.

Say you have a few various shaped "strips", "bits", "rings", etc. made of one or more WS2812 LEDs. Say you want to control them all, but from a single pin (because your project includes more and more devices that an ESP8266 or an Arduino can support).

So it would be nice to be able to define a single strip (say, 3x 24-LED rings in series, first one connected to a pin, second one connected to first one's output, et cetera), and be able to make segments out of it, animated separately. I'm thinking something similar to this:

```cpp
#include <WS2812FX.h>

#define LED_COUNT 72
#define LED_PIN 15

WS2812FX strip= WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
WS2812FXSegment segment1 = strip.segment(0 ,23);
WS2812FXSegment segment2 = strip.segment(24, 47);
WS2812FXSegment segment3 = strip.segment(48, 71)

void setup() {
  strip.init();
  strip.setBrightness(255);  // Modifying the strip object will modify all segments too
  strip.setSpeed(100);
  strip.setMode(FX_MODE_FLASH_SPARKLE);

  segment2.setMode(FX_MODE_RAINBOW_CYCLE); // setting the segment options after the strip will override those settings for said segment
  segment2.setSpeed(150);

  strip.start(); // starting the strip should go over all defined segments and start them separately
}

void loop() {
  strip.service(); // servicing the strip should go over all defined segments and service them too
}
```

And about the direction of animations. Especially on rings, alignment is a bitch. Say you want two 24-LED rings do the same animation, and attach them back to back. Well you can't, since the directions won't match. So it would be nice if separate WS2812FX objects (and, if implemented, segments too) could have a direction set (either NORMAL, which means the current direction and addressing `0..n`, and REVERSE, which means reversed direction addressing `n..0`).

kitesurfer1404, 2017-09-09:
I fully understand and I need a this in the near future for my Christmas decoration. I have three idetical windows that will get strips oround them. Running in circles etc. is a well needed improvement.
I can not promise to implement this immediately. But it's on the list.

fonix232, 2017-09-10:
Technically speaking, segmenting (at least in a properly OO language, say, C# or Java) is easy to solve - each segment is practically a WS2812FX object, but it maps only parts of the parent object's LEDs (such as, it translates e.g. 48 to 0, 49 to 1, 71 to 23, to extend my example).

Animation direction is again relatively simple, and I might give it a go in the near future.

jaekar99, 2017-09-12:
I would totally dig this.. I build ws2812 clocks..   I could break out the different segments for each of the hands and other animation..

I absolutely love this library.. It has been useful in many projects..  Your work makes mine a pleasure.

[image]
[image]
[image]

closed 2018-05-24 by moose4lord