kitesurfer1404/WS2812FX #43: using your serial and web examples on nodeMCU causes an error...

rhamblen, 2017-09-10:
I've had everything working using the basic examples you showed, but I get an error when using the serial and web examples.
When I complie using a DUO everything is fine, when I use the NodeMCU it reports an error...
section type conflict with __c
in looking into this it appears the apply() function is the issue and some definitions need to move to global and out of the function. [https://github.com/esp8266/Arduino/issues/2078](https://github.com/esp8266/Arduino/issues/2078)
Is it worth addressing the problem if you are able to identify the cause?

moose4lord, 2017-09-30:
I saw this problem as well. A workaround is presented in issue [#3369](https://github.com/esp8266/Arduino/issues/3369) regarding using PROGMEM. For esp8266_webinterface sketch, make the following changes:
Rename index.html.h to index.html.cpp and make this change at the top of the file:
```
#include <pgmspace.h>
char index_html[] PROGMEM = R"=====(
```
Similarly rename and change main.js.h:
```
#include <pgmspace.h>
char main_js[] PROGMEM = R"=====(
```
Then change the #include reference to these two files in esp8266_webinterface.ino to this:
```
extern char index_html[];
extern char main_js[]; 
```
This causes the PROGMEM global variables to be compiled in modules separate from the main ino module.

That worked for me.

k1gavrilov, 2017-10-06:
Bro, you are the WIZARD!
Thank you!