kitesurfer1404/WS2812FX #45: Platform.IO build issue

fonix232, 2017-09-12:
Working off of the latest master branch, and also the lib included in the PIO repos.

The following build error happens:

```
[Tue Sep 12 18:07:05 2017] Processing nodemcuv2 (platform: espressif8266; lib_deps: git+https://github.com/marvinroger/homie-esp8266.git#develop, git+https://github.com/kitesurfer1404/WS2812FX.git#master; board: nodemcuv2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Library Storage: [...]/170912-155002-nodemcuv2/.piolibdeps
LibraryManager: Installing homie-esp8266
Homie @ 5470b1b is already installed
Installing dependencies
Looking for ArduinoJson library in registry
Found: http://platformio.org/lib/show/64/ArduinoJson
LibraryManager: Installing id=64 @ ^5.10.0
ArduinoJson @ 5.11.1 is already installed
Looking for AsyncMqttClient library in registry
Found: http://platformio.org/lib/show/346/AsyncMqttClient
LibraryManager: Installing id=346 @ ^0.8.0
AsyncMqttClient @ 0.8.1 is already installed
Installing dependencies
Looking for ESPAsyncTCP library in registry
Found: http://platformio.org/lib/show/305/ESPAsyncTCP
LibraryManager: Installing id=305 @ ^1.0.1
ESPAsyncTCP @ 1.1.0 is already installed
Looking for Bounce2 library in registry
Found: http://platformio.org/lib/show/1106/Bounce2
LibraryManager: Installing id=1106 @ ^2.1.0
Bounce2 @ 2.1 is already installed
LibraryManager: Installing WS2812FX
WS2812FX @ 923123e is already installed
Installing dependencies
Looking for Adafruit NeoPixel library in registry
Found: http://platformio.org/lib/show/28/Adafruit NeoPixel
LibraryManager: Installing id=28 @ ~1.0.6
Adafruit NeoPixel @ 1.0.7 is already installed
Collected 46 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Homie> v2.0.0 ([...]/170912-155002-nodemcuv2/.piolibdeps/Homie)
|   |-- <ArduinoJson> v5.11.1 ([...]/170912-155002-nodemcuv2/.piolibdeps/ArduinoJson_ID64)
|   |-- <AsyncMqttClient> v0.8.1 ([...]/170912-155002-nodemcuv2/.piolibdeps/AsyncMqttClient_ID346)
|   |   |-- <ESPAsyncTCP> v1.1.0 ([...]/170912-155002-nodemcuv2/.piolibdeps/ESPAsyncTCP_ID305)
|   |   |   |-- <ESP8266WiFi> v1.0 ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi)
|   |-- <Bounce2> v2.1 ([...]/170912-155002-nodemcuv2/.piolibdeps/Bounce2_ID1106)
|   |-- <ESP8266WebServer> v1.0 ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer)
|   |   |-- <ESP8266WiFi> v1.0 ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi)
|   |-- <ESP8266HTTPClient> v1.1 ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266HTTPClient)
|   |   |-- <ESP8266WiFi> v1.0 ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi)
|   |-- <ESP8266mDNS> ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS)
|   |   |-- <ESP8266WiFi> v1.0 ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi)
|   |-- <DNSServer> v1.1.0 ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/DNSServer)
|   |   |-- <ESP8266WiFi> v1.0 ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi)
|   |-- <Ticker> v1.0 ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/Ticker)
|   |-- <ESP8266WiFi> v1.0 ([...]/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi)
|-- <WS2812FX> v1.0.0 ([...]/170912-155002-nodemcuv2/.piolibdeps/WS2812FX)
|   |-- <Adafruit NeoPixel> v1.0.7 ([...]/170912-155002-nodemcuv2/.piolibdeps/Adafruit NeoPixel_ID28)
xtensa-lx106-elf-g++ -o .pioenvs/nodemcuv2/lib/WS2812FX/WS2812FX.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=30500 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -U__STRICT_ANSI__ -I[...]/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include -I[...]/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include -I[...]/.platformio/packages/framework-arduinoespressif8266/cores/esp8266 -I[...]/.platformio/packages/framework-arduinoespressif8266/variants/nodemcu "-I.piolibdeps/Adafruit NeoPixel_ID28" -I.piolibdeps/WS2812FX .piolibdeps/WS2812FX/WS2812FX.cpp
In file included from .piolibdeps/WS2812FX/WS2812FX.h:41:0,
from .piolibdeps/WS2812FX/WS2812FX.cpp:54:
.piolibdeps/Adafruit NeoPixel_ID28/Adafruit_NeoPixel.h: In member function 'void WS2812FX::setLength(uint16_t)':
.piolibdeps/Adafruit NeoPixel_ID28/Adafruit_NeoPixel.h:159:5: error: 'uint16_t Adafruit_NeoPixel::numLEDs' is private
numLEDs,       // Number of RGB LEDs in strip
^
.piolibdeps/WS2812FX/WS2812FX.cpp:158:31: error: within this context
} while(!Adafruit_NeoPixel::numLEDs && b > 1);
^
In file included from .piolibdeps/WS2812FX/WS2812FX.h:41:0,
from .piolibdeps/WS2812FX/WS2812FX.cpp:54:
.piolibdeps/Adafruit NeoPixel_ID28/Adafruit_NeoPixel.h:159:5: error: 'uint16_t Adafruit_NeoPixel::numLEDs' is private
numLEDs,       // Number of RGB LEDs in strip
^
.piolibdeps/WS2812FX/WS2812FX.cpp:160:35: error: within this context
_led_count = Adafruit_NeoPixel::numLEDs;
^
*** [.pioenvs/nodemcuv2/lib/WS2812FX/WS2812FX.o] Error 1
============================================== [ERROR] Took 2.22 seconds ==============================================
```

Had to replace my path with `[...]`, since it includes work-related information falling under an NDA. It replaces both my `~` folder (in case of `/.platformio/` paths) and the path to my project (in case of `/170912-155002-nodemcuv2/` paths).

Apparently there's some incompatibility with the version of `Adafruit_Neopixel` pulled in by PIO itself.

fonix232, 2017-09-15:
@kitesurfer1404 if you need any specific help, I can give a hand in debugging and possibly fixing it.

euphi, 2017-09-22:
`numLEDs` is private in older versions of the Adafruit library, so the build fails. To solve it, just use the recent version :-)

PR #47 (2017-11-01): Use latest Adafruit NeoPixel library on pio (Fixes #45)
Updated library.json (for platformio) to use the latest Adafruit NeoPixel library. This fixes #45 (and possible some other problems)
commit 069139b9 (2017-11-01): Use latest Adafruit NeoPixel library on pio (Fixes #45)

Updated library.json (for platformio) to use the latest Adafruit NeoPixel library. This fixes #45 (and possible some other problems)
commit a765adf5 (2017-11-01): Merge pull request #47 from euphi/patch-2

Use latest Adafruit NeoPixel library on pio (Fixes #45)
closed 2017-11-01 by kitesurfer1404