kitesurfer1404/WS2812FX #49: main.js.h:1: error: main_js causes a section type conflict with __c

Pixelplanet, 2017-10-12:
Im trying to use this library on an ESP-01 but when trying to flash the webinterface example i get the following error, any idea whats going on?

also tried going back to a older version of the esp8266 arduino core thingy but didnt make a difference.

> Arduino: 1.8.1 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"
> 
> In file included from C:\Users\Tom\AppData\Local\Temp\arduino_modified_sketch_658435\esp8266_webinterface.ino:46:0:
> 
> main.js.h:1: error: main_js causes a section type conflict with __c
> 
>  const char main_js[] PROGMEM = R"=====(
> 
>             ^
> 
> In file included from C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/Arduino.h:243:0,
> 
>                  from C:\Users\Tom\AppData\Local\Temp\arduino_build_937800\sketch\esp8266_webinterface.ino.cpp:1:
> 
> C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/pgmspace.h:21:51: note: '__c' was declared here
> 
>  #define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
> 
>                                                    ^
> 
> C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/WString.h:38:76: note: in definition of macro 'FPSTR'
> 
>  #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
> 
>                                                                             ^
> 
> C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/WString.h:39:34: note: in expansion of macro 'PSTR'
> 
>  #define F(string_literal) (FPSTR(PSTR(string_literal)))
> 
>                                   ^
> 
> C:\Program Files (x86)\Arduino\libraries\WS2812FX-master/WS2812FX.h:227:50: note: in expansion of macro 'F'
> 
>        _name[FX_MODE_HALLOWEEN]                 = F("Halloween");
> 
>                                                   ^
> 
> Bibliothek ESP8266WiFi in Version 1.0 im Ordner: C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi  wird verwendet
> Bibliothek ESP8266WebServer in Version 1.0 im Ordner: C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WebServer  wird verwendet
> Bibliothek WS2812FX-master im Ordner: C:\Program Files (x86)\Arduino\libraries\WS2812FX-master (legacy) wird verwendet
> Bibliothek Adafruit_NeoPixel in Version 1.1.2 im Ordner: C:\Users\Tom\Documents\Arduino\libraries\Adafruit_NeoPixel  wird verwendet
> exit status 1
> main_js causes a section type conflict with __c

RudyFiero, 2017-10-22:
I have the same error.

M-Nau, 2017-10-28:
Fixed it for me:

Simply remove PROGMEM in the first lines of the main.js.h and in the HTML.h

the control is now working great for me, i am using a cheap chinese esp-1

dave-mueller, 2017-10-28:
M-Nau's fix worked for me as well. Thank you :)

RudyFiero, 2017-10-29:
All good now.

moose4lord, 2017-10-30:
If you want a solution that keeps the HTML and JavaScript strings in Flash memory (PROGMEM), see my solution mentioned in  #43. The esp8266 has quite a bit of RAM, so removing the PROGMEM from the webinterface demo program works fine. But if you create a lot of HTML or JavaScript, eventually you'll run out of RAM and you'll want to put the strings in Flash memory.

closed 2018-05-24 by moose4lord