kitesurfer1404/WS2812FX #50: Newer commits made code size blow

jrbenito, 2017-11-06:
Hi,

I had a arduino ATMega328P code compiling with this library (SHA 069139b9340e14cd46d8fafe14972b6c056273ec). Code size was 29948 bytes.  If I compile same code with latest, code size blows to 38694 bytes. 

The newer commits added about 8KB of code, this represents 1/3 of available code memory for an regular Arduino.

kitesurfer1404, 2017-11-09:
Good point, thanks. Please use the pinned version for now. I'm aware of this, too, because I ran into the same problem right after you opened this issue.
I think, I'll re-focus on the main purpose of this lib: collection of effects. There were nice features added, no question. But at the moment I can not maintain more than the basic lib. So I might withdraw most of them and put them into a special/different project or examples. Have to figure out a good strategy on that.

jrbenito, 2017-11-09:
Maybe some kind of plug in.

Thanks

moose4lord, 2017-11-18:
I have forked the project and made some changes to reduce the memory footprint. In the meantime, you can disabled some of the animations by commenting them out in the **ws2812fx.h** file like so:

```
      _name[FX_MODE_THEATER_CHASE_RAINBOW]     = F("Theater Chase Rainbow");
//    _name[FX_MODE_RUNNING_LIGHTS]            = F("Running Lights");
      _name[FX_MODE_TWINKLE]                   = F("Twinkle");
```

Note the double slashes to comment out the 'Running Lights' animation mode. After commenting it out, the linker will NOT include that mode in the compiled executable and save you about 1750 bytes. Here's some other animations modes that are 'expensive' memory-wise:
FX_MODE_RUNNING_LIGHTS (1754 bytes)
FX_MODE_FIREWORKS (1144 bytes)
FX_MODE_TWINKLE (868 bytes)
FX_MODE_BREATH (656 bytes)

Of course you won't be able to use the commented out animations in your application, but if you don't use them anyway, comment them out and save yourself some space.

closed 2018-05-24 by moose4lord