kitesurfer1404/WS2812FX #53: Problem with WS2812Fx library on Nodemcu 1.0 (ESP-12E)

benska13, 2017-11-18:
Look like that the values are inverted, when i write:
...
ws2812fx.setMode(255, 255, 255); 
...
the led is black. And if i set one value to 0, that color light up.

What do i wrong?

moose4lord, 2017-11-18:
The mode parameter sets the animation mode, not the color. You want something like:
```
ws2812fx.setColor(255, 255, 255);
ws2812fx.setMode(FX_MODE_COLOR_WIPE);
```

See the **ws2812fx.h** file for the complete list of animation modes.

moose4lord, 2017-11-19:
This simple sketch produces a white wipe animation for me using an ESP8266 processor and a strip of 30 2812 LEDs.
```
#include <WS2812FX.h>

#define LED_PIN   5
#define LED_COUNT 30

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setColor(255, 255, 255);
  ws2812fx.setMode(FX_MODE_COLOR_WIPE);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}
```
Give that a try. 

If it still doesn't work, what microprocessor and LED are you using? Tell me about your setup.

In terms of general debugging, make sure you have an adequate capacitor (470uF or larger) wired across the power and ground leads of the LED (as close to the LED as possible). You may also need a small (<300 ohm) resistor in series with the data lead to reduce noise,

See the troubleshooting guide here:
[https://learn.adafruit.com/adafruit-neopixel-uberguide/best-practices](https://learn.adafruit.com/adafruit-neopixel-uberguide/best-practices)

benska13, 2017-11-19:
Thanks for help.
I ment `setColor();`  in the first place.

I have a Nodemcu 1.0 (ESP-12E), led stip WS2812B and a 3,3 to 5v converter.
I use an external power supply to the led strip.

I also have a Arduino Pro Mini  5 V that i use to compare.

When I upload this to the Pro Mini, the strip gest blue.

`#include <WS2812FX.h>

#define LED_COUNT 30
#define LED_PIN 5

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(200);
  ws2812fx.setColor(0, 0, 255);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}`

If i upload the same to the Nodemcu I have to use the GPI05 pin = "D2",  connect the pin to the 3,3->5 V converter. If i connect the strip to this this signal, the strip only blinks.
Then I found out if you set a pin to "HIGH" on a ESP-8266 the pin gets "LOW". Then i use a inverter (74LS04), and now i can se that the strip understand the signal. But its not blue!
But, when i change the setColor to
`ws2812fx.setColor(255, 255, 0)`
 the strip gets blue... why? :)

moose4lord, 2017-11-19:
I have the exact same setup. My 3v-5v level shifter is the same as this one:
[https://www.amazon.com/3-3V-5V-Channels-Converter-Bi-Directional-Shifter/dp/B00ZC6B8VM](https://www.amazon.com/3-3V-5V-Channels-Converter-Bi-Directional-Shifter/dp/B00ZC6B8VM)

You mentioned using GPIO5 as the data pin, which for my NodeMCU is "D1", not "D2".
See [http://play.fallows.ca/wp/wp-content/uploads/sites/4/2016/11/NodeMCU-ESP8266-12E-Limitations_thumb.jpg](http://play.fallows.ca/wp/wp-content/uploads/sites/4/2016/11/NodeMCU-ESP8266-12E-Limitations_thumb.jpg)
You might want to check that you're using the correct data pin on the NodeMCU.

So my LED_PIN statement is this:
**#define LED_PIN D1**
The GPIO5 drives an input on the 3V side of the level shifter. The 5V output of the level shifter drives the LED strip through a 100 ohm resistor. I do not use an inverter.

I have never had a problem with this setup and the colors behave as they should.