kitesurfer1404/WS2812FX #58: Compile error: error: no matching function for call to 'max(uint16_t&, int)'

mishomvp, 2017-12-02:
Hi,

I try to compile examples from WS2812FX, but i can't because some comping errors i get.
I am using Arduino IDE 1.8.5. All library is up to date.

Error log:
C:\Users\devuser\Documents\Arduino\libraries\WS2812FX\WS2812FX.cpp: In member function 'void WS2812FX::service()':

C:\Users\devuser\Documents\Arduino\libraries\WS2812FX\WS2812FX.cpp:72:63: error: no matching function for call to 'max(uint16_t&, int)'

         SEGMENT_RUNTIME.next_time = now + max(delay, SPEED_MIN);

                                                               ^

C:\Users\devuser\Documents\Arduino\libraries\WS2812FX\WS2812FX.cpp:72:63: note: candidates are:

In file included from C:\Users\devuser\documents\arduino\hardware\espressif\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\devuser\Documents\Arduino\hardware\espressif\esp8266\cores\esp8266/Arduino.h:240,

                 from C:\Users\devuser\Documents\Arduino\libraries\Adafruit_NeoPixel/Adafruit_NeoPixel.h:23,

                 from C:\Users\devuser\Documents\Arduino\libraries\WS2812FX\WS2812FX.h:40,

                 from C:\Users\devuser\Documents\Arduino\libraries\WS2812FX\WS2812FX.cpp:54:

C:\Users\devuser\documents\arduino\hardware\espressif\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4236:5: note: template<class _Tp, class _Compare> _Tp std::max(std::initializer_list<_Tp>, _Compare)

     max(initializer_list<_Tp> __l, _Compare __comp)

     ^

C:\Users\devuser\documents\arduino\hardware\espressif\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4236:5: note:   template argument deduction/substitution failed:

C:\Users\devuser\Documents\Arduino\libraries\WS2812FX\WS2812FX.cpp:72:63: note:   mismatched types 'std::initializer_list<_Tp>' and 'short unsigned int'

         SEGMENT_RUNTIME.next_time = now + max(delay, SPEED_MIN);

                                                               ^

In file included from C:\Users\devuser\documents\arduino\hardware\espressif\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\devuser\Documents\Arduino\hardware\espressif\esp8266\cores\esp8266/Arduino.h:240,

                 from C:\Users\devuser\Documents\Arduino\libraries\Adafruit_NeoPixel/Adafruit_NeoPixel.h:23,

                 from C:\Users\devuser\Documents\Arduino\libraries\WS2812FX\WS2812FX.h:40,

                 from C:\Users\devuser\Documents\Arduino\libraries\WS2812FX\WS2812FX.cpp:54:

C:\Users\devuser\documents\arduino\hardware\espressif\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4231:5: note: template<class _Tp> _Tp std::max(std::initializer_list<_Tp>)

     max(initializer_list<_Tp> __l)

     ^

C:\Users\devuser\documents\arduino\hardware\espressif\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4231:5: note:   template argument deduction/substitution failed:

C:\Users\devuser\Documents\Arduino\libraries\WS2812FX\WS2812FX.cpp:72:63: note:   mismatched types 'std::initializer_list<_Tp>' and 'short unsigned int'

         SEGMENT_RUNTIME.next_time = now + max(delay, SPEED_MIN);

                                                               ^

In file included from C:\Users\devuser\documents\arduino\hardware\espressif\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\devuser\Documents\Arduino\hardware\espressif\esp8266\cores\esp8266/Arduino.h:240,

                 from C:\Users\devuser\Documents\Arduino\libraries\Adafruit_NeoPixel/Adafruit_NeoPixel.h:23,

[... 136 lines cut ...]
     min(const _Tp& __a, const _Tp& __b)

     ^

C:\Users\devuser\documents\arduino\hardware\espressif\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note:   template argument deduction/substitution failed:

C:\Users\devuser\Documents\Arduino\libraries\WS2812FX\WS2812FX.cpp:529:40: note:   deduced conflicting types for parameter 'const _Tp' ('int' and 'uint8_t {aka unsigned char}')

   b = map(b, 0, 64, min(25, _brightness), _brightness);

                                        ^

Using library WS2812FX in folder: C:\Users\devuser\Documents\Arduino\libraries\WS2812FX (legacy)
Using library Adafruit_NeoPixel at version 1.1.3 in folder: C:\Users\devuser\Documents\Arduino\libraries\Adafruit_NeoPixel 
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\devuser\Documents\Arduino\hardware\espressif\esp8266\libraries\ESP8266WebServer 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\devuser\Documents\Arduino\hardware\espressif\esp8266\libraries\ESP8266WiFi 
Using library ArduinoJson at version 5.11.2 in folder: C:\Users\devuser\Documents\Arduino\libraries\ArduinoJson 
Using library ArduinoOTA at version 1.0 in folder: C:\Users\devuser\Documents\Arduino\hardware\espressif\esp8266\libraries\ArduinoOTA 
Using library ESP8266mDNS in folder: C:\Users\devuser\Documents\Arduino\hardware\espressif\esp8266\libraries\ESP8266mDNS (legacy)
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

If you need another information just write to me :) 

Thanks  in advance!

ibreznik, 2017-12-02:
Interestingly, I am running into a similar issue with Generic ESP8266 @mishomvp if you resolve, please make sure to post here and I will do likewise.
I just updated all libs involved and using github definition for esp8266 (aka) https://github.com/esp8266/Arduino#using-git-version

I have not got a chance to dive very deeply into it but the errors seem to come from comparisons of two values that are cast differently. I seem to be able to get rid of them when I cast each value explicitly (eg  
`b = map(b, 0, 64, min(uint8_t(25), _brightness), _brightness); `
seems to get rid of error in WS2812FX.cpp:529 but there has to be an easier way? Seems very hacky and I am not 100% sure it is the issue (it is pretty late after a long day). Any help is appreciated.

Getting the following:

> /Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.cpp: In member function 'void WS2812FX::service()':
/Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.cpp:72:63: error: no matching function for call to 'max(uint16_t&, int)'
         SEGMENT_RUNTIME.next_time = now + max(delay, SPEED_MIN);
                                                               ^
/Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.cpp:72:63: note: candidates are:
In file included from /Users/ilyareznik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:240,
                 from /Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/../Adafruit_NeoPixel/Adafruit_NeoPixel.h:23,
                 from /Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.h:40,
                 from /Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.cpp:54:
/Users/ilyareznik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4236:5: note: template<class _Tp, class _Compare> _Tp std::max(std::initializer_list<_Tp>, _Compare)
     max(initializer_list<_Tp> __l, _Compare __comp)
     ^
/Users/ilyareznik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4236:5: note:   template argument deduction/substitution failed:
/Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.cpp:72:63: note:   mismatched types 'std::initializer_list<_Tp>' and 'short unsigned int'
         SEGMENT_RUNTIME.next_time = now + max(delay, SPEED_MIN);
                                                               ^
In file included from /Users/ilyareznik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:240,
                 from /Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/../Adafruit_NeoPixel/Adafruit_NeoPixel.h:23,
                 from /Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.h:40,
                 from /Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.cpp:54:
/Users/ilyareznik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4231:5: note: template<class _Tp> _Tp std::max(std::initializer_list<_Tp>)
     max(initializer_list<_Tp> __l)
     ^
/Users/ilyareznik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4231:5: note:   template argument deduction/substitution failed:
/Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.cpp:72:63: note:   mismatched types 'std::initializer_list<_Tp>' and 'short unsigned int'
         SEGMENT_RUNTIME.next_time = now + max(delay, SPEED_MIN);
                                                               ^
In file included from /Users/ilyareznik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/xtensa-lx106-elf/include/c++/4.8.2/algorithm:61:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:240,
                 from /Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/../Adafruit_NeoPixel/Adafruit_NeoPixel.h:23,
                 from /Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.h:40,
                 from /Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.cpp:54:
/Users/ilyareznik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:260:5: note: template<class _Tp, class _Compare> const _Tp& std::max(const _Tp&, const _Tp&, _Compare)
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
/Users/ilyareznik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:260:5: note:   template argument deduction/substitution failed:
/Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.cpp:72:63: note:   deduced conflicting types for parameter 'const _Tp' ('short unsigned int' and 'int')
[... cut ...]
/Users/ilyareznik/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:193:5: note:   template argument deduction/substitution failed:
/Users/ilyareznik/Documents/Arduino/libraries/WS2812FX-master/WS2812FX.cpp:529:40: note:   deduced conflicting types for parameter 'const _Tp' ('int' and 'uint8_t {aka unsigned char}')
   b = map(b, 0, 64, min(25, _brightness), _brightness);
                                        ^
exit status 1
Error compiling for board Generic ESP8266 Module.

moose4lord, 2017-12-04:
It doesn't look like the esp8266 platform package was installed properly. Your error log makes reference to the hardware platform files here: **C:\Users\devuser\documents\arduino\hardware\espressif\esp8266** but that's not where they should be. Did you install the esp8266 platform using the Board Manager or manually install it from GitHub?

I would recommend starting over and installing the esp8266 platform using the Board Manager as described here: [https://github.com/esp8266/Arduino#installing-with-boards-manager](https://github.com/esp8266/Arduino#installing-with-boards-manager). The Board Manager will put things where they belong. After installing the platform make sure it shows up in Board Manager:

[image]

If that doesn't help, what example file are you trying to compile?

mishomvp, 2017-12-04:
Hi,
Thank you for your answer. 
1. I correct path, but with git version not working.
2. With Board Manager all is OK. No errors.

But i prefer to using git.

moose4lord, 2017-12-04:
If you're interested in getting the latest esp8266 platform code (it does have some nice bug fixes and performance improvements), you don't have to clone it from Git. You can still use the Board Manager to install the 2.4.0 release candidates. See the **Obtaining 2.4.0-rc2 pre-release** section here:
[https://github.com/esp8266/Arduino/releases/tag/2.4.0-rc2](https://github.com/esp8266/Arduino/releases/tag/2.4.0-rc2)
But of course it's pre-release code, so it's not meant for the casual user.

toblum, 2018-01-06:
Hi @moose4lord,

this problem seems to exist in the final 2.4.0 (released on 02.01.2018) also. I'm not sure now how  that can be fixed, but will also have a look at it.

Regards
Tobias

moose4lord, 2018-01-07:
Thanks for the heads up that v2.4.0 has finally been released. Dang, I was hoping this would be fixed before it was officially released.

The problem stems from the way the arithmetic min() and max() functions are implemented. Before v2.4.0 they were implemented as macros and the macro parameters could be of different data types (byte, int, long or whatever). With v2.4.0 min() and max() are now implemented using the std::min and std::max library and both function parameters have to be the same data type, the compiler won't do implicit type conversion. :(

So probably the easiest solution is to explicitly cast one of the two parameters when they are different data types. The WS2812FX library needs to be tweaked in two places.
In WS2812FX.cpp on line 72 this:
```c++
SEGMENT_RUNTIME.next_time = now + max(delay, SPEED_MIN);
```
needs to be changed to this:
```c++
SEGMENT_RUNTIME.next_time = now + max((int)delay, SPEED_MIN);
```
And on line 538 this:
```c++
lum = map(lum, 0, 64, min(25, _brightness), _brightness);
```
needs to be changed to this:
```c++
lum = map(lum, 0, 64, min(25, (int)_brightness), _brightness);
```

There's other ways to fix this problem, so I'll let Harm do whatever he thinks best, but this should get you going again in the meantime.