kitesurfer1404/WS2812FX #60: Help me please

miao084, 2017-12-05:
Arduino: 1.8.5 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from /Users/buikhanh/Documents/Arduino/libraries/WS2812FX/WS2812FX.h:40:0,
                 from /Users/buikhanh/Documents/Arduino/libraries/WS2812FX/WS2812FX.cpp:54:
/Users/buikhanh/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.h: In member function 'void WS2812FX::setLength(uint16_t)':
/Users/buikhanh/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.h:159:5: error: 'uint16_t Adafruit_NeoPixel::numLEDs' is private
     numLEDs,       // Number of RGB LEDs in strip
     ^
/Users/buikhanh/Documents/Arduino/libraries/WS2812FX/WS2812FX.cpp:154:31: error: within this context
   } while(!Adafruit_NeoPixel::numLEDs && b > 1);
                               ^
In file included from /Users/buikhanh/Documents/Arduino/libraries/WS2812FX/WS2812FX.h:40:0,
                 from /Users/buikhanh/Documents/Arduino/libraries/WS2812FX/WS2812FX.cpp:54:
/Users/buikhanh/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.h:159:5: error: 'uint16_t Adafruit_NeoPixel::numLEDs' is private
     numLEDs,       // Number of RGB LEDs in strip
     ^
/Users/buikhanh/Documents/Arduino/libraries/WS2812FX/WS2812FX.cpp:157:42: error: within this context
   _segments[0].stop = Adafruit_NeoPixel::numLEDs - 1;
                                          ^
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

moose4lord, 2017-12-09:
I compiled a small demo sketch for the "Arduino/Genuino Mega or Mega 2560" with no errors. What's your sketch look like?

miao084, 2017-12-10:
my sketch: auto_mode_cycle in examples

moose4lord, 2017-12-10:
Hmmm, I'm not having any trouble compiling that example, although I'm on a Windows box and you seem to be on a Mac. Still, it doesn't seem like it's a Mac issue. It kind of looks like your Adafruit_NeoPixel library is old or corrupted. I would try reinstalling the latest version of Adafruit_NeoPixel in your libraries folder and see if that helps. [https://github.com/adafruit/Adafruit_NeoPixel](https://github.com/adafruit/Adafruit_NeoPixel). You could also try to compile one of the Adafruit_NeoPixel examples to see if it's a WS2812FX issue or an Adafruit_NeoPixel issue.

Other then that, without being able to reproduce the problem, I'm afraid I can't be much help. Sorry.

miao084, 2017-12-10:
thanks you very much hihi

miao084, 2017-12-22:
Thank you very much
I have updated adafruit and compiled it successfully, but the problem is not fixed speed is always by default run fast.

moose4lord, 2017-12-22:
You're right, most of the example sketches have not been updated to accommodate the new speed code. The examples typically set speed=200 by default, which runs the animations very fast. Change the default speed to 1000 and you should see better results.
```c++
ws2812fx.setSpeed(1000);
```

closed 2018-05-24 by moose4lord