kitesurfer1404/WS2812FX #61: RGBW Support?

stevepeart, 2017-12-07:
Hello, the Adafruit library supports RGBW LEDs but I get odd results because i cannot use the NEO_GRBW constant with your library. Should this already be supported? Or do you still have to implement it to catch up the the Adafruit library? Thanks! Your library is awesome, btw.

moose4lord, 2017-12-09:
Did you mean the "NEO_RGBW" constant? I compiled with that constant like so:
`WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_RGBW + NEO_KHZ800);`

I don't have any RGBW LEDs to test with, but a small demo sketch using that constant compiled fine and seemed to work with my 2812s (although the colors were off as you would expect). Are you sure you have RGBW LEDs?

stevepeart, 2017-12-11:
Absolutely, they are the warm white ones, but the colors being off is what I'm talking about. I also have 300 LEDs and using this lib it seems to only make use of 2/3 of them even though I set it properly to 300, like it's counting the white led's separately or something? It definitely compiles fine though, and I am using the correct constant. In my case it's the NEO_GRBW constant.

moose4lord, 2017-12-12:
Well, WS2812FX extends the Adafruit_Neopixel class, so RGBW LEDs should be supported in both. Can you paste you code here for me to have a look? Also, can you describe your hardware setup? What microprocessor are you using to drive the LED strip?

stevepeart, 2017-12-12:
The LED strip is a BTF-Lighting RGBW running SK6812 chipped Pixels at 5v. It's 5 meters, IP65 Waterproof. The white is the "Warm White" option.

These are plugged into my arduino nano for the digital pin 6, for the control signal. I am using a 5v, 4amp power supply to power both the arduino, and the LEDs. They glow super bright with this setup.

I simply attempted to load up some of the sample scripts, only editing the number of leds and the NEO_GRBW constant addition.

moose4lord, 2017-12-12:
What version of the nano do you have? The v2 nano only had 1k bytes of SRAM, so would not support a 300 LED strip of RGBW LEDs (each LED requires 4 bytes of SRAM, so 300 x 4 = 1200bytes, too much for the v2 nano). The v3 nano has 2k of SRAM, so should be able to handle your LED strip.

The v2 nano used a ATmega168 microprocessor and the v3 nano uses a ATmega328, so if you have good eyes you can read the tiny part number printed on the microprocessor chip and tell which version you have.

You might try setting LED_COUNT to a smaller value, say 30, instead of 300.
`#define LED_COUNT 30`
That will reduce the memory requirements (to fit in a v2 nano) and the color issues may go away.

rorist, 2017-12-13:
Hello,

I'm trying to make it work too, for instance this doesn't work, less than half of the LEDs lights up:
https://github.com/rorist/sk6812-ws2812fx-mqtt-esp8266/blob/master/ha_mqtt_ledstrip.ino#L225

Everything else is working like a charm (when not setting the W leds), it does work when using neopixel only.

Thanks for the awesome library though ;)

DimIsaev, 2017-12-13:
I support the question.

what needs to be changed?
in this code 
`WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_GRB + NEO_KHZ800);`

I replaced with

`NEO_GRBW`

This not work.

stevepeart, 2017-12-14:
I took a photo of my nano and zoomed WAY in and can make out the 328 on the chip so it must be V3. When using the Adafruit library directly, I can verify that the colors are fine and it uses every LED on the strip. Definitely need to use the external power for that many LEDs but it did work well. As another note, my LEDs are the SK6812 chipped ones, but that doesn't matter when using the adafruit lib. Ill try to take some pictures or a video of it failing and bring it on here later.

moose4lord, 2017-12-14:
OK, I was curious what all the hubbub was about with the RGBW leds, so I bought a strip. They were "BTF-LIGHTING RGBW RGBNW Natural White SK6812" from Amazon ([link](http://a.co/2EpjuaH)).

I plugged them in and, wow, are these things bright! Like hurt your eyes bright. After some experimenting (with brightness=32 to save my eyes :) I found there were a few modes that didn't work well (mostly the ones that used a fade effect like BREATH, FADE and TWINKLE). Looking over the code it was evident those modes weren't written with RGBW leds in mind and were definitely broken. But most of the modes worked just fine with my new RGBW led strip. The colors all looked fine (using the NEO_GRBW setting) and when I programmed R/G/B/W, I got R/G/B/W.

So I'm lost as to why some people are seeing such strange behavior with their RGBW strips. I have made some changes to the code to fix the modes that didn't work well and all modes are working properly now. Harm merged them with his GitHub repository in record time, so everyone should download a fresh copy of WS2812FX to get the latest fixes.

@stevepeart, you said you're using a 4 amp power supply, and that seems a little weak to me. Each SW6812 draws 60-80mA when each of the RGBW leds are at full intensity. A strip of 300 will draw 18-24 amps. I'm using a 150W (30A) power supply to run my strip of 300 (Amazon [link](http://a.co/gCsW0mf)). Not sure if that has anything to do with your problems, just FYI.

moose4lord, 2017-12-14:
If anyone's interested, here's the sketch I used to test my RGBW strip. I used an Arduino Leonardo and a strip of 60 RGBW leds. The sketch just barely fits in the Leonardo, so you might have to tweak it to get it to fit in a different Arduino.

```
/*
  WS2812FX segmented led strip demo.
  
  Keith Lord - 2017
  
  FEATURES
    * example of subdividing a strip of WS2812 LEDs into segments

  LICENSE

  The MIT License (MIT)

  Copyright (c) 2017  Keith Lord 

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in
  all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
  THE SOFTWARE.

  
  CHANGELOG
  2017-10-02 initial version
  
*/

#include <WS2812FX.h>

#define LED_PIN   6  // digital pin used to drive the LED strip
#define LED_COUNT 60  // number of LEDs on the strip

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRBW + NEO_KHZ800);

unsigned long lastTime = millis();

void setup() {
  Serial.begin(115200);
  while (!Serial) { } // wait for serial port to connect on Leonardo
  Serial.println("\r\n");

  ws2812fx.init();

// test - setup to run through each mode
  ws2812fx.setBrightness(64);
  ws2812fx.setSegment(0,  0, 14, FX_MODE_STATIC, (uint32_t) {0xFF000000}, 1000, false);
  ws2812fx.setSegment(1, 15, 29, FX_MODE_STATIC, (uint32_t) {0x00FF0000}, 1000, false);
  ws2812fx.setSegment(2, 30, 44, FX_MODE_STATIC, (uint32_t) {0x0000FF00}, 1000, true);
  ws2812fx.setSegment(3, 45, 59, FX_MODE_STATIC, (uint32_t) {0x000000FF}, 1000, true);

  ws2812fx.start();
}

void loop() {
  ws2812fx.service();

  unsigned long now = millis();
  if(now - lastTime > 10000) { // change mode every 10 seconds
    uint8_t mode = ws2812fx.getMode();
    mode = (mode + 1) % ws2812fx.getModeCount();

    ws2812fx.clear();
    ws2812fx.resetSegments();
    ws2812fx.setSegment(0,  0, 14, mode, (const uint32_t[]) {0xFF000000, 0x10000000, 0x0000FFFF}, 1000, false);
    ws2812fx.setSegment(1, 15, 29, mode, (const uint32_t[]) {0x00FF0000, 0x10000000, 0x0000FFFF}, 1000, false);
    ws2812fx.setSegment(2, 30, 44, mode, (const uint32_t[]) {0x0000FF00, 0x10000000, 0x0000FFFF}, 1000, true);
    ws2812fx.setSegment(3, 45, 59, mode, (const uint32_t[]) {0x000000FF, 0x10000000, 0x0000FFFF}, 1000, true);
  
    Serial.print("mode is ");
    Serial.println(ws2812fx.getModeName(mode));
    lastTime = now;
  }
}
```

stevepeart, 2017-12-15:
@moose4lord Did you run this on an Arduino nano by chance?

moose4lord, 2017-12-15:
Sorry, the only Arduino I have is a Leonardo.

stevepeart, 2017-12-15:
Does the setSegment functionality take a bunch more memory? When I try to run all 300, it just displays the serial messages without doing anything on LEDs, but if i do 100 LEDs, the segments run totally fine. I am going to test and see how many I can get up to before it fails.

moose4lord, 2017-12-15:
The new segment feature does eat up some additional RAM, around 350 bytes, I think. As I mentioned before, 300 RGBW leds requires 1200 byes just to store the color data. Arduinos typically have only 2048 bytes of RAM, so the memory requirements of such a long strip is problematic.

One thing you can try is changing the maximum number of segments setting in the WS2812FX.h file. It's set to 10 by default:
`#define MAX_NUM_SEGMENTS 10`

If you're only going to program the entire 300 led strip as one segment, you can reduce MAX_NUM_SEGMENTS to 1 and reclaim about 300  bytes of RAM. That might free up enough RAM to give you the 1200 bytes needed. Or you can pickup an Arduino Mega2560, which has a whopping 8192 bytes of RAM. It's the 800 pound gorilla in the Arduino world.

stevepeart, 2017-12-15:
ok, I've tested using 4 segments and the max number of LEDs I can use total is 232. The next level for 4 segments evenly is 236, which did NOT work. Again, I just see the Serial messages but no lights change/work at all. Since I am currently only using 4 segments, perhaps I can change that MAX_NUM_SEGMENTS value you suggested and get use of a few more of them.

stevepeart, 2017-12-15:
Taking it down to 1 segment only, I can push it to 256 LEDs max. May have to move to a MEGA, sucks i don't have one lying around not in use at the moment.

DimIsaev, 2017-12-15:
ESP8266  +  300 LEDS  SK6812  

ere such code works successfully

`#define PIN 3         
#define NUMLEDS 300`

`WS2812FX strip = WS2812FX(NUMLEDS, PIN, **NEO_GRBW** + NEO_KHZ800);
`

stevepeart, 2017-12-16:
I tried my code with an Arduino Mega, and I get the same results as before. I notice that the max value for uint8_t = 255, weird that matches up with 0-255 for how many pixels i can get to work. Is there something in either the WS2812FX lib that might be limiting the number of pixels i can use that is assigned as uint8_t or in thew Adafruit lib itself?

stevepeart, 2017-12-16:
@DimIsaev Can we see the whole file? Does it just compile? Or does it actually light up and run all 300 lights for you?

stevepeart, 2017-12-16:
I just tested not using segments at all and just ran the code on the readme in this repo, and I can get all 300 working. Is there a maximum length to segments?

moose4lord, 2017-12-16:
The code that deals with segment length uses 16bit integers (uint16_t), so program-wise a segment's maximum length is 65536 LEDs. I have run my longest strip (300 LEDs) as a single segment, but I am using an ESP8266 processor for that, not an Arduino (like @DimIsaev).

Even when programming the demo in the ReadMe, under the covers, the code is setting up a single segment. It should make no difference if you use the statements in the ReadMe or a single setSegment() statement. Can you show us the code that works for you and the code that doesn't?

DimIsaev, 2017-12-16:
@stevepeart 

The above code was published by @moose4lord 
Work all 300 LEDs.  
When connecting a tape, it is necessary to use a capacitor and a resistor
https://learn.adafruit.com/adafruit-neopixel-uberguide/best-practices

There is only a problem with the effects of white LEDs.
sometimes the ESP restarts

rorist, 2017-12-16:
> On the ESP there is only a problem with the effects of white LEDs. sometimes the ESP restarts

I have the same problem, at least when using "Color Wipe Random" for example.

closed 2018-05-24 by moose4lord

Chemita23, 2023-07-04:
If you want use my script. It can be usefull for you.

https://github.com/Chemita23/WS2812FX_web_GRBW