kitesurfer1404/WS2812FX #64: Add and Remove Web Interface Buttons

dmasek, 2017-12-29:
As a preface, I don't have much experience with coding, and in WAY over my head with this LED programing. But I've gotten the web interface up and running, so I feel fairly accomplished already.

But how would I add a button to the web interface that would allow me to cycle through multiple effects in sequence, say for 60 seconds each. I would like to pick the effects that are included, not just all of them.

I'm also wondering how I could remove some of the effect I don't plan on using from the web interface.

 Thanks in advance for the help!

moose4lord, 2018-01-04:
There's two web interface sketches in the examples folder, esp8266_webinterface and ws2812fx_segments_web. Which one are you referring to?

dmasek, 2018-01-04:
I'm using  esp8266_webinterface. Would ws2812fx_segments_web do this
better? I was unable to find a description of them both.

moose4lord, 2018-01-04:
The ws2812fx_segments_web example sketch could be used if you want to experiment with creating multiple segments in your led strip. If you just want to program your strip as one long segment, then esp8266_webinterface is a better example.

You mentioned customizing the list of effects and that can be done by modifying the modes_setup() function in esp8266_webinterface. Create an array of effect numbers and use that to generate the \<li\> elements in the web interface's list of effects. Something like this:

```c++
 // *** create a custom list of effects/modes (see WS2812FX.h for list of mode numbers)
uint8_t myModes[] = {1,5,31};
void modes_setup() {
  modes = "";
//for(uint8_t i=0; i < ws2812fx.getModeCount(); i++) {
  for(uint8_t i=0; i < sizeof(myModes); i++) { // *** loop over myModes[]
    uint8_t m = myModes[i]; // *** get myMode[] element
    modes += "<li><a href='#' class='m' id='";
//  modes += i;
    modes += m; // *** set HTML element id
    modes += "'>";
//  modes += ws2812fx.getModeName(i);
    modes += ws2812fx.getModeName(m);  // *** get mode name
    modes += "</a></li>";
  }
}
```

Your other question about a button to automatically cycle through a custom list of effects is a bit harder. Take a look at the auto_mode_cycle example sketch, which shows a nice way to cycle through all the effects. However, marrying that with a button on the web interface involves mucking with the HTML, JavaScript eventhandlers and AJAX network code. Not impossible, but not trivial either, especially for a beginner.

moose4lord, 2018-01-07:
@dmasek, I needed to make a change to the esp8266_webinterface example sketch, so I put some thought into your ideas while I was working on the code. I've added the "automatically cycle" button to the web interface and a means to create a custom list of effects. Harm has merged these changes with the latest code branch, so it's available for you to use. Just FYI.

dmasek, 2018-01-08:
@moose4lord , Thanks so much for adding these features!

closed 2018-05-24 by moose4lord