kitesurfer1404/WS2812FX #66: esp8266 hang and reset after running effect

phutruong1801, 2018-01-11:
Hi everyone. 
I need a little help about my project 
i'm using esp 8266 node mcu  to control 2 button, DHT22 and ws2812 via node-red MQTT pubsubclient
the problem is my esp being hang and sometime it will reset the esp. sometime not. 
i believe that the WS2112FX cause the hang. 
here is my code. please help me fix it. i waste 2 days to find the problem but i fall in love with the effect of WS2812FX

#include <PubSubClient.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <SoftwareSerial.h>
#include "DHT.h"
#include <Wire.h>
#include <WS2812FX.h>
#define LED_COUNT 216
#define LEDPIN D7
const int SPEED = 1000;

const char* ssid = "WifiChua";
const char* password = "Phuthuy@@1107";
const char* mqtt_server = "192.168.22.108";
int r;
int g;
int b;
const int Den_Pin1 = D1;
const int Den_Pin2 = D2;
const int Button1 = D5;
const int Button2 = D6;
int  den1status = HIGH;
int  den2status = HIGH;
int  State1 = LOW;
int  State2 = LOW;
//----------
#define DHTPIN D8
#define DHTTYPE DHT22
//boolean debug = false;
boolean debug = true;
WiFiClient espClientroom;
DHT dht(DHTPIN, DHTTYPE);
unsigned long lastMeasure = 0;
unsigned long lastMillis = 0;
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LEDPIN, NEO_GRB + NEO_KHZ800);
//long lastMsg = 0;
//char msg[50];
//int value = 0;
PubSubClient client(espClientroom);
void nhand1(char* tp, byte* nd, unsigned int length) {
  String topic(tp);
  String noidung = String((char*)nd);
  noidung.remove(length);

  if (topic == "LED/dk")
  {
    const int EF = noidung.toInt();
    switch (EF) {
      case 1:
        ws2812fx.setMode(FX_MODE_RAINBOW);
        break;
      case 2:
        ws2812fx.setMode(FX_MODE_HALLOWEEN);
        break;
      case 3:
        ws2812fx.setMode(FX_MODE_RANDOM_COLOR);
        break;
      case 4:
        ws2812fx.setMode(FX_MODE_STATIC);
        break;
      case 5:
[... 350 lines cut ...]
  float t = dht.readTemperature();
  float f = dht.readTemperature(true);
  // Check if any reads failed and exit early (to try again).
  if (isnan(h) || isnan(t) || isnan(f)) {
    Serial.println("Fall to read sensor");
    return;
  }
  float hic = dht.computeHeatIndex(t, h, false);
  static char temperatureTemp[7];
  dtostrf(hic, 6, 2, temperatureTemp);
  static char humidityTemp[7];
  dtostrf(h, 6, 2, humidityTemp);
  client.publish("Mainroom/Temp", temperatureTemp);
  client.publish("Mainroom/Hum", humidityTemp);
  Serial.print("Humidity: ");
  Serial.print(h);
  Serial.print(" %\t Temperature: ");
  Serial.print(t);
  Serial.print(" *C ");
  Serial.print(f);
  Serial.print(" *F\t Heat index: ");
  Serial.print(hic);
  Serial.println(" *C ");
  delay(50);
}

moose4lord, 2018-02-16:
I feel your pain. I tried to get MQTT and WS2812FX to work together, but finally gave up. I could never get it to work reliable, my ESP8266 would reboot at random times. I poked through the crash dumps, but wasn't able to figure out what was causing the reboots. Sorry, I wish I could be more help.

I think if I were to pursue it again, I would separate the MQTT functions from the WS2812FX functions, having the ESP8266 handle the MQTT part and a separate Arduino take care of the LEDs. Just a thought.

tobi01001, 2018-02-17:
Maybe this helps? 

https://github.com/euphi/ESP_Homie_WS2812FX

synssins, 2018-03-01:
Have any of you looked into McLighting v2? It uses the ws2812fx library and adds REST, MQTT and WebSockets, as well as a nifty webUI for controlling it.

moose4lord, 2018-03-02:
I started out using McLighting. It's quite nice. Easy to use and now it saves the last setup to eeprom, so it might be all you need. Doesn't support segments though :(

superbool, 2018-10-10:
I also encountered the problem，actually not the mqtt or the ws2812fx problem  but the DHT.At least it is to me.  When I removed all code of  DHT, it worked fine. I found out the four lines have problems:
```
dtostrf(hic, 6, 2, temperatureTemp);
dtostrf(h, 6, 2, humidityTemp);
Serial.print(h);
Serial.print(t);
```
but I don't know why and which line make it crashed.

I only subscribe one topic ,but you subscribe several topics, I don't know it is affect.

here is my code, it works fine:
```
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <WS2812FX.h>

#define LED_COUNT 300
#define LED_PIN 2
#define TIMER_MS 5000

const char *ssid = "xxxx";
const char *password = "xxx";

const char *mqtt_server = "xxxx";
const unsigned int mqtt_port = xxxx;
const char *mqtt_username = "xxx";
const char *mqtt_password = "xxxx";
long lastMsg = 0;
char msg[50];
int value = 0;
unsigned long last_change = 0;
unsigned long now = 0;
const uint32_t colors[] = {RED, BLUE, WHITE};
const uint32_t heart_colors[] = {RED, WHITE, BLACK};
const uint32_t colors_black[] = {BLACK};

WiFiClient espClient;
PubSubClient client(espClient);
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup_ws2812()
{
    ws2812fx.init();
    ws2812fx.setBrightness(255);
    ws2812fx.setSpeed(200);
    ws2812fx.setColor(0x000000);
    ws2812fx.setMode(FX_MODE_STATIC);
    ws2812fx.setSegment(0, 0, 53, 0, colors_black, 1000, false);
    ws2812fx.start();
}

void setup_wifi()
{
    delay(10);
    // We start by connecting to a WiFi network
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

uint *parseData(String str)
{
    static uint data[5];
    uint r = 0, t = 0;
    for (uint i = 0; i < str.length(); i++)
    {
        if (str.charAt(i) == ';')
        {
            data[t] = str.substring(r, i).toInt();
            r = (i + 1);
            t++;
        }
    }
    return data;
}

void callback(char *topic, byte *payload, uint length)
{
    Serial.print("Message arrived [");
    Serial.print(topic);
    Serial.print("] ");
    String str = "";
    for (uint i = 0; i < length; i++)
    {
        str += (char)payload[i];
    }
    Serial.print(str);
    Serial.println();

    uint *data = parseData(str);
    if (data[0] == 1)
    { 
        ws2812fx.setPixelColor(data[1], data[2], data[3], data[4]);
        ws2812fx.show();
    }
    else if (data[0] == 2)
    { 
        uint segment = data[1];
        if (segment == 0)
        { 
            ws2812fx.setSegment(0, 0, 53, data[2], data[4] == 1 ? colors : colors_black, data[3], false);
        }
        else if (segment == 1)
        { 
            ws2812fx.setSegment(1, 54, 118, data[2], data[4] == 1 ? heart_colors : colors_black, data[3], false);
        }
        else if (segment == 2)
        { 
            ws2812fx.setSegment(2, 119, 159, data[2], data[4] == 1 ? colors : colors_black, data[3], false);
        }else if (segment == 3)
        { 
            ws2812fx.setSegment(3, 173, 248, data[2], data[4] == 1 ? colors : colors_black, data[3], false);
        }else {
            ws2812fx.resetSegments();
        }
    }
}

void reconnect()
{
    // Loop until we're reconnected
    while (!client.connected())
    {
        Serial.print("Attempting MQTT connection...");
        // Attempt to connect
        if (client.connect("ESP8266Client", mqtt_username, mqtt_password))
        {
            Serial.println("connected");
            client.subscribe("light_up_led");
            // Once connected, publish an announcement...
            client.publish("esp_msg", "starting", true);
            // ... and resubscribe
        }
        else
        {
            Serial.print("failed, rc=");
            Serial.print(client.state());
            Serial.println(" try again in 5 seconds");
            // Wait 5 seconds before retrying
            delay(5000);
        }
    }
}

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT); // Initialize the BUILTIN_LED pin as an output
    Serial.begin(9600);
    setup_wifi();
    client.setServer(mqtt_server, mqtt_port);
    client.setCallback(callback);
    setup_ws2812();
}

void loop()
{
    //mqtt
    if (!client.connected())
    {
        reconnect();
    }
    client.loop();

    ws2812fx.service();

}
```

moose4lord, 2019-01-20:
I think this conversation has run its course. Closing.

closed 2019-01-20 by moose4lord