kitesurfer1404/WS2812FX #67: Store settings after power lost

Mulfarion, 2018-01-14:
I use arduino nano with a strip ws2812b and use "serial_control". And when the power goes out, the settings are reset to the default values. How I can use EEPROM for store settings and use it for control?

kitesurfer1404, 2018-01-14:
You might take a look at the Arduino examples 
https://www.arduino.cc/en/Tutorial/EEPROMRead
and
https://www.arduino.cc/en/Tutorial/EEPROMUpdate

On every settings-change update EEPROM. Read the FX settings from EEPROM in your setup() routine.

Mulfarion, 2018-01-17:
At the moment I was able to implement the storage and saving of settings. But I ran into a small problem. How to convert uint_32 into hex?

DeuxVis, 2018-01-18:
I think you cannot do it directly in an initialization.

This would work : 
```
    //Data to store.
    MyObject customVar = {
        value1,
        value2,
        "",
        value4
    };
    value3.toCharArray( customVar.name, 10 );
```

More details : https://stackoverflow.com/questions/7383606/converting-an-int-or-string-to-a-char-array-on-arduino

Mulfarion, 2018-01-18:
This is not a solution. I need to store hex colors in eeprom arduino

DeuxVis, 2018-01-18:
This is the solution for the error you have "error: cannot convert 'String' to 'char' in initialization".

You can still save customVar to eeprom after that.

Mulfarion, 2018-01-18:
You can not flood. And if you do not know the solution of the task, then it is not necessary to say obvious things. I know that this parameter causes an error, but the question is how to correctly save all the parameters. And what you are proposing is the same thing, that instead of castration, cut off the lower part of the body. Sorry for straightness.

Mulfarion, 2018-01-18:
Ok, how I can correctly save ALL the parameters to eeprom?

tobi01001, 2018-01-18:
As @DeuxVis said. You may need to  initialize with an empty char array `"",` - at least not with a String. Then you "convert" your string `value3` to this char array. Then you can write your struct to the EEPROM.
You could also initialize with a "default color" - just in case.

Since size of the "buffer" is fixed to 10 anyway (according to your struct).
Then in 
`//Data to store.
    MyObject customVar = {
      value1,
      value2,
      value3.c_str(),
      value4
    };`
should work as well. If again not during initialization, then with empty char array and assigning the `customVar.name = value3.c_str();` afterwards.

Mulfarion, 2018-01-19:
This is not work

moose4lord, 2018-01-19:
You don't need to construct your own object to store the settings, the WS2812FX library already has a struct called **WS2812FX::segment** that does that:
```c++
    typedef struct segment {
      uint8_t  mode;
      uint32_t colors[NUM_COLORS];
      uint16_t speed;
      uint16_t start;
      uint16_t stop;
      bool     reverse;
    } segment;
```
An array of such structs stores the settings for all segments:
```c++
segment _segments[MAX_NUM_SEGMENTS]
```
Although the _segments[] array is a private object, not directly accessible by your sketch, you can access the data using the getSegments() method.

More to your question about storing setup info to EEPROM, there's two pieces of data you need to store, 1) the **_segments[]** array mentioned above, and 2) the **_num_segments** value, which is a byte that stores the number of segments currently in use.

So use this method to store the current setup in EEPROM:
```c++
void saveToEEPROM() {
  EEPROM.put(0, ws2812fx.getNumSegments()); // save the num_segments data

  WS2812FX::segment *segs = ws2812fx.getSegments(); // save the segments data
  for(int i=0; i<ws2812fx.getNumSegments(); i++) {
    EEPROM.put(1 + i * sizeof(WS2812FX::segment), segs[i]);
  }
  EEPROM.commit(); // for ESP8266 (comment out if using an Arduino)
}
```
And use this method to restore the setup from EEPROM:
```c++
void restoreFromEEPROM() {
    ws2812fx.resetSegments(); // reset the segments data

    uint8_t num_segments; // restore the num_segments data
    EEPROM.get(0, num_segments);
    ws2812fx.setNumSegments(num_segments);

    WS2812FX::segment *segs = ws2812fx.getSegments(); // restore the segments data
    for(int i=0; i<ws2812fx.getNumSegments(); i++) {
      EEPROM.get(1 + i * sizeof(WS2812FX::segment), segs[i]);
    }
}
```

I've attached a small demo sketch to show these two methods in action. Note, the demo sketch is written for an ESP8266 processor, which implements the EEPROM a little different from an Arduino. The demo uses **EEPROM.begin(256)**; and **EEPROM.commit();** to initialize and write to the EEPROM. If you're using an Arduino, you'd use **EEPROM.begin();** and comment out the **EEPROM.commit()** command.

[ws2812fx_segments_eeprom_demo.zip](https://github.com/kitesurfer1404/WS2812FX/files/1647770/ws2812fx_segments_eeprom_demo.zip)

Mulfarion, 2018-01-19:
Thanks! It's cool work!

Mulfarion, 2018-01-20:
But I see what brightness this not save?

moose4lord, 2018-01-20:
Yes, I forgot about the global brightness setting. You can certainly add that to the **saveToEEPROM()** and **restoreFromEEPROM()** methods (use the **getBrightness()** and **setBrightness()** methods to access the brightness data).

I would probably not recommend the EEPROM functionality be added to the WS2812FX library. It adds a dependency on EEPROM.h to support what, to me, isn't a core feature. Even a small WS2812FX sketch uses over 90% of the available flash on an Arduino, so we have to be careful about making the library even bigger. But that's just my 2 cents.

Mulfarion, 2018-01-21:
Thanks for your reply. Can I ask you to show how can add brightness methods to saveToEEPROM() and restoreFromEEPROM()?

moose4lord, 2018-01-21:
I'm a firm believer in the "give a man a fish...teach a man to fish..." proverb, so I didn't want to just give you the solution, without giving you the opportunity to solve it on your own. I encourage you to give it a try. You'll learn something and nothing beats that sense of accomplishment when you succeed.

On the other hand, if you're just not into software development, I don't want to leave you hanging. So I've attached an updated version of my demo sketch, which handles brightness too.

[ws2812fx_segments_eeprom_demo.zip](https://github.com/kitesurfer1404/WS2812FX/files/1649776/ws2812fx_segments_eeprom_demo.zip)

Mulfarion, 2018-01-21:
The solution turned out to be so simple. And I tried another option that did not work. A week fought over this task, but you very much helped. Thank you very much! This is the first time I have encountered writing code for arduino, and coding in c#. In the near future, I want to make a website with my free program to manage led tape.

closed 2018-01-22 by Mulfarion