kitesurfer1404/WS2812FX #70: control with http request

dokho33, 2018-01-20:
Hello,

I'm using esp8266_webinterface sketch and i'd like to control it with http request like: "http://ip_of_wemos/set?b=255" to set brightness to 255.

Is it possible ?

Regards,

moose4lord, 2018-01-20:
Sure. This is the code to adjust brightness:
```c++
if(server.argName(i) == "b") { // adjusting brightness
  if(server.arg(i)[0] == '-') { // check parameter for '-' character
    ws2812fx.setBrightness(max((int)(ws2812fx.getBrightness() * 0.8), 5)); // decrease brightness by 20%
  } else { // if parameter is not '-' character, assume it's '+' character
    ws2812fx.setBrightness(min((int)(ws2812fx.getBrightness() * 1.2), 255)); // increase brightness by 20%
  }
}
```

You'd need to add some code to check for a numeric value (in addition to the '-' and '+' chars). Similar to the mode adjustment code, which uses strtol() to retrieve a numeric parameter:
```c++
uint8_t tmp = (uint8_t) strtol(&server.arg(i)[0], NULL, 10);
```

tobi01001, 2018-01-20:
@moose4lord 
Is there any particular reason why you use 
`uint8_t tmp = (uint8_t) strtol(&server.arg(i)[0], NULL, 10);`
instead of 
`uint8_t tmp = (uint8_t)server.arg(i).toInt();`?

Other than that `strtol `would accept blanks, signs... while `toInt `would just return `0`

Thanks.

moose4lord, 2018-01-20:
I can only guess that Harm used strtol() because it can convert strings representing hexadecimal numbers into ints. toInt() can't do that. See the code that adjusts color:

```c++
uint32_t tmp = (uint32_t) strtol(&server.arg(i)[0], NULL, 16);
```

dokho33, 2018-01-21:
Hello,
I've try with this code:
`if(server.argName(i) == "b") {
      if(server.arg(i)[0] == '-') {
        ws2812fx.setBrightness(max((int)(ws2812fx.getBrightness() * 0.8), 5));
      }
      if(server.arg(i)[0] == '+') {
        ws2812fx.setBrightness(min((int)(ws2812fx.getBrightness() * 1.2), 255));
      } else {
      uint8_t tmp = (uint8_t) strtol(&server.arg(i)[0], NULL, 10);
      ws2812fx.setBrightness(tmp);  
      }
      Serial.print("brightness is "); Serial.println(ws2812fx.getBrightness());
    }`

It's working with http request but when i try to increase/decrease brightness from web interface led goes down (brightness = 0 ?).

Another problem is that speed is working reverse on web interface (clik on + make speed decreasing).

PS: sorry for my bad english

moose4lord, 2018-01-21:
There's a couple problems with your code. First, you need to structure your "if" statements, so only one of the setBrightness() commands is executed. Use "if" and "else if" statements to do that.

Second, the "+" character has special meaning when used in a URL. Although the browser will send a "+" character, it will be transformed (URL decoded) into a space character by the web server. See [https://www.w3schools.com/tags/ref_urlencode.asp](https://www.w3schools.com/tags/ref_urlencode.asp). Welcome to the foibles of web interface programming :)  So the code needs to test for a space character, not a '+' character.

This should work better:
```c++
if(server.argName(i) == "b") {
  if(server.arg(i)[0] == '-') {
    ws2812fx.setBrightness(max((int)(ws2812fx.getBrightness() * 0.8), 5));
  } else if(server.arg(i)[0] == ' ') { // test for space char, not + char
    ws2812fx.setBrightness(min((int)(ws2812fx.getBrightness() * 1.2), 255));
  } else {
    uint8_t tmp = (uint8_t) strtol(&server.arg(i)[0], NULL, 10);
    ws2812fx.setBrightness(tmp);
  }
  Serial.print("brightness is ");
  Serial.println(ws2812fx.getBrightness());
}
```
Regarding the speed working in reverse, that's the way it's supposed to work. Increasing the speed value will make animations run slower.

dokho33, 2018-01-22:
Thank you, your code is working except if i put brightness to 0 with http request: after that it's impossible to increase it with web interface :(

i'll try to adapt your code for control speed from http request.

Regards,

tobi01001, 2018-01-22:
That is because `0 * 1.2` is still `0`.

You may need to catch the zero brightness:
```c
else if(server.arg(i)[0] == ' ') { // test for space char, not + char
  if(ws2812fx.getBrightness() == 0) {
    ws2812fx.setBrightness(5);  // 5 as the minimum value from above "-"
  }
  else {
    ws2812fx.setBrightness(min((int)(ws2812fx.getBrightness() * 1.2), 255));
  }
}
```

closed 2018-05-24 by moose4lord