kitesurfer1404/WS2812FX #71: REDUCED_MODES not working?

gazer000, 2018-01-23:
Using my Arduino Micro to control my WS2812b ring and the sketch size is just too big for it so I added #define REDUCED_MODES thinking it can shrink the sketch size, but before and after adding it, the sketch size does not shrink but remain the same size. Also I only want a certain effects, can I add more effects I wish to be replaced with STATIC into the library?

moose4lord, 2018-01-24:
Make sure you add the "#define REDUCED_MODES" statement before the "#include <WS2812FX.h" statement in your sketch. When I do that to a simple sketch like the ws2812fx_segments example, the sketch size drops from 25346 bytes to 21720  bytes.

If you need to reduce the sketch size even more, you can add more modes to the #ifdef REDUCED_MODES code block in WS2812FX.h. Or you can simply comment out the modes you don't need in the constructor in WS2812FX.h like so:

```c++
      _mode[FX_MODE_RUNNING_RED_BLUE]        = &WS2812FX::mode_running_red_blue;
//    _mode[FX_MODE_RUNNING_RANDOM]          = &WS2812FX::mode_running_random;
      _mode[FX_MODE_LARSON_SCANNER]          = &WS2812FX::mode_larson_scanner;
```
The linker will remove the modes that have been commented out from the compiled sketch.

gazer000, 2018-01-24:
Thanks for your reply. By limiting it to only 6 modes the sketch size reduced to only 10kb, which gives me a lot of space to add in more codes for my manual-controlled 4 pins RGB. I also liberate some memory by deleting the getModeName in the cpp file and all the lines starting with _NAME in WS2812fx.h. I guess this thread can be closed now.

closed 2018-01-24 by gazer000