kitesurfer1404/WS2812FX #72: Stop button do nothing.

j00e, 2018-02-08:
Hello my Stop button in web interface do nothing. I would very much like it to completely disable all the effects and the tape was turned off.

moose4lord, 2018-02-14:
The play and stop buttons at the bottom of the esp8266_webinterface example web page are for starting and stopping the sketch feature that automatic rotates the mode every 10 seconds. Clicking the stop button just stops updating the mode, so the last mode continues to run. If you want to stop the animation and clear the LED strip, you'll need to tweak the sketch.

To stop the animation and clear the LEDs you need to call ws2812fx.stop() in the code that handles the stop button (near the bottom of the sketch), like so:

```c++
if(server.arg(i)[0] == '-') {
  auto_cycle = false;
  ws2812fx.stop(); // add this to stop and clear the animation
 } else {
```
Unfortunately, this has the undesirable side effect that you can't restart the animations. To fix that, add a call to ws2812fx.start() at the top of the srv_handle_set() function, like so:

```c++
void srv_handle_set() {
  ws2812fx.start(); // add this to restart animations
  for (uint8_t i=0; i < server.args(); i++){
```

j00e, 2018-02-14:
Sorry for my noob question, in what file i must put this code?

moose4lord, 2018-02-14:
In the webinterface sketch file, esp8266_webinterface.ino. I assumed you were working on that sketch.

closed 2018-05-24 by moose4lord

Evan-Ray, 2019-05-27:
How would I add an off button at the top of the page, so that the stop button is still usable?