kitesurfer1404/WS2812FX #73: How can i add custom effects?

Primus007, 2018-02-10:
How can i add custom effects like this?
`for(int x = 0; x <30; x++)
{
int flicker = random(0,40);
int r1 = r-flicker;
int g1 = g-flicker;
int b1 = b-flicker;
if(g1<0) g1=0;
if(r1<0) r1=0;
if(b1<0) b1=0;
strip.setPixelColor(x,r1,g1, b1);
}
strip.show();
`delay(random(50,150));``

thanks for help....

moose4lord, 2018-02-14:
Adding a custom effect/mode involves quite a few steps, so to make it easier you might want to start with replacing an existing effect with your own.  For example, to replace the blink effect with your custom effect, edit the WS2812FX.cpp file in the WS2812FX library folded (i.e. Arduino\libraries\WS2812FX\WS2812FX.cpp). Change the code for the blink effect from this:

```c++
uint16_t WS2812FX::mode_blink(void) {
  return blink(SEGMENT.colors[0], SEGMENT.colors[1], false);
}
```
to this (using your code snippet as an example):

```c++
uint16_t WS2812FX::mode_blink(void) {
	for(uint16_t i=SEGMENT.start; i <= SEGMENT.stop; i++) { // loop over every LED in the segment
		int flicker = random(0, 40);
		int r1 = r - flicker;
		int g1 = g - flicker;
		int b1 = b - flicker;
		if(g1<0) g1=0;
		if(r1<0) r1=0;
		if(b1<0) b1=0;
		Adafruit_NeoPixel::setPixelColor(i, r1, g1, b1);
	}
	return random(50, 150);
}
```
Note, this doesn't compile, because your code snippet includes three variables (r, g and b), which are undeclared. You'll have to initialize those three variables somehow. Even so, your snippet keeps subtracting a random amount from each RGB color, so eventually they will all be zero and all the LEDs will be off. So you have some thinking to do. But hopefully this will get you started.

ChristianKnorr, 2018-05-02:
Hi @All
Does anyone have added the FastLED Fire2012 into CustomMode?
Thank You

ChristianKnorr, 2018-05-04:
Thank you but it wont work for me.
All LEDs are red. thats it :(

moose4lord, 2018-05-05:
Try this.
[ws2812fx_Fire2012.zip](https://github.com/kitesurfer1404/WS2812FX/files/1977068/ws2812fx_Fire2012.zip)

kitesurfer1404, 2018-05-24:
Thanks to @moose4lord there is now also a tutorial/userguide. I'll close this issue. Feel free to reopen if more help is needed.

closed 2018-05-24 by kitesurfer1404