kitesurfer1404/WS2812FX #74: pixel lenght of animation

typhoonss92, 2018-02-13:
is there a way for lets say the larson scanner to make the animated pixels longer from the default 4 or 5 pixels to lets say 10 pixels long. I have looked thru the code and tried playing with numbers but none of them make the length longer they just make the animation go off the strip and come back after a few seconds. 
Thanks

moose4lord, 2018-02-14:
Probably the easiest way to affect the length of "on" LEDs in the Larson scanner is to change the fade_out() function. Currently it's this (see the WS2812FX.cpp file in the WS2812FX library):

```c++
void WS2812FX::fade_out() {
  for(uint16_t i=SEGMENT.start; i <= SEGMENT.stop; i++) {
    uint32_t color = Adafruit_NeoPixel::getPixelColor(i);
    color = (color >> 1) & 0x7F7F7F7F;
    Adafruit_NeoPixel::setPixelColor(i, color);
  }
}
```
It decreases the moving LED's brightness by 50% for each animation loop, causing the "tail" to fade away. The algorithm is fast, but not very flexible. Something like this might be what you're looking for:

```c++
void WS2812FX::fade_out() {
  for(uint16_t i=SEGMENT.start; i <= SEGMENT.stop; i++) {
    uint32_t color = Adafruit_NeoPixel::getPixelColor(i);
    byte w = (color >> 24) & 0xFF;
    byte r = (color >> 16) & 0xFF;
    byte g = (color >>  8) & 0xFF;
    byte b = (color        & 0xFF);
    Adafruit_NeoPixel::setPixelColor(i, r*0.9, g*0.9, b*0.9, w*0.9);
  }
}
```
It fades out the tail by decreasing each LED's brightness by 10% for each animation loop, producing a longer tail. Adjust the "0.9" factor in the setPixelColor() function to get the length you desire.

typhoonss92, 2018-02-14:
This is perfect, thank you. Basically I am only using red and I am going to change the patterns via siri thru home assistant. This is going in my truck where I have 16 relays and 5 temp sensors and MSGEQ7 chip for 3 sets of red leds to move to music. There is a RPi running as a AP for the network and a hotspot it connects to for internet. Having this will just add to the fun, for now this will just be in the grill but I may expand to other parts of the truck with the neopixel lights. Thanks for all your help, greatly appreciated

closed 2018-02-14 by typhoonss92