kitesurfer1404/WS2812FX #75: OSC trigger different effetcs

leexorsist, 2018-02-20:
Hello, im making a project whit osc touch and neopixels in  a nodemcu V3, 
mi question is : 

 how  can i trigger diferent effects whit a diferent osc adress in each one?

the code is running pretty good but i only work whit one effect, (the main declared in setMode) 

i cant find where or how put the different efeccts, because i want it an especific adress react with especific effect.

any ideas?
thanks for any help!

`#include <WS2812FX.h>
#include <Adafruit_NeoPixel.h>
#include <ESP8266WiFi.h>
#include <OSCBundle.h>  
#include <WiFiUdp.h>

#define LED_COUNT 9
#define LED_PIN 15

char ssid[] = "-----------";                 // your network SSID (name)
char pass[] = "-----------";  

WiFiUDP Udp;                          
const unsigned int localPort = 8005;  
const unsigned int destPort = 9005;

bool oldState = HIGH;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

unsigned long last_trigger = 0;
unsigned long now = 0;

void setup() {

   Serial.begin(115200);
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);
    WiFi.begin(ssid, pass);
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
 Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    Serial.println("Starting UDP");
    Udp.begin(localPort);
    Serial.print("Local port: ");
    Serial.println(Udp.localPort());

  ws2812fx.init();
  ws2812fx.setBrightness(50);
  ws2812fx.setMode(FX_MODE_BLINK);

  
  
  // put your setup code here, to run once:

}

void loop() {

  ws2812fx.service();
  
  OSCMessage msgIN;
  int size;
  if((size = Udp.parsePacket())>0){
    while(size--)
      msgIN.fill(Udp.read());
    if(!msgIN.hasError()){
      msgIN.route("/1/toggleLED", osc1);  
    }
  }
}

 void osc1 (OSCMessage &msg, int addrOffset){     //////1er msg
  oldState = (boolean) msg.getFloat(0);
  if(msg. isFloat ( 0 )) {
  ws2812fx.trigger();
  OSCMessage msgOUT("/1/toggleLED");
  msgOUT.add(oldState);
  
  
       if (oldState) {
    Serial.println("LED on");
  }  
 
  else {
    Serial.println("LED off");
  }
  
 oldState = !oldState; 

  
 Udp.beginPacket(Udp.remoteIP(), destPort);
 msgOUT.send(Udp); // send the bytes
 Udp.endPacket(); // mark the end of the OSC Packet
 msgOUT.empty(); // free space occupied by message
  }
  }`

moose4lord, 2018-02-22:
I don't think you're using the trigger() function properly. trigger() is used to manually initiate one animation step. It doesn't really have anything to do with changing effects.

I don't know anything about OSC, but looking over your code I think you want something like this:
  1) setup routes to have OSC addresses execute functions to change effect
```c++
msgIN.route("/0", doStatic);
msgIN.route("/1", doBlink);
msgIN.route("/2", doBreathe);
msgIN.route("/3", doColorWipe);
etc...
```

  2) create the functions that change effects:
```c++
void doStatic(OSCMessage &msg, int addrOffset) {
  ws2812fx.setMode(FX_MODE_STATIC);
}
void doBlink(OSCMessage &msg, int addrOffset) {
  ws2812fx.setMode(FX_MODE_BLINK);
}
etc...
```

leexorsist, 2018-02-22:
Thanks for the answer!, I try to doit in that way, 

- i make the function to change the effect like your step 2, but all the pixels are in red,
- when i quit the trigger-nothing happens
- when the setMode are in the setup and the trigger there´s ,  works fine

maybe theres a trouble with the boolean ```oldState ```?

many thanks for your help

```
void doBlink (OSCMessage &msg, int addrOffset){
  ws2812fx.setMode(FX_MODE_BLINK_RAINBOW);
  oldState = (boolean) msg.getFloat(0);
  if(msg. isFloat ( 0 )) {
  ws2812fx.trigger(); 
  OSCMessage msgOUT("/1/toggleLED");
  msgOUT.add(oldState);```

moose4lord, 2018-02-23:
I dug into OSC a little bit and created a demo sketch which changes the mode.

```c++
#include <WS2812FX.h>
#include <Adafruit_NeoPixel.h>
#include <ESP8266WiFi.h>
#include <OSCBundle.h>
#include <WiFiUdp.h>

#define LED_COUNT 30
#define LED_PIN 5

char ssid[] = "xxxxxxxx"; // your network SSID (name)
char pass[] = "xxxxxxxx";

WiFiUDP Udp;
const unsigned int localPort = 8005;
const unsigned int destPort = 9005;

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println("Starting UDP");
  Udp.begin(localPort);
  Serial.print("Local port: ");
  Serial.println(Udp.localPort());

  // setup the initial ws2812 config
  ws2812fx.init();
  ws2812fx.setBrightness(50);
  ws2812fx.setMode(FX_MODE_BLINK);
  ws2812fx.setColor(GREEN);
  ws2812fx.setSpeed(1000);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();

  if(Udp.parsePacket() > 0) {
    OSCMessage msgIN;
    char buffer[30];
    int length = Udp.read(buffer, sizeof(buffer));
    msgIN.fill((uint8_t *)buffer, length);
    if(!msgIN.hasError()){
      msgIN.route("/1/mode", setMode);
    }
  }
}

void setMode(OSCMessage &msg, int addrOffset){
  Serial.print("setmode message received - size:"); Serial.print(msg.size());
  Serial.print(" type:"); Serial.print(msg.getType(0));
  Serial.print(" length:"); Serial.println(msg.getDataLength(0));
  if(msg.isInt(0)) {
    int mode = msg.getInt(0);
    if(mode >= 0 && mode < ws2812fx.getModeCount()) {
      Serial.print("setting mode to "); Serial.println(mode);
      ws2812fx.setMode(mode);
    }

    OSCMessage msgOUT("/1/mode");
    msgOUT.add(mode);

    Udp.beginPacket(Udp.remoteIP(), destPort);
    msgOUT.send(Udp); // send the bytes
    Udp.endPacket(); // mark the end of the OSC Packet
    msgOUT.empty(); // free space occupied by message
  }
}
```

I used an Android program called QuickOSC (from the Google play store) to do my testing and it seems to work well. Note, from what I've seen looking at OSC message examples, the format should be something like "/1/mode/0" (like a RESTful API). However, the only way I could get things to work was by programming the buttons in QuickOSC to send messages in the format "/1/mode 0". Note, the last slash character is replaced with a space character. Not sure if this is the way it's supposed to work, or if it's a quirk in the Arduino OSC library.

Hope this gets your project back on track.

leexorsist, 2018-02-24:
Well im keep a while, I had to use Float Oscmessage because I pretend to send musical notes througth osc, and I want the fx to react while the note is pushed,

finally i  did the Fx change triggering when i push a button with Osc code,
i would like to modify the uint8_t "m" to put it to specific fx, i tried whit number but i dont know how works this well,
```
void setup() {
 ws2812fx.init();
  ws2812fx.setBrightness(50);
  ws2812fx.setColor(GREEN);

}

void loop() {
ws2812fx.service();

  OSCMessage msgIN;
  int size;
  if((size = Udp.parsePacket())>0){
    while(size--)
      msgIN.fill(Udp.read());
    if(!msgIN.hasError()){
      msgIN.route("/1/toggleLED", osc1);  
    }
  }
}
    
 void osc1 (OSCMessage &msg, int addrOffset){ 
oldState = (boolean) msg.getFloat(0);
  if(msg. isFloat ( 0 )) {
  uint8_t m = (ws2812fx.getMode() + 1) % ws2812fx.getModeCount();
  ws2812fx.setMode(m);
  ws2812fx.trigger(); 
  
  }

  
  OSCMessage msgOUT("/1/toggleLED");
  msgOUT.add(oldState);
  
  

  
 oldState = !oldState; 

  
 Udp.beginPacket(Udp.remoteIP(), destPort);
 msgOUT.send(Udp); // send the bytes
 Udp.endPacket(); // mark the end of the OSC Packet
 msgOUT.empty(); // free space occupied by message
  }

 ````

i tried things like that https://github.com/toblum/simple_ws2812_button_light/blob/4d1fcf9f94b6fe0283a8d257efaf537b472b747a/simple_ws2812_button_light/simple_ws2812_button_light.ino

but doesnt work, 

btw, i omit all the serials code
many thanks for your help @moose4lord i think im close to the solution, or maybe is not possible

leexorsist, 2018-02-25:
OK, i could already change the modes with different osc adresses, i use the guide of your first sugestion @moose4lord, but now the modes aren´t animated, 
-how can i do for obtain the modes are animated with each pushing osc?
this is what i do
```
void setup() {
    
  ws2812fx.init();
  ws2812fx.setBrightness(50);
  
}

void loop() {
  
ws2812fx.service();

  OSCMessage msgIN;
  int size;
  if((size = Udp.parsePacket())>0){
    while(size--)
      msgIN.fill(Udp.read());
    if(!msgIN.hasError()){
      msgIN.route("/1/toggleLED", osc1);
      msgIN.route("/2/toggleLED", osc2); 
      msgIN.route("/3/toggleLED", osc3);    
      
    }
  }
}
    
 void osc1 (OSCMessage &msg, int addrOffset){
  ws2812fx. setMode (14); 
  ws2812fx. setColor ( YELLOW );
  oldState = (boolean) msg.getFloat(0);
  if(msg. isFloat ( 0 )) {
  
    ws2812fx.trigger();
  
  OSCMessage msgOUT("/1/toggleLED");
  msgOUT.add(oldState);
  
  

  
 oldState = !oldState; 

  
 Udp.beginPacket(Udp.remoteIP(), destPort);
 msgOUT.send(Udp); // send the bytes
 Udp.endPacket(); // mark the end of the OSC Packet
 msgOUT.empty(); // free space occupied by message
  }
 }

  void osc2 (OSCMessage &msg, int addrOffset){
  ws2812fx. setColor ( BLUE );
  ws2812fx. setMode (11); 
  oldState = (boolean) msg.getFloat(0);
  if(msg. isFloat ( 0 )) {
  }
if (msg. isFloat ( 0 )) {
 
    ws2812fx.trigger();
  }
  
  OSCMessage msgOUT("/2/toggleLED");
  msgOUT.add(oldState);
  
  

  
 oldState = !oldState; 

  
 Udp.beginPacket(Udp.remoteIP(), destPort);
 msgOUT.send(Udp); // send the bytes
 Udp.endPacket(); // mark the end of the OSC Packet
 msgOUT.empty(); // free space occupied by message
  }

void osc3 (OSCMessage &msg, int addrOffset){
  ws2812fx. setColor ( GREEN );
  ws2812fx. setMode (FX_MODE_SCAN); 
  oldState = (boolean) msg.getFloat(0);
  if(msg. isFloat ( 0 )) {
  }
if (msg. isFloat ( 0 )) {
 
    ws2812fx.trigger();
  }
  
  OSCMessage msgOUT("/3/toggleLED");
  msgOUT.add(oldState);
  
  

  
 oldState = !oldState; 

  
 Udp.beginPacket(Udp.remoteIP(), destPort);
 msgOUT.send(Udp); // send the bytes
 Udp.endPacket(); // mark the end of the OSC Packet
 msgOUT.empty(); // free space occupied by message

  }

moose4lord, 2018-02-25:
You forgot to call ws2812fx.start() in your setup() function. That starts the animation.

leexorsist, 2018-02-25:
yes that works! i add this too
´´´
if(msg. isFloat ( 0 )) {
  
    ws2812fx.trigger();
    last_trigger = now;

now modes changes and animation running!, but i want to the animation are foward everytime i push the osc button, this is better for have midi notes and have rhythm, maybe if i chance the unsigned long "now"... thanks a lot @moose4lord

leexorsist, 2018-04-01:
i finally could make  triggered the osc whit that messages  i could change the set modes, it works well but, im want to the fx running while the osc is active, but any time i press the button to send osc msg the fx begins form the top, i want to stop the fx when i doesnt pressed the osc signal button and then when i will press  again and still hold it the fx running from where it stop last time,

i tried modify start from .ccp erasing reset _runtime( but doesnt work :( )

i modify  trigger too from .ccp (this makes reactive the fx to the message, i dont know if its correct but works)

any idea of how i can do this?

in .ccp :
```
void WS2812FX::trigger() {
  SEGMENT_LENGTH; 
  _triggered = true;
  _running = true;
}

void WS2812FX::start() {
  RESET_RUNTIME; // i tried erase it, but n othing happens
  _running = true;
}
```

this is mi code from the loop:

```
void setup() {   
  ws2812fx.init();
  ws2812fx.setBrightness(50);
}
void loop() {   
ws2812fx.service();

  OSCMessage msgIN;
  int size;
  if((size = Udp.parsePacket())>0){
    while(size--)
      msgIN.fill(Udp.read());
    if(!msgIN.hasError()){
      msgIN.route("/3/toggleLED", osc3); 
    }
  }
}
    
void osc3 (OSCMessage &msg, int addrOffset){
  
  oldState = (boolean) msg.getFloat(0);
  if(msg. isFloat ( 0 )) {
  ws2812fx. setColor ( GREEN );
  ws2812fx. setMode (37); 
   ws2812fx.trigger();
   
  OSCMessage msgOUT("/3/toggleLED");
  msgOUT.add(oldState);
  
 oldState = !oldState; 

  
 Udp.beginPacket(Udp.remoteIP(), destPort);
 msgOUT.send(Udp); // send the bytes
 Udp.endPacket(); // mark the end of the OSC Packet
 msgOUT.empty(); // free space occupied by message

  }

  
}
````
any idea would be awesome, thanks!

moose4lord, 2018-04-02:
You're calling ws2812fx.trigger() in your osc3() function, which is incorrect. To start and stop the animation, you should call the ws2812fx.start() and ws2812fx.stop() functions. It seems like you want something like this:

```c++
boolean state = true;
void osc3(OSCMessage &msg, int addrOffset) {
  if(state) {
    ws2812fx.setColor ( GREEN );
    ws2812fx.setMode (37); 
    ws2812fx.start();
  } else {
    ws2812fx.stop();
  }
  state = !state;
}
```

closed 2018-05-24 by kitesurfer1404