kitesurfer1404/WS2812FX #77: Running Lights

salvonos, 2018-03-02:
Hi there,
firstly tank you for your beautiful work.
I was wonder how to modify the preset "Running Lights" in order to go to white instead of black..

I am having a bit of trouble to achieve this.

moose4lord, 2018-03-02:
Running Lights is a tricky effect to wrap your head around. Lots of math :)
To get it run with white you have to invert all the colors, which can be done by subtracting the RGB color from 255 before setting the pixel. Like so:
```c++
red = 255 - red;
green= 255 - green;
blue= 255 - blue;
ws2812fx.setPixelColor(ledIndex, red, green, blue);
```

I used the new custom effect feature to create this function:
```
uint8_t stepCount = 0;
uint16_t mode_running_lights_white(void) {
  uint8_t w = 255 - ((ws2812fx.getColor() >> 24) & 0xFF);
  uint8_t r = 255 - ((ws2812fx.getColor() >> 16) & 0xFF);
  uint8_t g = 255 - ((ws2812fx.getColor() >>  8) & 0xFF);
  uint8_t b = 255 - (ws2812fx.getColor()         & 0xFF);

  float radPerLed = (2.0 * 3.14159) / LED_COUNT;
  for(uint16_t i=0; i < LED_COUNT; i++) {
    int lum = map((int)(sin((i + stepCount) * radPerLed) * 128), -128, 128, 0, 255);
    ws2812fx.setPixelColor(LED_COUNT - i - 1,
      255 - ((r * lum) / 256),
      255 - ((g * lum) / 256),
      255 - ((b * lum) / 256),
      255 - ((w * lum) / 256));
  }
  stepCount = (stepCount + 1) % LED_COUNT;
  return (ws2812fx.getSpeed() / LED_COUNT);
}
```
See the ws2812fx_custom_effect sketch in the WS2812FX examples folder to see how to implement a custom effect. Make sure you download the latest version of WS2812FX.

salvonos, 2018-03-04:
HI thank you for your help.

Sorry about my delaying in replaying. I have tried to use your code but I get full white without animation. I have also tried to add this to the library replacing static mode that I won't use. 

```
uint16_t WS2812FX::mode_static(void) {
  
    uint8_t w = 255 - ((SEGMENT.colors[0] >> 24) & 0xFF);
    uint8_t r = 255 - ((SEGMENT.colors[0] >> 16) & 0xFF);
    uint8_t g = 255 - ((SEGMENT.colors[0] >>  8) & 0xFF);
    uint8_t b = 255 - (SEGMENT.colors[0]         & 0xFF);

    float radPerLed = (2.0 * 3.14159) / SEGMENT_LENGTH;
    for(uint16_t i=0; i < SEGMENT_LENGTH; i++) {
        int lum = map((int)(sin((i + SEGMENT_RUNTIME.counter_mode_step) * radPerLed) * 128), -128, 128, 0, 255);
        
            Adafruit_NeoPixel::setPixelColor(SEGMENT_LENGTH - i - 1,
                                             255 - ((r * lum) / 256),
                                             255 - ((g * lum) / 256),
                                             255 - ((b * lum) / 256),
                                             255 - ((w * lum) / 256));
        
    }
    SEGMENT_RUNTIME.counter_mode_step = (SEGMENT_RUNTIME.counter_mode_step + 1) % SEGMENT_LENGTH;
    return (SEGMENT.speed / (SEGMENT_LENGTH));
}
```

However, I am still getting full white without animation.

moose4lord, 2018-03-05:
Hmmm, that's odd. Here is the complete sketch that's working for me. Maybe you can spot a difference between my sketch and yours.

```c++
#include <WS2812FX.h>

#define LED_COUNT 30
#define LED_PIN 5

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setColor(BLUE);
  ws2812fx.setSpeed(1000);
  ws2812fx.setCustomMode(mode_running_lights_white);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}

uint8_t stepCount = 0;
uint16_t mode_running_lights_white(void) {
  uint8_t w = 255 - ((ws2812fx.getColor() >> 24) & 0xFF);
  uint8_t r = 255 - ((ws2812fx.getColor() >> 16) & 0xFF);
  uint8_t g = 255 - ((ws2812fx.getColor() >>  8) & 0xFF);
  uint8_t b = 255 - (ws2812fx.getColor()         & 0xFF);

  float radPerLed = (2.0 * 3.14159) / LED_COUNT;
  for(uint16_t i=0; i < LED_COUNT; i++) {
    int lum = map((int)(sin((i + stepCount) * radPerLed) * 128), -128, 128, 0, 255);
    ws2812fx.setPixelColor(LED_COUNT - i - 1,
      255 - ((r * lum) / 256),
      255 - ((g * lum) / 256),
      255 - ((b * lum) / 256),
      255 - ((w * lum) / 256));
  }
  stepCount = (stepCount + 1) % LED_COUNT;
  return (ws2812fx.getSpeed() / LED_COUNT);
}
```

closed 2018-05-24 by moose4lord