kitesurfer1404/WS2812FX #78: add FastLED noise functions to WS2812FX

XerionXavier, 2018-03-03:
its beyond my ability but would anyone be interested in adding the functionality to WS2812FX similar to FastLED noise()? It creates random lava lamp like colors. I am using it in a similar esp-01 project that has a web interface to select different animations. I find WS2812FX more complete for a light strip ESP-01 LED-Web Based projects. See examples of functions inoise8() and inoise16().
Thanks

`// FastLED setup ----------   FastLED has to be declared BEFORE the Webserver     ---------------------
#include "FastLED.h"
FASTLED_USING_NAMESPACE

#if defined(FASTLED_VERSION) && (FASTLED_VERSION < 3001000)
#warning "Requires FastLED 3.1 or later; check github for latest code."
#endif

#define DATA_PIN      0     // for Huzzah: Pins w/o special function:  #4, #5, #12, #13, #14; // #16 does not work :( 
#define LED_TYPE      WS2812B
#define COLOR_ORDER   GRB
#define NUM_LEDS      60
CRGB leds[NUM_LEDS];

bool gReverseDirection = false;
static uint16_t dist;         // A random number for our noise generator.
uint16_t scale = 30;          // Wouldn't recommend changing this on the fly, or the animation will be really blocky.
uint8_t maxChanges = 48;      // Value for blending between palettes.

uint8_t frequency = 100;                                       // controls the interval between strikes
uint8_t flashes = 6;                                          //the upper limit of flashes per strike
unsigned int dimmer = 1;

uint8_t ledstart;                                             // Starting location of a flash
uint8_t ledlen;

CRGBPalette16 currentPalette(CRGB::Black);
CRGBPalette16 targetPalette(CloudColors_p); //pasted in from noise fastled end

//#define BRIGHTNESS       160
int BRIGHTNESS =           160;   // this is half brightness
int new_BRIGHTNESS =       160;   // shall be initially the same as brightness

#define MILLI_AMPERE      1800    // IMPORTANT: set here the max milli-Amps of your power supply 5V 2A = 2000
#define FRAMES_PER_SECOND  120    // here you can control the speed. With the Access Point / Web Server the 
// animations run a bit slower.

int ledMode = 13;                  // this is the starting animation

// Fire2012 by Mark Kriegsman, July 2012
// as part of "Five Elements" shown here: http://youtu.be/knWiGsmgycY
////
// This basic one-dimensional 'fire' simulation works roughly as follows:
// There's a underlying array of 'heat' cells, that model the temperature
// at each point along the line.  Every cycle through the simulation,
// four steps are performed:
//  1) All cells cool down a little bit, losing heat to the air
//  2) The heat from each cell drifts 'up' and diffuses a little
//  3) Sometimes randomly new 'sparks' of heat are added at the bottom
//  4) The heat from each cell is rendered as a color into the leds array
//     The heat-to-color mapping uses a black-body radiation approximation.
//
// Temperature is in arbitrary units from 0 (cold black) to 255 (white hot).
//
// This simulation scales it self a bit depending on NUM_LEDS; it should look
// "OK" on anywhere from 20 to 100 LEDs without too much tweaking.
//
// I recommend running this simulation at anywhere from 30-100 frames per second,
// meaning an interframe delay of about 10-35 milliseconds.
//
// Looks best on a high-density LED setup (60+ pixels/meter).
//
//
// There are two main parameters you can play with to control the look and
// feel of your fire: COOLING (used in step 1 above), and SPARKING (used
// in step 3 above).
//
[... 871 lines cut ...]

  delay(random8(frequency) * 100);                            // delay between strikes

} // lightning()

void blur() {

  uint8_t blurAmount = dim8_raw( beatsin8(3, 64, 192) );      // A sinewave at 3 Hz with values ranging from 64 to 192.
  blur1d( leds, NUM_LEDS, blurAmount);                        // Apply some blurring to whatever's already on the strip, which will eventually go black.

  uint8_t  i = beatsin8(  9, 0, NUM_LEDS);
  uint8_t  j = beatsin8( 7, 0, NUM_LEDS);
  uint8_t  k = beatsin8(  5, 0, NUM_LEDS);

  // The color of each point shifts over time, each at a different speed.
  uint16_t ms = millis();
  leds[(i + j) / 2] = CHSV( ms / 29, 200, 255);
  leds[(j + k) / 2] = CHSV( ms / 41, 200, 255);
  leds[(k + i) / 2] = CHSV( ms / 73, 200, 255);
  leds[(k + i + j) / 3] = CHSV( ms / 53, 200, 255);

  FastLED.show();

} // blur()
`

moose4lord, 2018-03-04:
I'm not sure what you mean. Are you looking for an implementation of the inoise_() functions built into FastLED? Or are you looking to have WS2812FX implement some of the effects demonstrated in the sketch you reference?

XerionXavier, 2018-03-04:
The latter. It would be cool to implement a similar type of effect in ws2812fx.

moose4lord, 2018-03-04:
Any of the 17 effects in particular?

moose4lord, 2018-03-04:
Ah, I should have guessed. Those are spectacular effects. Almost hypnotic. Kudos to you for using the noise function in such an innovative way.

I'll take a shot at reproducing something like those in WS2812FX, but I'm not hopeful. I looked at the noise functions in FASTled and the math makes my puny brain hurt. :) Even so, the WS2812FX library is pretty big as it is, and just barely fits in the standard Arduino's 32k flash RAM. I think all the code necessary for a noise function would blow up the library's memory footprint to the point where it wouldn't fit on an Arduino anymore, which would be unacceptable. Still, it may be able to be implemented as a custom effect for ESP implementations.

moose4lord, 2018-03-05:
Well, I have something working. My initial plan was to "borrow" the noise code from FastLED , but that quickly turned into a mess. So instead I created a sketch that included all of FastLED and wrote some code to bridge the FastLED's CRGB object and WS2812FX's plain old RGB ints. So you create the effect using FastLEDs features, then translate the CRGB colors to ints for WS2812FX to display.

Here is my demo implementation of your fillnoise8 effect:

[ws2812fx_noise.zip](https://github.com/kitesurfer1404/WS2812FX/files/1781284/ws2812fx_noise.zip)

As I suspected, compiling for Arduino creates a program that's too big to fit in the standard Arduino's 32k flash RAM. So the effects aren't something that can be added to the WS2812FX library. But you can certainly create a sketch using this technique for an ESP implementation.

tobi01001, 2018-03-10:
Awesome... May play with that a bit.

Is there a particular reason why you init the FASTLed first in setup? 
I don't think thats actually needed because you map the CRGB struct anyway to the particular neopixel.

moose4lord, 2018-03-10:
I think you're right. The lines that init a single FastLED led aren't needed.

tobi01001, 2018-03-11:
And no need to copy over if you just set the CRGB to the Neopixel address.

```C++
#include <FastLED.h>
#include <WS2812FX.h>

#define LED_COUNT 30
#define LED_PIN 5

void fillnoise8(void);
uint16_t myCustomEffect(void);

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

// declare some parameters for the FastLED functions

CRGB *leds; // Pixel address for FastLED CRGB

uint8_t gHue = 0; // rotating "base color" used by many of the patterns
static uint16_t dist = random(12345); // A random number for our noise generator.
uint16_t scale = 30;          // Wouldn't recommend changing this on the fly, or the animation will be really blocky.
uint8_t maxChanges = 48;      // Value for blending between palettes.
CRGBPalette16 currentPalette(CRGB::Black);
CRGBPalette16 targetPalette(CloudColors_p);

void setup() {
  Serial.begin(115200);

  
  // init WS2812FX to use a custom effect
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setColor(BLUE);
  ws2812fx.setSpeed(500);
  ws2812fx.setCustomMode(myCustomEffect);

  // FastLed Pixel-mapping
  leds = (CRGB*)ws2812fx.getPixels();
  
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();
}

uint16_t myCustomEffect() {
  fillnoise8(); // run fillnoise8 algorithm

  // update some FastLED properties in the background
  EVERY_N_MILLISECONDS(20) { // slowly cycle the "base color" through the rainbow
    gHue++;
  }

  EVERY_N_MILLISECONDS(10) { // Blend towards the target palette
    nblendPaletteTowardPalette(currentPalette, targetPalette, maxChanges);
  }

  EVERY_N_SECONDS(5) { // Change the target palette to a random one every 5 seconds.
    targetPalette = CRGBPalette16(CHSV(random8(), 255, random8(128, 255)),
                                  CHSV(random8(), 255, random8(128, 255)),
                                  CHSV(random8(), 192, random8(128, 255)),
                                  CHSV(random8(), 255, random8(128, 255)));
  }
  // set the animation speed based on the ws2812fx speed setting
  return (ws2812fx.getSpeed() / LED_COUNT);
}

void fillnoise8() {
  for (int i = 0; i < LED_COUNT; i++) {
    uint8_t index = inoise8(i * scale, dist + i * scale) % 255;
    leds[i] = ColorFromPalette(currentPalette, index, 255, LINEARBLEND);
  }
  dist += beatsin8(10, 1, 4);
}
`

moose4lord, 2018-03-11:
That's true, but only because, by happenstance, the internal structure of the CRGB object aligns with the Adafruit_Neopixel library's internal RGB array.  I might not be a good idea to depend on that always being true. Your shortcut would fail, for instance, with RGBW LEDs, which use 4 bytes per LED. LEDs that don't conform to the NEO_GRB ordering might not work either.

tobi01001, 2018-03-11:
Of course this has to be handled with some special care and will not work for RGBW strips. I think the order is not a problem as this should be handled low level in the library.

So it is and has to be a custom effect from library point of view and is user / application specific.