kitesurfer1404/WS2812FX #82: auto cycle

cergant, 2018-03-21:
How to make, that at inclusion was an auto-cycle, and at a choice in a web interface of any mode it and remained?

moose4lord, 2018-03-21:
Sorry, I don't understand your question. Are talking about the esp8266_webinterface example? Maybe you could rephrase your question.

cergant, 2018-03-21:
Yes, about him. I'm through Google. it is necessary that the alternation of all effects is switched on automatically or in a random order. and when one of them was selected in the webinterface, it was a cyclic

cergant, 2018-03-22:
Я уже сделал. boolean auto_cycle = true;
А в веб интерфейсе просто перед сменой эффекта нажимать "stop". После нажатия "play" все эффекты опять по кругу. Спасибо.

moose4lord, 2018-03-22:
Google translation:
"I did it already. boolean auto_cycle = true;
And in the web interface, just before changing the effect, press "stop". After pressing "play" all the effects are again in a circle. Thank you."

I think I understand now. You were trying to change modes in the web app and stop/start the auto-cycle feature. Glad you figured it out.

closed 2018-03-23 by cergant