kitesurfer1404/WS2812FX #86: trigering with PIR sensors

Andriejus, 2018-04-18:
Hello. Im not good in coding, so maybe I can find some help here. I want to light stairs at home with ws2812 leds and control effects from my phone. I have tried to use esp8266 webinterface with my ws2812 LED strip and it works fine. But I want to trigger effect with 2 PIR motion  sensors (one on the top and one on the bottom of stairs). And maybe add some standby mode (when no motion is detected). Has anyone done porject like this? Could you please share information how to do it?

moose4lord, 2018-04-19:
I have a couple PIR sensors in my parts drawer. I'll take a stab at it and get back to you in a day or two. What kind of ESP8266 and PIR sensor do you have?

Andriejus, 2018-04-20:
I have ESP8266 v1,0 12E. And HC-SR05 pir sensors. Also I want to use photoresistor:). Im thinking to make something like this
http://www.instructables.com/id/LED-Stair-Lighting/
It has standby mode (so I want to have this possibility to switch ON or OFF in webinterface:)
2 PIR sensors= if you trigger sensor on the bottom of the  stairs, so all animations should start from bottom. 
Also I have question about segments. My Stairs has 15 steps and Im thinking to use about 50leds per step, So is it possible to make 15 segments? Or only 10? I would like to add some effects for segments in my webinterface sketch. I think it would look very nice with segments:)

moose4lord, 2018-04-21:
Yikes! 15 steps with 50 LEDs per step is a LOT of LEDs. You'd need a 50 Amp power supply to power all those LEDs at full intensity. Maybe you're not planning on driving them at full intensity, but still, I would try to scale that back a little. How about using WS2812 strips that have 30 LEDs per meter? That would give you around 25-30 LEDs per step, which I think is more manageable and would work fine for your application.

I looked at the instructable that you referenced and it seems like that project has a solid design. Take note of how he wired the power to all of his strips. Notice he didn't just daisy chain the power from one step to the next, but instead feeds power to each strip individually. There's a small, but noticeable voltage drop in the power supply voltage as you go down the length of a strip. For very long strips you may have a solid 5 volts at the beginning of the strip, but only 3 volts or less at the end. Creating separate power supply feeds to each strip, as he has done, avoids that problem. This is mentioned in Adafruit's Neopixel best practices guide ([here](https://learn.adafruit.com/adafruit-neopixel-uberguide/best-practices)), which a must read.

Regarding the number of segments, the maximum is set to 10 in the WS2812FX library just to accommodate the tiny amount of memory in an Arduino. Since you're using an ESP8266, you have lots of memory and can safely increase the maximum number of segments to 20 or more by changing the constant at line 55 in the library's WS2812FX.h file:
```cpp
#define MAX_NUM_SEGMENTS 10
```

moose4lord, 2018-04-25:
I had a chance to play with my PIR sensors and WS2812Bs. A mock-up of your stair lighting application is working pretty well. I've attached a fritzing diagram to show how I wired up my ESP8266 dev board (a NodeMCU v1.0). First I recreated the sketch that @snikogos had created in his instructable project. that worked pretty well, so I then refactored the sketch to use the WS2812FX library. See attachments.

If you're having trouble with your project, take a look at what I did and it may help.

[image]
[stairs_demo.zip](https://github.com/kitesurfer1404/WS2812FX/files/1948814/stairs_demo.zip)
[stairs_ws2812fx.zip](https://github.com/kitesurfer1404/WS2812FX/files/1948815/stairs_ws2812fx.zip)

moose4lord, 2018-04-28:
The web interface example is just supposed to be a demonstration of how to create a web interface. Adding code for PIR and photo sensors seems pretty application specific, so probably doesn't really belong in the demo sketch. My 2 cents. But if you create a spiffy application, please share.

closed 2018-05-24 by kitesurfer1404