kitesurfer1404/WS2812FX #87: Variable brightness development idea

sushukka, 2018-04-25:
Hello,
I really love this library. Has more than enough different lighting animations so that there are no need to code your own algorithms anymore. Also customizability is in a good level. 

However, there is now one project which gives some troubles. I have a home theater room where there are 2x300 WS2812 LEDs along both sides of the room. Now the problem is that when the leds are turned on the projector image on big screen will be quite washed out. Instead of dimming the whole led strips down to very minimal level, it would be very nice to have the dimming so that the brightness changes gradually from set level to zero from beginning to the end of the strip (when approaching the big screen). Also I assume this would be a nice feat in other lighting projects too. 

I'm trying to figure out how to fork the library to do that, but haven't been able to figure it inner functionality yet. Maybe something like this:

- Having a method to set a variable telling if there are variable/gradual brightness mode activated
- If this variable is set: in service loop: call decreaseBrightness every time before calling Adafruit_NeoPixel::show()...problem probably with this approach is that the brightness changes would be in random places in different animations...so maybe:
- Having a separate variable brightness setting method where do a check which led position will currently be updated and apply normalized brightness to this position. Something like: this_led_brightness = current_led_pos * global_brightness_level / led_strip_length. This would gradually brighten the led strip when moving to the end. To do the opposite (dim when moving to the end) naturally: this_led_brightness = (led_strip_length-current_led_pos) * global_brightness_level / led_strip_length. This method would be then run before every NeoPixel::show call in service loop. 

User would have then dimToEnd and dimToBeginning methods to activate the gradual dimming thorough the led strip and all the animations. 

This is just an idea! If you don't seem this feasible/useful, just let my mumbling be. :smiley:  However, if I do this by my own, would something like this work, or will it break the whole internal structure somehow.

moose4lord, 2018-04-26:
You have an interesting idea. I think what you're talking about is running a post-process after the normal LED mode function that would gradually dim the entire strip of LEDs from beginning to end.

It sounds doable, but I can foresee a couple problems.

1) Some of the mode functions use a fade function to gradually dim some of the LEDs (like COMET, LARSON_SCANNER, TWINKLE and FIREWORKS). Adding a dimming post-process would essentially "double fade" the LEDs, causing them to fade very quickly.

2) Brightness is not linear. Brightness=100 isn't half as bright as brightness=200. In fact, brightness between 50-250 look pretty much the same to my tired, old eyes. Setting 300 LEDs to a gradient from 0-255 would still be a very bright string of LEDs. You might try a sketch something like this to see if you get the effect you're after:

```c++
#include <Adafruit_NeoPixel.h>

#define LED_COUNT 300
#define LED_PIN D1

Adafruit_NeoPixel pixels = Adafruit_NeoPixel(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  pixels.begin();
  for(int i=0; i<LED_COUNT; i++) {
    int intensity = i * 256 / LED_COUNT;
    pixels.setPixelColor(i, intensity, intensity, intensity);
  }
  pixels.show();
}
```

I created a demo sketch that applies the dimming post-processing (see attached). It's implemented as a custom effect and can be used to try out various WS2812FX effects with the dimming applied. There's one "gotcha" though. The WS2812FX library doesn't normally expose the animation functions that it uses internally. For my demo sketch to work it needs access to the lib's internal animation functions, which can be accomplished by inserting "public:" before line 355 in the ws2812fx.h file, like so:
```c++
  private:
    void
      strip_off(void),
      fade_out(void);

  public: // add this line
    uint16_t
      mode_static(void),
```
Yes, it's a bit of a hack, but good enough for some quick testing. :)
[ws2812fx_custom_effect.zip](https://github.com/kitesurfer1404/WS2812FX/files/1953062/ws2812fx_custom_effect.zip)

sushukka, 2018-04-30:
Niiice, the demo works just as expected! Just changed the polarity of dimming by "i - seg.start" -> "seg.stop - i". But as you said I need to think about the intensity algorithm a little bit. The linear 0-255 dimming is not good --> have to make it steeper or just use only those lower level dimming values. 
Also it seems that this dimming overlay breaks all the wipe methods where you should have the older value in the background. Now the wipes ends to be variations of the almighty larson scanner. I assume that the background colors are not saved and from WS2812FX pov have RGB(0,0,0) or null. The dimming overlay method then clears the strips backround color and ends up showing only a moving dots...maybe... Rainbow_cycle works but it probably need to have the old colors saved for transitioning. 

Also one additional question raised when progressing in this project...multiple WS2812FX instances.
I have two 292x WS2812 strips installed in separate pins on ESP8266. I tried to copycode dummystyle like this:
`
/*
  Demo sketch for creating custom effects.
  
  Keith Lord - 2018

  LICENSE

  The MIT License (MIT)

  Copyright (c) 2018  Keith Lord 

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sub-license, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in
  all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
  THE SOFTWARE.
  
  CHANGELOG
  2018-02-26 initial version
*/

#include <WS2812FX.h>

#define LED_COUNT 292
#define LED_PIN1 D3
#define LED_PIN2 D2

WS2812FX ws2812fx1 = WS2812FX(LED_COUNT, LED_PIN1, NEO_RGB + NEO_KHZ800);
WS2812FX ws2812fx2 = WS2812FX(LED_COUNT, LED_PIN2, NEO_RGB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  //-- Turn relay=LED power source on --
  pinMode(D0, OUTPUT);
  digitalWrite(D0, HIGH); 
  
  ws2812fx1.init();
  ws2812fx2.init();
  
  ws2812fx1.setBrightness(255);
  ws2812fx2.setBrightness(255);

  ws2812fx1.setCustomMode(myCustomEffect1);
  ws2812fx2.setCustomMode(myCustomEffect2);
  
  ws2812fx1.setSegment(0, 0, LED_COUNT - 1, FX_MODE_CUSTOM, (const uint32_t[]) {GREEN, BLUE, YELLOW}, 5000, false);
  ws2812fx2.setSegment(0, 0, LED_COUNT - 1, FX_MODE_CUSTOM, (const uint32_t[]) {GREEN, BLUE, YELLOW}, 5000, false);

  ws2812fx1.start();
  ws2812fx2.start();
}

void loop() {
  ws2812fx1.service();
  ws2812fx2.service();
}

uint16_t myCustomEffect1(void) {
  // start by runnng a built-in mode function
  uint16_t delay = ws2812fx1.mode_larson_scanner();
  //uint16_t delay = ws2812fx1.mode_tricolor_chase();

  // then overlay with the dimming effect
  WS2812FX::Segment seg = ws2812fx1.getSegment(); // get the current segment
  int seg_length = seg.stop - seg.start + 1;
  for(uint16_t i=seg.start; i<=seg.stop; i++) {
    uint32_t color = ws2812fx1.getPixelColor(i);
    int r = (color >> 16 & 0xff) * (seg.stop - i) / seg_length;
    int g = (color >>  8 & 0xff) * (seg.stop - i) / seg_length;
    int b = (color       & 0xff) * (seg.stop - i) / seg_length;
    ws2812fx1.setPixelColor(i, r, g, b);
  }
  return delay;
}
uint16_t myCustomEffect2(void) {
  // start by runnng a built-in mode function
  uint16_t delay = ws2812fx2.mode_larson_scanner();
  //uint16_t delay = ws2812fx2.mode_tricolor_chase();

  // then overlay with the dimming effect
  WS2812FX::Segment seg = ws2812fx2.getSegment(); // get the current segment
  int seg_length = seg.stop - seg.start + 1;
  for(uint16_t i=seg.start; i<=seg.stop; i++) {
    uint32_t color = ws2812fx2.getPixelColor(i);
    int r = (color >> 16 & 0xff) * (seg.stop - i) / seg_length;
    int g = (color >>  8 & 0xff) * (seg.stop - i) / seg_length;
    int b = (color       & 0xff) * (seg.stop - i) / seg_length;
    ws2812fx2.setPixelColor(i, r, g, b);
  }
  return delay;
}
`
Only "fx2" strip will do anything, fx1 stays black. Probably some noob error from me...

moose4lord, 2018-04-30:
I see what you mean about the wipes not working. I hadn't looked at that mode. Wipes are one of the modes that don't update the entire string of LEDs, only the LEDs that change from one animation frame to the next. The twinkle and chase modes have this problem as well.

Adafruit has a nice write-up about  the nonlinear brightness issue and how to correct for it in their [Gamma Correction](https://learn.adafruit.com/led-tricks-gamma-correction/the-issue) article.

Running two instances of ws2812fx simultaneously is usually not a problem. Running wipe on one strip and chase on another works fine. Unfortunately, custom effects are a special case. You can only configure one custom effect in a sketch. Trying to configure two custom effects, like you've done, results in only one of them being executed. I'll have to look at fixing that.

As an aside, you might want to avoid using D3 (GPIO0) in your sketch. D3 is sampled at boot time and can cause odd problems depending on what circuit you connect it to. It's safer to use D1 and D2, which have no restrictions. Google "esp8266 bootloader" if you want to read the gory details.

sushukka, 2018-05-03:
Hmm, have had some strange problems with some ESP8266 pins. Don't remember if it has been D3. Have tried to avoid D0, but probably I have mixed GPIOs and actual pins in my mind here... 
I tried the normal auto-cycle example sketch and it worked nicely with two ws2812fx instances just like you said. 
But about the dimming stuff, this seems to mess up the whole library so it's better not to disturb you anymore with this. I'll choose some animations I like the most and make own modded versions of them.  However, I would suggest you adding your example-variable_brightness-larson-scanner at least in the list of your animations as it's slightly different than the others and looks cool. :)

moose4lord, 2018-05-04:
Actually, I'm intrigued by your idea of overlaying effects. I think it has wider applications then just the dimming scheme, so I'm going to pursue it a little more. The gamma correction thing has been rattling around in my head for a while, so I need to do something with that too, especially since recent versions of the Adafruit_NeoPixel library includes a gamma correction table. Stay tuned...

sushukka, 2018-05-04:
Heh, nice to hear. :)
Having post-processing option/overlay would definitely broaden the possibilities enormously. Maybe squeeze a little bit the available "stock" animations and give the possibility to combine basic sweeps, wipes etc, with twinkles, dimming... 

That gamma correction problem is a good topic. I have faced this with multiple projects where you need to have a brightness option...and so far made it in the lazy linear way. :/ However, in some projects where I have had a breathing effect I've coded some "slowiness" in the lower end of the brightness scale and ruled out the smallest values to keep leds turning off completely making the breath effect smooth enough. Have to check NeoPixels update if it can streamline this manual tuning.  

The problem with this library now is its diversity; it's hard to find the best animations to the project on the hand. I'm gradually starting to remember what sweep, blink or scanner does what, but still I spend quite some time testing the different alternatives every time I use this library. It would be nice to have a simple youtube video showing all the alternatives or if one wants to be very ambitious, some small javascript running those algorithms and showing the chosen animation with a small virtual led strip. But again, don't take any pressure. :) Just throwing some ideas on the table. :)

sushukka, 2018-05-11:
Amag! 😮 Lots of fade/gamma related updates have appeared. 😃 👍 Seems to require little time for familiarizing the new possibilities. Now terrible busy with all boring life responsibilities, but I'll try to dig in to this new options at the start of the next week. However, no need to keep this open as this wasn't even a real issue but feature request. Nice work! Also Domoticz developers managed to get two-way parameter passing working with MySensors nodes which means that WS2812FX user could now just send the wanted animation id (0-56) directly from homeautomation controller to the node making the sketches more versatile.

closed 2018-05-11 by sushukka