kitesurfer1404/WS2812FX #89: Async and delays dont mix!

debsahu, 2018-04-27:
I was trying out Async stuff and it **crashes and burns** ESP8266 because of delay in line 78. Is there a way we could do either below?

```
#ifndef ESP8266
  delay(1);
#endif
```

or

```
#if defined(ESP31B) || defined(ESP32)
  delay(1);
#endif
```

Let me know if you want me to create a PR or caveats of doing something like this?

DeuxVis, 2018-04-27:
I'm not able to answer, but could you elaborate on the "crashes and burns" meaning ? Sounds a bit too alarming as it is.

debsahu, 2018-04-27:
I mean WDT reset (it was an exaggeration!) and the traceback of the cause on ESP Exception Decoder is yield() called in delay(). Source of these errors: https://github.com/marvinroger/async-mqtt-client/blob/master/docs/1.-Getting-started.md#fully-featured-sketch There are no delays anywhere in this code except in service(). That is there because of ESP32 specific issue and not related to ESP8266.

moose4lord, 2018-04-28:
I'd be interested in seeing what you're doing. As far as I know, as long as you're not calling delay() or yield() in one of your async callbacks, it should work ok. You should not be calling service() in an async callback.

debsahu, 2018-04-28:
Also forgot to mention that sometimes the WDT resets originate from TaskScheduler as well. In the [documentation] (https://github.com/arkhipenko/TaskScheduler/wiki/Full-Document), delay() will cause WDT resets.

moose4lord, 2018-04-29:
I don't know. This makes me uneasy. I don't understand why removing delay(1) would fix a WDT reset problem. Delay() feeds the watchdog timer, so it should fix flakey reset problems, not create them. And calling delay() before the NeoPixel show() function disables interrupts seems like a prudent thing to do. I look at it as sort of giving one last chance for the background (WiFi) processes to run before interrupts are disabled as the LEDs are updated.

I've seen too many wiki posts about odd ball, unexplainable issues that were miraculously fixed by inserting a delay() at the right spot in a sketch, to feel good about removing the delay().

debsahu, 2018-04-29:
There is no need of delay() for esp8266, see https://github.com/adafruit/Adafruit_NeoPixel/blob/4d0987ede8f8c204ff333de9a2fcbb70c646b7a5/Adafruit_NeoPixel.cpp#L113 show() is handled by the sdk.

moose4lord, 2018-04-29:
The NeoPixel code potentially pauses before disabling interrupts, but it's not guaranteed and doesn't call esp_yield() to take care of managing background processes (most notably the TCPIP stack). So I don't think it's the same as calling delay().

debsahu, 2018-04-30:
I think it's the burden on users to add delays, yield etc if they see issues. The idea of a library is to wrap complex things into a easy to use interface for users. Our own wiki for this repo touts that there are no delays, which is not true as of now. An ESP32 specific issue, needs to be addressed by a complier flag. This delay was introduced [in this PR](https://github.com/kitesurfer1404/WS2812FX/commit/fa640e91ccd9408cc2758782c40ef7613036c983).

moose4lord, 2018-07-19:
This conversation seems to have run its course, so I'm closing it. If someone has something to add, please reopen.

closed 2018-07-19 by moose4lord

sprior, 2019-01-01:
Just a new data point.  I'm seeing intermittent wdt resets inside WS2812FX::service(), usually less than once/hour.  I'm running a Wemos D1 R2 (8266) with a total of 93 Neopixels, external power supply driving the NeoPixels.  
Brightness set to 50
Segments as follows:
   seg 0, start 0, end 0, mode 11, speed 1000
   seg 1, start 1, end 8, mode 42, speed 1000
   seg 2, start 9, end 20, mode 42, speed 1500, reversed
   seg 3, start 21, end 36, mode 42, speed 2000
  seg 4, start 37, end 60, mode 42, speed 4000, reversed
  seg 5, start 61, end 92, mode 42, speed 6000

I'm printing a message to Serial before the call to service() and right after its return and caught the wdt reset between the two.

moose4lord, 2019-01-01:
@sprior, you should probably open an new issue and post your sketch. Unless you're doing a lot of Async network processing, your WDT resets probably have nothing to do with this issue.

sprior, 2019-01-02:
Done, thanks.