kitesurfer1404/WS2812FX #91: Copy constructor doesn't work

Kenny11CZ, 2018-05-05:
## Problem
Copy constructor doesn't copy attributes but it is not permitted, which leads to fails. 

## Example
```c++
#include <ws2812fx.h>

class LED {
public:
  LED(int pin, int count) : pin(pin), count(count) {}

  bool setup() {
    if(count = 0) {
      return false;
    }
    ws2812fx = WS2812FX(60, 5, NEO_RGB + NEO_KHZ800);
    ws2812fx.init();
    ws2812fx.setBrightness(255);
    ws2812fx.setSpeed(1000);
    ws2812fx.setColor(0x007BFF);
    ws2812fx.setMode(FX_MODE_STATIC);
    ws2812fx.start();
    ready = true;
    return true;
  }

  void loop() {
    if(!ready) {
      return;
    }
    ws2812fx.service(); // Here it fails
  }

private:
  bool ready = false;
  int count = 0;
  WS2812FX ws2812fx = WS2812FX(0,0,0);
  int pin;
};
// LED::setup() is called in main setup() function, same for loop()
```

## Solution:
Permit default copy constructor (`WS2812FX ( const WS2812FX & ws) = delete;`) or implement deep copy.

moose4lord, 2018-05-05:
Not sure what you mean. I don't see where a copy constructor would be used in your code snippet.

Be that as it may, the WS2812FX library barely fits on some Arduinos, so I'm very conscious about adding features that have limited use. I think a copy constructor falls in that category; nice to have, but I don't think many people would use it, so probably not worth the cost in bytes of flash memory. But you're welcome to do a pull request and see if it gets traction.

Kenny11CZ, 2018-05-05:
Oh, actually there is called operator=, but since that isn't overriden/implemented, default copy constructor is called, creating shallow copy. But i see why you don't want to implement that, but you could write some kind of warning to the end of the readme or something :smiley: 

Anyway, thanks for response and for this great library, peace :v: .

closed 2018-05-05 by Kenny11CZ