kitesurfer1404/WS2812FX #94: Divide by zero in mode_breath?

debsahu, 2018-05-11:
https://github.com/toblum/McLighting/issues/166 pointed out that if brightness is zero, mode_breath crashes. I'm guessing maybe we got to check divide by zero
https://github.com/kitesurfer1404/WS2812FX/blob/6ce37240ec411b24baa0444597bb88d828555569/src/WS2812FX.cpp#L555-L558

flagstone78, 2018-05-12:
mode_fade() does the same thing on line 576 to 579.

moose4lord, 2018-05-12:
Thanks for catching this. I've never been happy with the breath or fade modes. Breath doesn't work the way it's supposed to, since it tries to set the frame delay less the MIN_SPEED. And both breath and fade incorrectly apply the brightness setting twice, once in the WS2812FX code and again in the Adafruit_NeoPixel setPixelColor() function. They both look OK, but the implementation is flawed. I'll have to go back and look at those.

debsahu, 2018-05-24:
Closing, thank you.

closed 2018-05-24 by debsahu