kitesurfer1404/WS2812FX #95: How exactly is MODE_MULTI_STROBE working?

szmolnar, 2018-05-15:
Hi!

I'm trying to use MODE_MULTI_STROBE.
ws2812fx.setSegment(0, 0, 23, FX_MODE_MULTI_STROBE, BLUE, 100, false);

It suppose to make changes based on the speed settings (100) but changing it it looks like it has no effect at all.
Can you post a sample code?

Thanks!

kitesurfer1404, 2018-05-15:
That mode might be broken, since speed was changed to 16 bit values and was inverted in it's behavior.
In general the idea was: speed 0-10: single flash from slow to fast, speed 11-20 double flash from slow to fast, 21-30 tripple flash... and so on. Don't quote me on the exact values.

I think this needs fixing.

moose4lord, 2018-05-21:
This is indeed a bug introduced when the behavior of the speed parameter was changed a few months ago. I have fixed it and will deploy an update soon.

For future reference, the multi-strobe mode has two alternating phases, a flashing phase and an off phase. The duration of the flashing phase depends on the speed value, and is approximately (speed * 0.0007) seconds. The duration of the off phase depends on the last digit of the speed value, where "0" gives about a 1 second off phase, all the way up to "9" which gives about a 0.2 second off phase.

closed 2018-05-24 by kitesurfer1404