kitesurfer1404/WS2812FX #96: SetSegment Problem

Derodin, 2018-05-16:
How do I solve the following problem?

I have created a command which makes the following:
```
server.on("/all", []() {
   strip.setSegment(0, 0, 119, FX_MODE_STATIC, (const uint32_t[]) {0xffffff, 0xffffff, 0xffffff}, 255, false);
});
```

i Created 2 more Commands with:

```
server.on("/front", []() {
   strip.setSegment(0, 0, 59, FX_MODE_STATIC, (const uint32_t[]) {0xffffff, 0xffffff, 0xffffff}, 255, false);
   strip.setSegment(0, 60, 119, FX_MODE_STATIC, (const uint32_t[]) {0x000000, 0x000000, 0x000000}, 255, false);
})
```
This makes the first 0-60 LED lights up and the next 60-120 are off. This is also done in the test :)

**BUT**

With this Command:
```
server.on("/back", []() {
   strip.setSegment(0, 0, 59, FX_MODE_STATIC, (const uint32_t[]) {0xffffff, 0xffffff, 0xffffff}, 255, false);
   strip.setSegment(0, 60, 119, FX_MODE_STATIC, (const uint32_t[]) {0x000000, 0x000000, 0x000000}, 255, false);
})
```
This makes the first 0-60 LED lights off and the next 60-120 are on. But the first 60 leds going dont off.

how can i fix this Problem?

Lord-FEAR, 2018-05-16:
segment numbers must be different
```
server.on("/back", []() {
   strip.setSegment(0, 0, 59, FX_MODE_STATIC, (const uint32_t[]) {0x000000, 0x000000, 0x000000}, 255, false);
   strip.setSegment(1, 60, 119, FX_MODE_STATIC, (const uint32_t[]) {0xffffff, 0xffffff, 0xffffff}, 255, false);
})
```

Derodin, 2018-05-17:
Okay i try this.

closed 2018-05-24 by moose4lord