kitesurfer1404/WS2812FX #97: email still getting bounced back

moose4lord, 2018-05-21:
Hi Harm, my emails to your *.org address are still getting bounced back. I've sent a couple emails to your hotmail address. Did you get them?

Thx,
Keith

kitesurfer1404, 2018-05-22:
I cannot deactivate the forwarding - because hotmail now bounces the verification mail. I get your mails, I also replied from hotmail. Don't care about the bounces. I'll try to deactivate or whitelist (not so easy, because outlook/hotmail/live now sends from 10 billion cloud servers :( whitelisting all increases spam by magnitudes )

kitesurfer1404, 2018-05-22:
Now I'm locked out completely. M$ wants me to add my phone number - just to tell me that that number is already associated with my account. Well, how many phone numbers do one need these days?! 
--> let's try the gmail way: kitesurfer1404 @ gmail.com

closed 2018-05-24 by moose4lord