kitesurfer1404/WS2812FX #101: Number of LEDs limited to 278

Deadeye5589, 2018-06-02:
Hello,

I would like to use this library in combination with a 500 LED SK6812 RGBW stripe. Upon testing the library with the default examples provided,  I encounted what seems to be a limitation to a maximum of 278 LEDs. Any bigger number and there won't be any output from the program. Is there a technical limitation that results in this number? 278 is already bigger than 8 bit and works well. So it does not seem to be a problem with the size of the variable. Also the neo pixel library which is used in the background works with the whole 500 pixels. 

BR Chris

kitesurfer1404, 2018-06-02:
Hi. Most likely you're running out of RAM. See https://learn.adafruit.com/adafruit-neopixel-uberguide/the-magic-of-neopixels#faq-8
The bare NeoPixel lib uses some RAM, than additional to that comes the WS2812FX, every additional code (the examples) eat RAM as well when running. So in the end it highly appends on all these factors on the given hardware, how many LEDs you can drive. There is no limitation to 278 LEDs in this lib.

You might want to add some lines of code to print the free RAM while running your code for debugging his issue.

Hope this helps
Harm

kitesurfer1404, 2018-06-02:
PS: What kind of hardware are you using? Which Arduino?

Deadeye5589, 2018-06-02:
Hi kitesurfer,

I'm using a simple arduino uno. So not the most potent hardware. But the RAM might be indead the root cause for this problem. If I configure the code to RGB pixels instead of RGBW I can increase the number of LEDs 370 before the software stops working. This fits the math 278 * 4 / 3 = 370,66. I will give it another try with some more potent hardware. I still should have some ESPs and Megas lying around.