kitesurfer1404/WS2812FX #106: Delay with segments

Andriejus, 2018-07-19:
Hello. Can someone give me directions how to change the code (what function or syntax to use.. im not a programmer) in order to show segments with delay. I have 15 segments and I want that each segment is shown with some delay for example 1 second. 

I have something like this now:
void loop() {
  ws2812fx.service();
    for(int i=0; i<NUM_STEPS; i++) {
      ws2812fx.setSegment(i, i * PIXELS_PER_STEP, (i * PIXELS_PER_STEP) + PIXELS_PER_STEP - 1,
        FX_MODE_COLOR_WIPE,(const uint32_t[]) {GREEN, RED, BLACK}, DEFAULT_SPEED, false);
    }
  }

moose4lord, 2018-07-19:
If I understand your question correctly, you want to have only one segment active at a time. You can do that by creating a loop which changes the segment settings every iteration, making one segment active and all the other segments static black.

Here's my code snippet:
```cpp
#define LED_FIRST (i * PIXELS_PER_STEP)
#define LED_LAST  ((i * PIXELS_PER_STEP) + PIXELS_PER_STEP - 1)

// init some global variables
uint32_t colors[] = { GREEN, RED, BLACK };
int stepDelay = 2000; // delay between changing the active segment (2000ms)
long lastStep = 0; // timestamp of last active segment update
int activeStep = 0; // the active segment number

void loop() {
  ws2812fx.service();

  long now = millis();
  if(now > lastStep + stepDelay) { // time to update the active segment?
    for(int i=0; i<NUM_STEPS; i++) { // loop through the array of segments
      if(i == activeStep) { // if it's the active segment, initialize it
        ws2812fx.setSegment(i, LED_FIRST, LED_LAST, FX_MODE_COLOR_WIPE, colors, DEFAULT_SPEED, false);
      } else { // if not the active segment, make it static black
        ws2812fx.setSegment(i, LED_FIRST, LED_LAST, FX_MODE_STATIC, (uint32_t) BLACK, DEFAULT_SPEED, false); // static black
      }
    }
    ws2812fx.start(); // reset the segment runtimes
    lastStep = now; // update the last timestamp
    activeStep = (activeStep + 1) % NUM_STEPS; // increment the active segment number
  }
}
```

Andriejus, 2018-07-19:
Thank you for answer. Maybe i can use it later, but now i want that all segments would be active. These segments are on my stairs and i want that second segment would show animation after 1 second after the first segment and so on. So lets say my last segment number 15 would show animation after 15 seconds. I hope you understood what i want:) thank you again.

moose4lord, 2018-07-19:
Ah, I see. I think you can use a similar technique. Modify the two lines in my previous snippet as shown here:
```cpp
#define LED_FIRST (i * PIXELS_PER_STEP)
#define LED_LAST  ((i * PIXELS_PER_STEP) + PIXELS_PER_STEP - 1)

// init some global variables
uint32_t colors[] = { GREEN, RED, BLACK };
int stepDelay = 2000;
long lastStep = 0;
int activeStep = 0;

void loop() {
  ws2812fx.service();

  long now = millis();
  if(now > lastStep + stepDelay) {
      if(i <= activeStep) { // modify this line
        ws2812fx.setSegment(i, LED_FIRST, LED_LAST, FX_MODE_COLOR_WIPE, colors, DEFAULT_SPEED, false);
      } else {
        ws2812fx.setSegment(i, LED_FIRST, LED_LAST, FX_MODE_STATIC, (uint32_t) BLACK, DEFAULT_SPEED, false);
      }
    }

    ws2812fx.start();
    lastStep = now;
    activeStep++; // modify this line
  }
}
```
That should cause the segments to animate in sequence after a short delay.

Andriejus, 2018-07-20:
Thank you very much! I will try to play with this code and develope my project further!

moose4lord, 2018-10-23:
This discussion seems to have run its course, so closing.

closed 2018-10-23 by moose4lord