kitesurfer1404/WS2812FX #107: start different animations

djaronbenjamin, 2018-07-20:
Hello,

I'm trying to start different animations with the push of a button.
I can start a main Animation with no problems, but when i push a button my next animation won't start until i release the button and it only runs for a second and returns to the main one again.

I'm trying to make a wheel of fortune with a Rainbow Cycle always running and when the wheel begint tot turn it starts a different Animation like a Strobe or something.

Thanks in advance for any help.

Here is my code i'm using now:

#include <WS2812FX.h>

#define LED_COUNT 40
#define LED_PIN 7
#define BUTTON_PIN 10
#define TIMER_MS 2800

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

unsigned long last_change = 0;
unsigned long now = 0;

// variables will change:
int ButtonState = 0;         // variable for reading the pushbutton status

void setup() {
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setSpeed(200);
  ws2812fx.setColor(0xFFFFFF);
  ws2812fx.setMode(0);
  ws2812fx.start();
}

void loop()
{
  now = millis();

  ButtonState = digitalRead(BUTTON_PIN);

  ws2812fx.service();

  if (now - last_change > TIMER_MS)
  {
    ws2812fx.setMode(FX_MODE_RAINBOW_CYCLE);
    last_change = now;
  }

  if (ButtonState == 1)
  {
    ws2812fx.setMode(26);
  }
}

moose4lord, 2018-07-21:
Hmmm, I'm not sure if this is exactly what you're looking for, but here's a sketch that monitors a button. If the button is not pressed, the rainbow animation is run, and if the button is pressed, the strobe animation is run. I assumed BUTTON_PIN has a pull-up resistor to VCC and pushing the button pulls the GPIO pin to ground.

```cpp
#include <WS2812FX.h>

#define LED_COUNT 30
#define LED_PIN 5
#define BUTTON_PIN D2

WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

// variables will change:
int ButtonState = 1; // variable for reading the pushbutton status

void setup() {
  pinMode(BUTTON_PIN, INPUT);

  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setSpeed(200);
  ws2812fx.setColor(0xFFFFFF);
  ws2812fx.setMode(FX_MODE_RAINBOW_CYCLE);
  ws2812fx.start();
}

void loop() {
  ws2812fx.service();

  if(ButtonState != digitalRead(BUTTON_PIN)) {  // check if the button state has changed
    ButtonState = !ButtonState;
    if(ButtonState) {
      ws2812fx.setMode(FX_MODE_RAINBOW_CYCLE);
    } else {
      ws2812fx.setMode(FX_MODE_STROBE);
    }
    ws2812fx.start();
  }
}
```
When you run setMode() in the loop() section to change the mode dynamically, it's a good idea to rerun ws2812fx.start() as well, which resets the animation.

djaronbenjamin, 2018-08-10:
Thank you very much! That worked for me! @moose4lord

closed 2018-08-10 by djaronbenjamin